meatpiHQ/wican-fw #1: Some Questions

mickeyl, 2022-05-18:
Congratulations for releasing the interface and the software!

Having just purchased the WiCAN, I wonder about some things ­– perhaps you can clarify.

1.) What is "the native" protocol on the TCP socket? Right now it looks like "realdash" and "saavycan" protocols are implemented ­– guessing that the former refers to https://github.com/janimm/RealDash-extras/blob/master/RealDash-CAN/realdash-can-protocol.md, could you point me to a documentation for the saavycan protocol?

2.) Is the current firmware implementing ISOTP on the controller level, so that we can send the full 4095 bytes of payload in one go?

3.) Can we configure the CAN bitrates through the API? For the use with arbitrary vehicles, it would be a pity, if customers would have to edit the settings via web.

4.) Is there any documentation beyond the https://github.com/meatpiHQ/programming_examples/tree/master/CAN ­– which seems to refer to VSCAN API. Is this the same API that slcand speaks?

5.) Does the device support BLE5 queued writes?

6.) Are all the supported protocols active _in parallel_?

Thanks a lot. Looking forward to working with the device.

meatpiHQ, 2022-05-18:
@mickeyl Thanks for the support!

**1.) What is "the native" protocol on the TCP socket? Right now it looks like "realdash" and "saavycan" protocols are implemented ­– guessing that the former refers to https://github.com/janimm/RealDash-extras/blob/master/RealDash-CAN/realdash-can-protocol.md, could you point me to a documentation for the saavycan protocol?_**

There is no native protocol you can choose one on the 3 currently supported protocols(ReadDash, slcand and saavycan), the link you referred to is for ReadDash.
SavvyCAN is an alternative to BUSmaster, a monitoring application that supports windows/mac/linux. Here’s the link.
https://github.com/collin80/M2RET/blob/ad9d5af67d8e7c3330c0281e0403c2750e023f1c/CommProtocol.txt

**2.)Is the current firmware implementing ISOTP on the controller level, so that we can send the full 4095 bytes of payload in one go?**
The current firmware doesn’t support ISOTP, but this something I need look into.

**3.)	Can we configure the CAN bitrates through the API? For the use with arbitrary vehicles, it would be a pity, if customers would have to edit the settings via web.**

Yes, if you’re using slcand protocol it can be set by API please refer to the provided “programming examples”. Slcand is a very simple, you can set up the channel by just opening a tcp connection and sending an ascii command. 
If you’re using RealDash, you’ll have to set it manually in the web interface since realdash protocol doesn’t not offer any method to set the bitrate. 

**4.)	Is there any documentation beyond the https://github.com/meatpiHQ/programming_examples/tree/master/CAN ¬– which seems to refer to VSCAN API. Is this the same API that slcand speaks?**

Yes the VSCAN api is the same as slcand, here’s the documentation link from VSCOM.
https://www.vscom.de/429_download.htm

**5.)	Does the device support BLE5 queued writes?**

For now the ble function is very basic and it uses indication to send data, to make it compatible with Realdash. Please have a look at the firmware and the refence api
https://github.com/meatpiHQ/wican-fw/blob/main/main/ble.c
https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32c3/api-reference/bluetooth/esp_gatts.html

**6.)	Are all the supported protocols active in parallel?**

No, currently you can choose only one in the web interface. But this might be a useful feature to consider.

Daniboum, 2023-02-28:
I recently got my WiCAN USB. Great stuff! You made a real good product. 

1 functionality I'm also missing is ISOTP, ISO 15765-2, in the firmware, question 2 here above. 
Any progress, thoughts an that since the last post?  

Thank you for the very good job!

meatpiHQ, 2023-02-28:
@Daniboum no progress on ISOTP yet. This [library](https://github.com/lishen2/isotp-c) looks very interesting. 

Do you know which is best software to test with?

Daniboum, 2023-03-03:
> @Daniboum no progress on ISOTP yet. This [library](https://github.com/lishen2/isotp-c) looks very interesting.
> 
> Do you know which is best software to test with?

Thanks for the link.

I don't have references of a software to test it. I'm more thinking of a web app able to run UDS services, the firmware dealing with the ISOTP part of it. I see a huge potential in things like your [webcan](http://webcan.meatpi.com/#/home) not depending on the user platform.