meatpiHQ/wican-fw #3: Suggest engage the Home Assistant community to encurage an integration for WiCAN and other CAN Bus OBD-II/OBD2 adapters

Hedda, 2022-08-25:
Not exactly an issue, however, I would like or suggest and recommend that you try to engage with the Home Assistant community to encourage a Home Automation "integration" for WiCAN specifically and a generic integration for CAN Bus OBD-II/OBD2 adapters which supports the same APIs and protocols for third-party integrations.

Check out feature request disacussion -> https://community.home-assistant.io/t/support-for-can-bus-peak-ixxat-canable-etc/249639/

Home Assistant is the world's most popular open-source home automation software application primarily because it makes it relatively easy for developers to make third-party integrations for it.

https://en.wikipedia.org/wiki/Home_Assistant

Home Assistant community members have requested support for different OBD-II/OBD2 adapters for a long time in order to both use it as a presence detection of their cars and trucks (when it connects you to your Wi-Fi at home) as well as get data information about things like battery and fuel level status which is useful in home automation use cases like sending status notifications when the car/truck arrive at home or when battery or fuel is low.

See example:

https://developers.home-assistant.io/docs/development_index/

https://www.home-assistant.io/integrations/device_sun_light_trigger/

https://www.home-assistant.io/integrations/mqtt_room/

Hedda, 2022-08-25:
Here are by the way some somewhat similar sensor integrations for HA even through these do not offer presence detection:

https://www.home-assistant.io/integrations/#car

https://www.home-assistant.io/integrations/torque/

https://www.home-assistant.io/integrations/bmw_connected_drive/

https://www.home-assistant.io/integrations/dacia/

https://www.home-assistant.io/integrations/mazda/

https://www.home-assistant.io/integrations/nissan_leaf/

https://www.home-assistant.io/integrations/subaru/

https://www.home-assistant.io/integrations/volvooncall/

https://brands.home-assistant.io/_/nsw_fuel_station/logo.png

https://www.home-assistant.io/integrations/openevse/

https://www.home-assistant.io/integrations/juicenet/

https://www.home-assistant.io/integrations/velbus/

These other integrations for Home Assistant do offer some type of device tracker and/or presence detection:

https://www.home-assistant.io/integrations/traccar/

https://www.home-assistant.io/integrations/fleetgo/

https://www.home-assistant.io/integrations/ubus/

https://www.home-assistant.io/integrations/mqtt_room/

https://www.home-assistant.io/integrations/device_sun_light_trigger/

https://www.home-assistant.io/integrations/asuswrt/

meatpiHQ, 2022-08-25:
@Hedda I'm interested in this, I'll engage with HASS community to understand the requirements.

Hedda, 2022-08-26:
> @Hedda I'm interested in this, I'll engage with HASS community to understand the requirements.

@meatpiHQ  Cool! I am sure that the Home Assistant have loads of eager early adopters that would be willing to buy your WiCAN and be willing alpha/beta testers once there is a compatible custom integration component available.

Would be awesome if you could either write a new custom integration component yourself or try to find an other interested developer in the Home Assistant community to develop an integration component which can talk to WiCAN over WiFi and/or BLE using one of the protocols and/or APIs that you already support in the existing firmware.

Again, I think what is wanted is to have a new custom integration component that can present several "sensors", "binary sensor", and "device tracker" or "presence detection" as a device with several entities inside Home Assistant to be able to see both state changes of those entries as well as location (at least tracking connection status to Wi-Fi and/or Bluetooth). See:

* https://www.home-assistant.io/integrations/sensor
* https://www.home-assistant.io/integrations/select
* https://www.home-assistant.io/integrations/button
* https://www.home-assistant.io/integrations/switch
* https://www.home-assistant.io/integrations/lock
* https://www.home-assistant.io/integrations/device_tracker
* https://www.home-assistant.io/integrations/person
* https://www.home-assistant.io/integrations/zone
* https://www.home-assistant.io/integrations/proximity
* https://www.home-assistant.io/integrations/locative
* https://www.home-assistant.io/integrations/device_sun_light_trigger

Tip if you are not wanting to continue developing all by yourself is to make it a little more generic "OBD and CAN Bus" custom integration than one specific to only WiCAN, and as such you might want to make either a generic "OBD and CAN Bus" custom integration that can be extended by both yourself and other volunteering developers in the future to support multiple protocols (like BUSMASTER, slcand, SavvyCAN, and RealDash) in one single unified generic "OBD and CAN Bus" custom integration, or just make a generic custom integration for a specific for only one of the protocol and/or API that WiCAN support, like for example a dedicated "BUSMASTER", "slcand", "SavvyCAN", or "RealDash" custom integration specifcially only for that protocol/API working together with the BUSMASTER, slcand, SavvyCAN, or RealDash communities.

RoganDawes, 2022-08-29:
Depending on the interface exposed via WiFi, you might also want to consider looking at the ESPHome project as a firmware environment, which integrates tightly with Home Assistant.

Hedda, 2022-08-29:
> Depending on the interface exposed via WiFi, you might also want to consider looking at the ESPHome project as a firmware environment, which integrates tightly with Home Assistant.

Yes but using an open protocol standard like MQTT or similar would probably make it easier to integrate for developers of other home automation projects/products that are not Home Assistant (like example OpenHAB, Domoticz, Hubitat, Homey, etc. ).

meatpiHQ, 2022-08-29:
@RoganDawes @Hedda I prefer to keep the integration generic, and yes MQTT makes a lot of sense for this application.

I think here we have 2 possible applications.

- **CAN MQTT gateway:** 
we have a sensor or device that communicates over CAN in this case frames will be translated into MQTT frame and published to a topic something like: **wican/xxxxxx/ttt/yyyyyyyy**
**xxxxxx:** MAC address
**yyyyyyyy:** CAN ID
**ttt:** frame type, rtr or data

- **OBD2 protocol :** 
Here the implantation might not be as simple, this is how I think this should be done. First I'll need to add some configuration in the settings page to allow user to select the PIDs that needs to be requested. 
For example the user would select 0x2F or "Fuel tank level input" PID. WiCAN adapter will request this PID while it's awake, lets say every 10 seconds or a configurable time, and will try to publish the value if it's connected to WiFi if not it will store the latest value and it will publish when it connects later. The topic will be **wican/xxxxxx/obd/Fuel_tank_level_input**

Hedda, 2022-08-31:
FYI, if you ever wanted to use a native integration specific to Home Assistant without need to MQTT then there is "Home Assistant WebSocket API" which for example is used by the ESPHome integration and the Z-Wave (Z-Wave) integration in Home Assistant:

https://www.home-assistant.io/integrations/websocket_api/

@meatpiHQ suggest however checking out Home Assistant's existing MQTT-based integrations for different device types and services, so if you could add MQTT client support and publish relevant functions/features as MQTT topics in firmware first then you (or someone else) could probably make a new generic "Car/Truck OBD and CAN Bus" custom integration as a virtual "hub" for Home Assistant that makes use of all different existing MQTT device types and services by subscribing to the correct MQTT topics and present the WiCAN Can Bus adapter as "hub" that present "devices" with multiple "entities" that stream "events" and different support "device types" (example "sensor", "binary sensor", etc.) to Home Assistant:

https://www.home-assistant.io/integrations/#hub

See example existing integrations "Velbus", "Modbus", and "OpenTherm Gateway" integrations which all acts as a "hub" integration (hubs == gateways/bridges with APIs, meaning they can abstract multiple devices which they, in turn, can present to Home Assistant in a uniformed matter to Home Assistant), as well as the "MQTT Alarm Control Panel" which is an "alarm" category integration:

https://www.home-assistant.io/integrations/velbus/
https://www.home-assistant.io/integrations/modbus/
https://www.home-assistant.io/integrations/opentherm_gw/
https://www.home-assistant.io/integrations/manual_mqtt/

Anyway, there are already loads of MQTT integrations which in turn can be used from other integrations as dependencies:

https://www.home-assistant.io/integrations/#search/mqtt
https://www.home-assistant.io/integrations/mqtt/
https://www.home-assistant.io/integrations/mqtt_eventstream/
https://www.home-assistant.io/integrations/mqtt_statestream/
https://www.home-assistant.io/integrations/binary_sensor.mqtt/
https://www.home-assistant.io/integrations/sensor.mqtt/
https://www.home-assistant.io/integrations/device_trigger.mqtt/
https://www.home-assistant.io/integrations/number.mqtt/
https://www.home-assistant.io/integrations/select.mqtt/
https://www.home-assistant.io/integrations/button.mqtt/
https://www.home-assistant.io/integrations/switch.mqtt/
https://www.home-assistant.io/integrations/light.mqtt/
https://www.home-assistant.io/integrations/scene.mqtt/
https://www.home-assistant.io/integrations/lock.mqtt/
https://www.home-assistant.io/integrations/alarm_control_panel.mqtt/
https://www.home-assistant.io/integrations/siren.mqtt/
https://www.home-assistant.io/integrations/fan.mqtt/
https://www.home-assistant.io/integrations/climate.mqtt/
https://www.home-assistant.io/integrations/device_tracker.mqtt/
https://www.home-assistant.io/integrations/mqtt_json/

Also check out other "helper", "utility", and "automation" integrations that they have which can be used in your integration:

https://www.home-assistant.io/integrations/#helper
https://www.home-assistant.io/integrations/#utility
https://www.home-assistant.io/integrations/#automation

PS: Might also be a nice feature to include the export integration diagnostics for troubleshooting the integration for HA, see:

https://www.home-assistant.io/integrations/diagnostics/

meatpiHQ, 2022-10-17:
[MQTT](https://github.com/meatpiHQ/wican-fw#mqtt) is now supported. I also decided against make implementing OBD2 protocol with auto requests.

I will post a working example on how to use the WiCAN MQTT with HA.

meatpiHQ, 2022-11-23:
I've added some instruction on how to use WiCAN with [Home assistant](https://github.com/meatpiHQ/wican-fw#home-assistant).

amuntavant, 2023-03-07:
I just received my first wican! I'm very impressed with this project. I love it!
Thank you very much for all the effort you are making!

Yesterday, I implanted the Wican in my Peugeot Expert Motorhome which is 2 years old. The vehicle has a 4G Router and Homeassistant as its brain. This morning I just received the outside ambient temperature for the first time by MQTT/Node Red (following your instructions). I am very excited!.

And I'm looking forward to see more Canbus information on Homeassistant.
Who knows, maybe one day we can turn the car volume up and down from Home Assistant!

I will closely follow the project and the future integration with Home Assistant.
If you need help, or data from a vehicle with Wican connected 24/7, do not hesitate to contact me!

Thanks!

flyrmyr, 2023-03-21:
@amuntavant I have a similar setup: Converted Promaster Van, 5G router, RPi4 running HAssio. Everything running full time (except the engine..). Got my WiCan, getting data through to my MQTT, setting up reading with NodeRed. But, its my only use for NodeRed, and I feel like it could just be its own integration instead of needing the NodeRed parsing the CanBus output. Have you heard of anyone bypassing that, with HA just reading the bus data directly?

typxxi, 2024-03-28:
any updates about the integration idea cause that was the key to let me buy the stick to get a fully integrated solution in the end independent from Volkswagen cloud.

Would be nice to see such solution happen cause right now I feel a bit lost without a way to get things started.
thx.

Hedda, 2024-09-03:
> any updates about the integration idea cause that was the key to let me buy the stick to get a fully integrated solution in the end independent from Volkswagen cloud.

FYI, because this issue turned out to become more of a general discussion (and as such should be moved to [discussions](https://github.com/meatpiHQ/wican-fw/discussions) instead), I have now posted a new issue as a more formal request asking for an official "WiCAN integration for Home Assistant" from MeatPi:

* https://github.com/meatpiHQ/wican-fw/issues/194