meatpiHQ/wican-fw #4: ETA and status on ELM327 support?

Gamester17, 2022-08-31:
What is your roadmap of development plans and timeline for adding ELM327 support to your WiCAN OBD adapter?

Noticed that ELM327 support was on your feature wishlist but that there was no informtion on development status or estimated date on when ELM327 support will be available? I am sure more potential buyers than me like to know if and when ELM327 support will be available before ordering multiple adapters from your crowdfunding campaign on Crowd Supply:

https://github.com/meatpiHQ/wican-fw/blob/main/README.md#features-wishlist

> #### **_Features Wishlist:_**
> * _Support ELM327 OBD2 protocol_
> * _Support ELM327 protocol and Realdash protocol simultaneously on WiFi and BLE_

PS: As we have two relatively old cars I use my ELM327 WiFi/BLE OBD2 adapters a lot which I have lying around in the glove box in each of our cars all the time in order to read + clear trouble codes from diagnostics logs and turn off the MIL. Thus it would be great to have one unified adapter that did it all, and I would be one for each car and my parent's cars + recommend it to others too if there was one all-in-one inexpensive adapter that could do it all and is compatible with the simple smart phone car diagnostics apps that already support ELM327 over WiFi and BLE.

meatpiHQ, 2022-09-01:
@Gamester17 I can't give you an exact date. I do have a basic working code, but I decide to restructure that part of the code to make it easier to add ELM327 commands and future support for some cool features such as Multiple PID Requests and adaptive timing. Hopefully I can get that ready before the campaign ends, if not I will release a new firmware version and users can update the their units.

You mentioned you had 2 old cars, please note that the hardware is only capable of supporting these protocols:
- ISO 15765-4 CAN (11 bit ID, 500 kbit/s)
- ISO 15765-4 CAN (29 bit ID, 500 kbit/s)
- ISO 15765-4 CAN (11 bit ID, 250 kbit/s)
- ISO 15765-4 CAN (29 bit ID, 250 kbit/s)

meatpiHQ, 2022-09-13:
@Gamester17  ELM327 is now supported on wifi only, ble needs more work. Not all ELM327 commands are yet supported, but enough to work with apps such as Torque, Car Scanner ELM and Realdash.

Hedda, 2022-11-18:
Noticed that you mentioned ELM327 in this update too:

https://www.crowdsupply.com/meatpi-electronics/wican/updates/preparing-for-shipment-and-new-firmware-features

"_I’ve also added support for the OBDII protocol (ELM327) over BLE, and I’ve been using it with RealDash on my car. The performance so far is great, I haven’t had any issues. There are still some improvements to be made with ELM327 emulation, like Multi PID request, which is a bit complex to implement but will greatly improve the responsiveness._"

[image]

Xitro01, 2023-02-15:
Support seems to be available right now, can I just buy a random ELM327 wifi adapter online and use this firmware?
How does the firmware upgrade proces work? Can it be done OTA?

Hopefully someone already tried it here and can give me some pointers.

misel228, 2023-02-15:
> Support seems to be available right now, can I just buy a random ELM327 wifi adapter online and use this firmware? How does the firmware upgrade proces work? Can it be done OTA?

I believe it's meant the other way around. WiCan can emulate an ELM327 so that any other software out there can work with this device, too.

Hedda, 2023-02-15:
> I believe it's meant the other way around. WiCan can emulate an ELM327 so that any other software out there can work with this device, too.

That is correct. The WiCAN firmware is only compatible with ESP32 based SoCs. This is not firmware for none-ESP32 devices.