meatpiHQ/wican-fw #9: [Next-Gen] Look into using ESP32-C6 instead of ESP32-C3 in next-generation hardware?

Hedda, 2023-01-20:
Suggest looking into if possible to replace ESP32-C3 SoC with the new ESP32-C6 in next-generation of WiCAN hardware for the future.

https://www.espressif.com/en/products/socs/esp32-c6

https://www.cnx-software.com/2023/01/12/esp32-c6-wifi-6-ble-802-15-4-module-development-board/

ESP32-C6-WROOM-1 module can now be available on Aliexpress at $14.25(US) for a pack of 5 modules (lower price if buy bulk).

The main reason to use ESP32-C6 instead of ESP32-C3 is that it should be more future proof because it supports more modern interfaces yet it should be backwards compatible while remaining competitive in price so should not raising total cost of the product my much, at least not compared to the added value for that small additional sum. 

- WiFi 6 radio (instead of WiFi 6 radio, though still only 2.4GHz) - Fully backward-compatible.
- Bluetooth 5.0 LE (instead of Bluetooth 4.2 LE) - Should be backward-compatible.
- 802.15.4 IoT radio which supports Zigbee and Thread/OpenThread (inc. Matter over Thread) though not at the same time.
- Dual core? " _It consists of a high-performance (HP) 32-bit RISC-V processor, which can be clocked up to 160 MHz, and a low-power (LP) 32-bit RISC-V processor, which can be clocked up to 20 MHz._"
- 512KB SRAM instead of 400KB SRAM (internal RAM)
- 320KB ROM instead of 384 KB (internal flash storage)?

ESP32-C6-WROOM-1 radio module has ESP32-C6 single-core 32-bit RISC-V clocked up to 160 MHz 320KB ROM, 512KB SRAM, low-power RISC-V core @ up to 20 MHz Wireless – 2.4 GHz WiFi 6, Bluetooth 5.0 LE, 802.15.4 radio for Zigbee 3.0 and Thread. Matter compatible. PCB antenna, 8MB SPI flash.

As an IoT (Internet-of-Things) device it would great if could enable basic capability over Thread via the new Matter standard for IoT:

https://en.wikipedia.org/wiki/Matter_(standard)

Obviously, the initial Matter 1.0 standard is rather limited so mostly just hype now but future versions of the standard do have potential.

https://csa-iot.org/all-solutions/matter/

For example; as of right now version 2.0 of the Matter specification is expected to be published in March or April 2024 so progress is moving relatively fast now in extending the capability to expose new device types (clusters) and more attributes in future versions.

meatpiHQ, 2023-01-29:
@Hedda Thanks for the suggestions.

I like the idea of having WiFi 6 and Zigbee+Matter+Thread without adding much cost. I'll try to get my hands on some ESP32-C6-MINI modules when it's released to try it out.

Hedda, 2023-01-31:
@meatpiHQ Nice! ESP32-C6-MINI-1 module as well as the 8MB SPI Flash version of ESP32-C6-WROOM-1-N4 Engineering Module look to be sold out everywhere, but FYI, can still buy the smaller 4MB SPI Flash version of ESP32-C6-WROOM-1-N4 Engineering Module from a couple places like Adafruit and Mouser for around $3(US) if you act very fast:

https://www.adafruit.com/product/5670

https://eu.mouser.com/ProductDetail/Espressif-Systems/ESP32-C6-WROOM-1-N4?qs=8Wlm6%252BaMh8R1JRAtXlt87w%3D%3D

You can otherwise order "MOQ 3250" in bulk directly from Espressif or possibly even get free samples if you tell them about your products/projects:

https://www.espressif.com/en/contact-us/sales-questions

ESPRESSIF Official Store on Aliextress as well as Digi-Key is completely out of stock of all ESP32-C6 modules and devkit boards.

https://www.aliexpress.com/store/1100220184

PS: ESP32-C6-MINI-1 is by the otherwise the smallest ESP32-C6 based module:

https://www.espressif.com/en/products/modules

### ESP32-C6 Series

32-bit RISC-V MCU & 2.4 GHz Wi-Fi 6 & Bluetooth 5 (LE) & IEEE 802.15.4 

- 32-bit RISC-V single-core processor that operates at up to 160 MHz
- State-of-the-art power and RF performance
- 320 KB ROM, 512 KB SRAM, 16 KB Low-power SRAM on the chip, and works with external flash
- 30 (QFN40) or 22 (QFN32) programmable GPIOs, with support for SPI, UART, I2C, I2S, RMT, TWAI and PWM

Module | Buy | Description | Chip Embedded | Dimensions (mm) | Pins | Flash (MB) | PSRAM (MB) | Antenna | Development Board | Footprint
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
[image] ESP32-C6-MINI-1 |   | ESP32-C6-MINI-1 is a ESP32-C6 based module, which supports Wi-Fi 6 in 2.4 GHz band, Bluetooth 5, Zigbee 3.0 and Thread. It's small-sized and pin-to-pin compatible with the ESP32-C3-MINI series module.  With low power consumption, it is an ideal choice for a variety of IoT devices. | ESP32-C6FH4 | 13.2×16.6×2.4 | 53 | 4 | N/A | PCB antenna | ESP32-C6-DevKitM-1 | N/A
[image] ESP32-C6-WROOM-1 |   | ESP32-C6-WROOM-1 is a ESP32-C6 based module, which supports Wi-Fi 6 in 2.4 GHz band, Bluetooth 5, Zigbee 3.0 and Thread. It's pin-to-pin compatible with the ESP32-WROOM series module. With low power consumption, it is an ideal choice for a variety of IoT devices. | ESP32-C6 | 18×25.5×3.2 | 28 | 4, 8, 16 | N/A | PCB antenna | ESP32-C6-DevKitC-1 | N/A

meatpiHQ, 2023-03-02:
I'm keen to try the C6, I wish it had more IOs and a second core though.

I've already ordered C6-mini modules it seems to be somewhat pin compatible with the C3-mini, I should be able to swap the module and try it out.

https://www.aliexpress.com/item/1005005244950665.html?spm=a2g0o.order_list.order_list_main.5.6170180208ntgl

Hedda, 2023-04-05:
FYI, [01Space](https://github.com/01Space) is sell a tiny ESP32-C6 IoT dev board based on ESP32-C6-WROOM-1 wireless module with two 10-pin headers + USB-C for power:

https://www.aliexpress.us/item/3256805225446433.html

[image]

ESP32-C6 Arduino support will require ESP IDF 5.1 framework which [is still under development:

https://github.com/espressif/arduino-esp32/issues/7713

https://github.com/espressif/arduino-esp32/issues/7852

meatpiHQ, 2023-04-06:
@Hedda I've already received the C6-Mini modules, but I don't have time to work on the firmware now. If you like I can send a modified board if you want to try it out?

Hedda, 2023-08-21:
FYI, Espressif ESP-IDF v5.1 with ESP32-C6 (and ESP32-H2) has now been released and supports most of its functions upstream:

https://github.com/espressif/esp-idf/releases/tag/v5.1

https://github.com/espressif/esp-idf/issues/10423

Arduino ESP32 Core support of ESP-IDF 5.1 (for  ESP32-C6) is however still a work in progress:

https://github.com/espressif/arduino-esp32/issues/7852

https://github.com/espressif/arduino-esp32/issues/7713

https://github.com/orgs/espressif/projects/3

Hedda, 2024-08-30:
@meatpiHQ Will your upcoming WiCAN Pro adapter not be based on ESP32-C6?

* https://www.crowdsupply.com/meatpi-electronics/wican-pro
  * https://www.crowdsupply.com/meatpi-electronics/wican-pro/updates/schematics-and-design-features-highlight

If not when wondering if you at least consider using an ESP32-C6 in your upcoming ESPNetLink LTE/GPS adapter?

* https://www.crowdsupply.com/meatpi-electronics/espnetlink

The reason for the request is that the ESP32-C6 has Wi-Fi 6 + BLE 5 and well as a 802.15.4 radio for Thread or Zigbee SoC

* https://www.espressif.com/en/products/socs/esp32-c6

meatpiHQ, 2024-08-30:
Hey @Hedda 

I really wanted to use the C6, but it did not have what it takes, especially the USB host and the PSRAM.

What I really want to do is use the P4 with the C6. That would be the perfect combination. And I might still release a new version for C6 but without crowd funding just to see how well it performs.

Unfortunately the ESPNetlink is also based on the S3 :)

Btw you should check the WiCAN latest firmware update, now the integration with home assistant is easier than ever.

https://github.com/meatpiHQ/wican-fw?tab=readme-ov-file#home-assistant