meatpiHQ/wican-fw #10: MQTT not sending any requests

mukowman, 2023-02-02:
Recently received the WiCAN module. I was able to get the battery alert setup and publishing to MQTT successfully, but can't seem to get the MQTT module to work by itself. Using the same MQTT settings I cannot see anything published to the MQTT server. Not sure on the best way to check the logs on this device. Any help would be appreciated.

mukowman, 2023-02-02:
Reviewing the code I did notice in the mqtt.c file seems to have some hard coded values in the unit as opposed to the sleep mode pulls them from the config

`void mqtt_init(char* id, uint8_t connected_led, QueueHandle_t *xtx_queue)
{
    esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtt://192.168.31.2",
		.port = 1883,
		.username = "meatpi",
		.password = "meatpi",
		.disable_auto_reconnect = false,
		.reconnect_timeout_ms = 5000,
		.out_buffer_size = 1024*5,
        .buffer_size = 1024*5,
		.lwt_topic = mqtt_status_topic,
		.lwt_msg = "{\"status\": \"offline\"}",
		.keepalive`

meatpiHQ, 2023-02-02:
@mukowman yup looks like that's the problem. I'll release a new version with the fix.

meatpiHQ, 2023-02-02:
@mukowman please try this firmware and let me know.

[wican-fw_v164_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/10569438/wican-fw_v164_hv300.zip)

mukowman, 2023-02-02:
Working perfect now. Thank you for the quick response

spali, 2023-02-03:
Had the same problem, can also confirm the above firmware works and status is now published as expected.

zebraelch, 2023-02-24:
I just downloaded and installed fw 1.64 ..but i forgot i have the wican_usb .. now, not sure this fw is compatible? I do see the wican topic in mqqt-explorer though :)

meatpiHQ, 2023-02-24:
@zebraelch firmware is not fully compatible, v1.64 is already released for USB and OBD

https://github.com/meatpiHQ/wican-fw/releases/tag/v1.64

zebraelch, 2023-02-24:
thank you, all good now :) time to get the home assistant part going.

closed 2023-03-08 by meatpiHQ

karlmikko, 2024-02-26:
I had the same issue - then installed the firmware above - now I get mqtt messages to my broker but carscanner can no longer connect.