meatpiHQ/wican-fw #11: i cant see any canbus messages

zebraelch, 2023-02-24:
Ive been trying for 2 days now to get anything out of my meatpi-usb, I tripple checked the connections, but neither mqqt not the monitor tab or the webcan.meatpi.com spit out anything, neither with engine running or key in pre-starting position. All i found online about Ford canbus says the 500k is correct. What could I be missing?
I use mqqt-explorer for monitoring mqtt, all I get on mqtt wican is the status online/offline. I also tried the obd2 port with a bluedriver and get data from it. 
Any tips for debugging this would be awesome, I imagined the monitor tab should show a stream on canbus data when the truck is running?

meatpiHQ, 2023-02-24:
> I also tried the obd2 port with a bluedriver and get data from it.

So you received data on bluedriver?

zebraelch, 2023-02-24:
Yes I can see live data as soon as engine starts
[image]

meatpiHQ, 2023-02-24:
@zebraelch ok perfect

Try sending this message on mqtt and see if you get a response.

`{ "bus": "0", "type": "tx", "frame": [{ "id": 2015, "dlc": 8, "rtr": false, "extd": false, "data": [2, 1, 70, 170, 170, 170, 170, 170] }] }`

> I imagined the monitor tab should show a stream on canbus data when the truck is running?

Yes you should be able see data in the tab monitor. But some cars don't expose the internal CAN ECU communication on the OBD port. bluedriver uses OBD2 protocol to request data. The above mqtt message requests the ambient temperature.

zebraelch, 2023-02-24:
i have a few tx messages like the above in my mqtt history, but not a single rx.  could i use webcan.meatpi.com to trigger a response?

meatpiHQ, 2023-02-24:
Yes you can, but make sure to set the protocol to slcan.

meatpiHQ, 2023-02-24:
@zebraelch try this:

1- Set mode to AP+Station 
2- select "ELM327" protocol
3- Enable MQTT
4- Install "Car Scanner" APP, or similar app that allows you to connect to using WiFi
5- Setup the connection in "Car Scanner" to connect to WiCAN IP and OBD2 port 3333. (You can find the station IP address in the status page)
6- In mqtt explorer ( I use MQTT spy) subscribe to topic: `wican/+/can/rx` 
7- Now Connect with Car Scanner app

You should see frames coming in on mqtt explorer. Please post a sample of the received messages, it will help with debugging.

[image]

zebraelch, 2023-02-24:
okay more weirdness, i installed car scanner and got it to get the elm-connection but it wouldnt connect to ecu, I happen to have a elm327 bluetooth adapter and tried that one, it connected just fine to elm connection and ecu connection to show all the data.. 

now im wondering if my canbus is weird, does the elm327 dongle use canbus or maybe some of the other pins instead?

zebraelch, 2023-02-25:
okay, it seems the lack of canbus might be the issue :D the elm327 shows SAEJ1850PWM at 41.6 baud as the ecu protocol, is this going over different pins maybe?

zebraelch, 2023-02-25:
"SAE J1850 PWM
Diagnostic bus/protocol used mostly on Ford. Uses pins 1 and 2, communication signal is differential and it's rate is 41.6kB/sec." 
here lies my trouble..  :(  looks like i wont get my data into home assistant..shoot

meatpiHQ, 2023-02-25:
Hang on a second, were you using bluedriver with WiCAN adapter?

zebraelch, 2023-02-25:
no with the bluedriver dongle which i also have, i guess it also can read other pins

zebraelch, 2023-02-25:
just dont get why i have the can pins 6 and 14 populated in my obd outlet if they dont carry data..

meatpiHQ, 2023-02-25:
I always thought it was rare to find a car with no CAN bus. What is your car's make and model, just for reference ?

zebraelch, 2023-02-25:
Its a 2002 Ford E550 - a RV, I was hoping to get the whole canbus thing usable, because all my RV runs on home assistant, so having access to data like indicators, breaking, speed and so on would be useful for automations.. like close the roof vents when driving etc :D

zebraelch, 2023-02-25:
the obd port has only a few pins populated, among them the can ones, so im confused why i dont get anything from them

meatpiHQ, 2023-02-25:
> just dont get why i have the can pins 6 and 14 populated in my obd outlet if they dont carry data..

It's very unlikely that the pins are populated and there's no CAN data or at least OBD protocol. I'll send you  couple of things to try. 

The project is so cool, it has to work.

zebraelch, 2023-02-25:
sweet! Thanks!!

meatpiHQ, 2023-02-25:
Realdash has OBD2 monitor, we can try that. 

Lets disable MQTT for now. Follow the instruction to setup Realdash with OBD2 adapter. 

https://github.com/meatpiHQ/wican-fw#obd2-in-realdash

Make sure to select the OBD2 protocol, we have to try all supported OBD2 CAN protocols.

[image]

Then Click on "OBD2 Monitor" and send the command 0100, if successful you should see some green messages come in. Try this for all supported OBD2 CAN protocols.

[image]

zebraelch, 2023-02-25:
https://user-images.githubusercontent.com/19243711/221329799-a2ae2e6b-1bb8-4a28-ba05-65eb3968bb0e.mp4

Only way it did anything at all was with elm protocol, but with running engine, it seems there is just no data on the canbus

zebraelch, 2023-02-25:
did you mean to try all obd2 protocols available in realdash or wican? i tried wican, maybe i should go through the realdash ones..

zebraelch, 2023-02-25:
I quickly tried all obd2 protcols in realdash with running engine, no data on any of them :(

zebraelch, 2023-02-25:
would it be possible to connect the other 2 pins with the SAEJ1850PWM at 41.6 baud to the wican and test the according protocol in realdash or would this not work as its pwm? just wondering

meatpiHQ, 2023-02-25:
> would it be possible to connect the other 2 pins with the SAEJ1850PWM at 41.6 baud to the wican and test the according protocol in realdash or would this not work as its pwm? just wondering

No that won't work.

Lets try using WiCAN as a sniffer. We have to try all possible CAN bitrates.

Connect your ELM327 together with WiCAN-USB to your car CAN bus, set protocol to "slcan" and enable mqtt. Turn on your MQTT-Explore then use whatever APP you use with ELM327. Change the CAN bitrate in WiCAN configuration page then try the ELM327 App again.

FishTuning, 2023-09-13:
FWIW, in a 07 Dodge Cummins, the CAN bus is there for OBDII communications, but the truck itself does not use it and as far as I know, once the truck starts and switches to the Dodge bus, the only way there will be CAN traffic is if you send requests on the 7E0/7E8 address pair. I forget which address is the scanner and which is the ECM.  Surprised the heck out of me as in a bench ECM from the same truck the CAN bus is active as all get out.