meatpiHQ/wican-fw #12: RC2/RC3 output possible?

JoeBargo, 2023-02-27:
First off, my USB WiCAN and the code you wrote is amazing. 

I'm trying to see if I can send data to an app "RaceChrono" via Wi-Fi. I think it uses a format called RC2/RC3 but I'm not sure how it could be implemented. Here is an explanation from the app developer: https://racechrono.com/forum/discussion/1380/input-format-rc2-and-rc3-used-by-racedac-and-diy-loggers

Would it be possible to do something like RealDash where you define where the data is in an XML doc or something? 

Also, I tried ELM but it's extremely slow compared to the other modes (2000+ vs 40-50).

meatpiHQ, 2023-02-28:
@JoeBargo I need bit more information on this. The format seems simple enough but I see the fields describe analog and digital channels. Do you want send the CAN data bytes in this format?

JoeBargo, 2023-03-01:
> @JoeBargo I need bit more information on this. The format seems simple enough but I see the fields describe analog and digital channels. Do you want send the CAN data bytes in this format? 
> 
> 

So I don't know what the difference is between analog and digital in this regard. Isn't anything through the CAN bus digital? Even if it's originally an analog voltage, we are reading a digital value? I'm essentially trying to get data from my CAN bus (brake pressure, steering angle, etc) into the app for quick data evaluation at the track. 

If you are willing to look into it, I'll send you whatever the app costs. If it's too much work or not possible, don't worry about it.

Also, I have most of the CAN bus mapped for my car, and some of the data starts in the middle of one byte and ends in the middle of another. I've had issues with realdash mapping these but I'm pretty sure that's not an issue with your code. However if you're able to map the bits/bytes for the above format, the ability to do math down to the bit level would be preferred.

meatpiHQ, 2023-03-04:
@JoeBargo 

> I'm essentially trying to get data from my CAN bus (brake pressure, steering angle, etc) into the app for quick data evaluation at the track.

Can you give me an example of how to you send pressure to racechrono? in RC2/RC3 format ?

> If you are willing to look into it, I'll send you whatever the app costs. If it's too much work or not possible, don't worry about it.

I'm always happy to add support for more protocols and apps. Depending on how complex it is, it will be sometime until I get to it.

> However if you're able to map the bits/bytes for the above format, the ability to do math down to the bit level would be preferred.

I have to think of a way to do this efficiently and user-friendly way.