meatpiHQ/wican-fw #13: [SUGGESTION] Offer "ESP Web Tools" (ESP32 web installer and finder) via WiCAN website

Hedda, 2023-02-28:
FYI, "[ESP Web Tools](https://esphome.github.io/esp-web-tools/)" for ESP8266/ESP32 flashing is a free and open source project that is released and maintained by Nabu Casa as custodians of ESPHome and Home Assistant.

https://esphome.github.io/esp-web-tools/

ESP Web Tools allow firmware re-flashing and updating of ESP32/ESP8266 SoC projects via an web installer that prduct developers can easy offer on their homepage, allowing users to get access to the tools directly from their browser without having to install anything.

Suggest @meatpiHQ offer "ESP Web Tools" (ESP32 web installer and finder) via WiCAN website as an option for manually flashing firmware upgrades, alternative alpha/Beta firmware, or firmware recovery.

ESP Web Tools has already been adopted by more and more ESP-based the great projects ever since the introduction of ESP Web Tools, as for example check out the website of these ESP8266/ESP32 based products thjat now offers "ESP Web Tools" (ESP32 firmware flasher) via their website:

* [ESPHome](https://esphome.github.io/esp-web-tools/)
* [Tasmota](https://tasmota.github.io/install/)
* [ESPeasy](https://td-er.nl/ESPEasy/)
* [WLED](https://install.wled.me/)
* [ESPresense](https://espresense.com/firmware).
* [CanAirIO](https://canair.io/installer.html)
* [Squeezelite-ESP32](https://sle118.github.io/squeezelite-esp32-installer/)
* [2Smart](https://2smart.com/docs-resources/platform-updates/platform-updates-13-07-2022)
* [ZigStar](https://install.zig-star.com)

PS: They recently expanded that project to support additional MCU devices that are not based on ESP8266/ESP32, and now they also introduced [SL Web Tools](https://github.com/NabuCasa/sl-web-tools) for Silicon Labs EFR32 based based chips such as EFR32MG21.

* https://github.com/NabuCasa/sl-web-tools

meatpiHQ, 2023-02-28:
@Hedda ESP Web Tools use serial... I'm not sure this is a good idea. The WiCAN-OBD is not really designed to be powered from the its USB port, the power supply is marginal on 5V, you might notice it sometimes might resets. The USB version can be powered from the USB port but RST and DTR pin are not connected to put the ESP in bootloader mode, so that won't work as well.

This will probably be in #9 V2  :)

meatpiHQ, 2023-03-02:
@Hedda 

> Should it not work over USB as long as hardware is using a USB to Serial converter / USB to UART bridge for a virtual COM port?
> 
> Would be user-friendly if was able to flash firmware over just USB alone in order to recover failed/corrupted firmware.
> 
> It would also make it easier for users to temporarily try an Alpha/Beta firmware and then restore stable firmware later.

For the WiCAN-OBD its using the C3 internal USB to serial, the problem is the power supply chip which might cause resets sometimes if powered from the USB connector, but if you [pull down GPIO9 (on the 2x6 connector)](https://github.com/meatpiHQ/wican-fw/edit/main/README.md#2-usb-flash) you can flash it.

> Would it help if the next-gen V2 variant of WiCAN hardware included a USB-to-Serial converter chip / USB-to-UART bridge chip like CH340B chip from WCH, FT231 series chips by FTDI, or CP210x series chips by Silicon Labs / Silabs?
> 
> PS: All those, including CH340B variant, do have an EEPROM that supports adding a unique "Product Description String" for USB.

I think I will keep using the internal USB to serial on V2, unless I find an issue with that, no need to add extra cost. But I'll improve the power supply so that it works with 5V.