meatpiHQ/wican-fw #16: Support custom IP address range

agners, 2023-03-03:
When in AP mode, the device uses 192.168.80.x as address range. It would be helpful to have an option to change the address range used, in case that address range happens to be used :sweat_smile:

meatpiHQ, 2023-03-04:
@agners When you're connected to WiCAN AP your device gets an IP from WiCAN dhcp something like 192.168.80.2. Are you connected to WiCAN AP and another network at the same time?

agners, 2023-03-07:
> Are you connected to WiCAN AP and another network at the same time?

Yes, that is what I would like todo, but I get routing issues because I use the same range. Reconfigure my whole network is a bit painful :cold_face:

meatpiHQ, 2023-03-08:
@agners If you send me the IP range you want I can give you a hardcoded version to get you going, and I'll add this option in the next release.

agners, 2023-03-08:
Thanks for the offer! I can manage for now. An option in the next release is fine.