meatpiHQ/wican-fw #24: [Suggestion] Support for extra wifi settings

mrbostad, 2023-03-12:
Some of us have wifi in car and in garage / home, it would be a nice feature if one could setup multiple wifi configurations so the wican obd could jump over to the house/garage wifi when the car is home and turned off / the in-car-wifi is off.

jnimmo, 2023-04-25:
Yeah I love this idea too

meatpiHQ, 2023-04-26:
@mrbostad @jnimmo Noted

TommySharpNZ, 2023-06-09:
Both great suggestions!

mrbostad, 2023-06-11:
> Both great suggestions! Although possibly we'd also need options to set different MQTT connection settings for each wifi AP?

Good point, i did not think of that since my car-routers have GRE tunnel to my home router, so my mqtt broker is always available. So yeah, maybe a "failover-mqtt-broker" option? (if mqtt1 is not answering, try mqtt2 or something like that?)