meatpiHQ/wican-fw #30: Open Source License

scytacki, 2023-04-12:
I noticed that the license on all of the source files is GPL, while the top level repository LICENSE says public domain. 

I wonder if you'd consider changing all or some of the source files to be MIT? I imagine some of them would be useful to other projects/companies such as the elm327.c file. With an MIT license these other projects could more easily use the file and hopefully contribute changes or at least identify issues.

I just wanted to ask since I prefer contributing to MIT licensed code. That is mainly because I like my efforts to benefit as many people as possible and in my experience haven't found copy left licenses help with this goal. 

I totally understand if you don't want to make the change. I hope to contribute either way. I can imagine you might be concerned about some other company taking the firmware and making a clone of the WiCAN without contributing back.

meatpiHQ, 2023-04-21:
@scytacki I prefer to keep the GPL license, this is an open source project it's only fair if anyone uses this code in their product to keep it opensource.

scytacki, 2023-04-24:
I would note that the current GPL license requires those people using to also open source all of their code too. 
In any case, I think the license file at the top level of the repository should be changed to GPL, so people know the license without having to look at individual source files.

closed 2024-04-30 by scytacki