meatpiHQ/wican-fw #33: Initital CAN bitrate

rv10guy, 2023-04-19:
I am attempting to use the WICAN to interface with an CAN bus running the [rv-c spec](www.rv-c.com). When I plug in the device to my CAN bus, other devices on the bus become unavailable and eventually things on the bus stop communicating. My guess is that this is because the device defaults to the 500K bitrate when it boots and the RV-C standard uses 250K and this mismatch is causing problems on the bus. Is there a way to set a different default bitrate on boot? Is there a reason it is not retaining the previously set rate?

flyrmyr, 2023-04-19:
Can confirm, when I update my bitrate, it forces a Reboot, bringing it back to 500k. So bitrate changes never take effect.

meatpiHQ, 2023-04-20:
@rv10guy @flyrmyr 

Fixed, please try and let me know

[wican-fw_v170.zip](https://github.com/meatpiHQ/wican-fw/files/11284532/wican-fw_v170.zip)

rv10guy, 2023-04-20:
@meatpiHQ 
So it seems to hold the settings now on the settings page but the status is still showing 500K after reboot. So I am not sure what setting is being followed. Also, the monitor page always seems to be default on 500K regardless of what the settings page shows.

meatpiHQ, 2023-04-21:
@rv10guy I've tested and it's working for me, what browser are you using? please share a screenshot of your settings page.

The monitor page always defaults to 500K, that's by design. You're supposed to set the baudrate before you start monitoring and reboot the device after you finish with the monitor tool.

rv10guy, 2023-04-21:
I just reran the tests with both Safari and Chrome. I did a second test where I switched it to 250K and had the same result. Attached are the screenshots. This is the USB version.
[image]
[image]
[image]

meatpiHQ, 2023-04-21:
@rv10guy how about the actual bitrate used? Does the settings reflect the correct bitrate?

rv10guy, 2023-04-21:
@meatpiHQ I am not in a position to test today. All of these screenshots have the device disconnected from the bus. I unplugged the device from my can bus because it was causing communications errors. I can test again in the next few connected to the bus.

meatpiHQ, 2023-04-21:
@rv10guy I've fixed this issue in the latest release [v1.71](https://github.com/meatpiHQ/wican-fw/releases/tag/v1.71)

rv10guy, 2023-04-21:
@meatpiHQ Got it flashed and the interface is working like I would expect. I will get it tested with the bus by EOD tomorrow.

Also quick question, that version had new partition-table, ota_data_initial and bootloader files. Did that need to be flashed over USB or would the update of just the fw file via OTA work? I used USB because I wasn't sure.

rv10guy, 2023-04-23:
This is fixed in 1.71

closed 2023-04-23 by rv10guy