meatpiHQ/wican-fw #34: CAN Bus to MQTT Message flow

rv10guy, 2023-04-19:
It appears from the documentation that the current way the MQTT integration is written, it works on a question/response standard where a request is sent (ie for a temp) and a response comes in. In my application I need to see new messages as they come across the bus. I would like to see the device watch for events that meet a filter and then send those directly to MQTT.

Thanks.

flyrmyr, 2023-04-19:
Perhaps my device is different (OBD model) but I have it sending to my Home Assistant MQTT broker. When listening on the `wican/my_device_id/can/rx` topic, without NodeRED or anything else using MQTT, I'm getting just under 1,000 messages a second.
It would be absolutely amazing if the device could parse out different topics instead of spamming the single topic, or offer any information on how to parse the millions of lines/sec of bus data.

meatpiHQ, 2023-04-20:
@rv10guy as flyrmyr noted, this is the case now, the request/response is just an example for OBD2 protocol.

@flyrmyr I think a better way is to add CAN filter/mask which is more standard for CAN.

rv10guy, 2023-04-21:
Got this tested and it does send all the canbus info over to MQTT.

closed 2023-04-21 by rv10guy