meatpiHQ/wican-fw #36: Share hardware pcb/schenatics

filzek, 2023-04-21:
Hi,

It's possible to share hardware pcb and schematics, bom list. 

So we can learn from and also improve it?

mickeyl, 2023-04-21:
I'd love to see that as well, in particular because I would want to derive an own adapter with some additional goodies to make it work with BMW ENET and CAN FD (which many automakers seem to have settled on)… that said, perhaps @meatpiHQ is interested in a collaboration? I have an alpha release firmware ready (with WiCAN support, btw.) with a custom protocol that transfers data via L2CAP. That way I'm >= 50% faster than an ELM327-based device for ECU reflashing.

meatpiHQ, 2023-04-26:
I haven't had a chance to clean up the schematics yet, been swamped with projects. 

Here's the ESP32-C3 used GPIO's,

CAN RX -> GPIO3
CAN TX -> GPIO0
VBAT ADC -> GPIO4
Blue LED -> GPIO7
Green LED -> GPIO8
Yellow LED -> GPIO9
CAN transceiver Standby -> GPIO6

meatpiHQ, 2023-04-26:
@mickeyl I'm always happy to collaborate, you can contact me https://www.meatpi.com/contact if you like.

> in particular because I would want to derive an own adapter with some additional goodies to make it work with BMW ENET and CAN FD

Sounds cool. Is CAN FD being widely used in new cars? I have an unfinished design for CAN FD adapter, it was never released because I did not think it would be very popular, and the CAN FD controller chip was out of stock.

mickeyl, 2023-04-27:
@meatpiHQ Yes, according to my partners, many "mid to upper class" vendors are in the process of migrating, hence it would be more future-proof to switch to an MCP2518FD or something along the lines. I can't see Espressif coming up with a more modern TWAI-controller in the near future.

Right now I'm fighting with adding a W5500 via SPI to realize the ENET/WiFi forwarder necessary for BMW and some other vendors. The "fun" thing is… for some operations, we will need both CAN and ENET to work with certain transmission control units. In order to make this seamless for the customer, the adapter needs to do both in parallel (that's why I also have a custom L2CAP protocol to free the WiFi for the other task).

brainstorm, 2023-11-21:
As I [mentioned over discord](https://discord.com/channels/1097095962213765220/1097097200967888916/1176377562629423234), I'd also love to have access to those schematics and pcb layout to figure out if there's a 120R on board before I dive into reversing it and adding one myself :-S

meatpiHQ, 2023-11-21:
@brainstorm on WiCAN-OBD there's no 120R termination resistor. Usually if you're plugging in a car you won't need the termination resistor the bus is already terminated. 

WiCAN-USB has an optional termination resistor.

brainstorm, 2023-11-21:
Yeah, I'm not connecting it to a car today... not a typical user setup:

[image]

I wanted the schematics layout to avoid doing an inelegant bodge between CAN_L/CAN_H OBD connector I'll later regret X"D... I thought perhaps there was a more convenient location to put a switch/jumper on the board instead...

So, is it possible to have the schematics or shall I reverse & DIY?

brainstorm, 2023-11-21:
For instance, there's a few of unpopulated pads near the CAN transceiver:

[image]

I'll pull the datasheet for that SOP8 and figure out how it's connected but I'd rather save that effort to do my project 😀

meatpiHQ, 2023-11-21:
@brainstorm should be easy to do. All you need is to solder your jumper and resistor on the obd2 connector pin 6 and 14. I'll send you a photo of a modified board later tonight.

brainstorm, 2023-11-21:
> @brainstorm should be easy to do. All you need is to solder your jumper and resistor on the obd2 connector pin 6 and 14. I'll send you a photo of a modified board later tonight.

Sure, I figured it as well, just wanted to encourage you to release the schematics, but my plan clearly failed. Nice choice of a transceiver IC, btw, I've only used the TJA1040 types ;)

meatpiHQ, 2023-11-21:
@brainstorm it's not encouragement I need just more time :). The schematics needs a clean up, some parts doesn't have any values in the schematics.. resistors not marked as DNP and stuff like that. Embarrassing really :)

ppisa, 2024-11-14:
Please, is there some progress with schematics availability for C3 version? We have lot of experience with CAN, CAN FD, Linux (I am author of CAN bittiming code), RTEMS, NuttX. We use C3 even with iCE40 FPGA etc. My car is broken and I consider to buy diagnostics but I would prefer to have it open and can switch frmware to NuttX based with our CAN drivers etc.

How is it with USB port/C3 pins on the OBD version, are they available? I would like to use ACM UART  and OCD debugging? 

In the longer term, we can help with CAN FD a lot, but I would wait till chips are available.

To see our CAN projects range https://canbus.pages.fel.cvut.cz/ , other projects https://gitlab.fel.cvut.cz/otrees/org/-/wikis/knowbase

We have people working on official Skoda Auto diagnostics tools and bus analyzers at Czech Technical University in Prague. But I have no NDA signed and want to support and use open solutions.