meatpiHQ/wican-fw #37: Reading out battery voltage?

mickeyl, 2023-04-21:
Is it possible with the current hardware design to read the voltage present at the OBD2 port? For ECU reflashing (which can take hours for some vehicles), you typically want to monitor battery closely while operating and stopping when the voltage goes under a certain threshold.

meatpiHQ, 2023-04-23:
@mickeyl What is your hardware version? If it's v3 and above you can read the battery voltage.

I think you might have an older version I know you're one of the first supports of this project :).

mickeyl, 2023-04-25:
I have one v210 and four of the v300 versions.

meatpiHQ, 2023-04-26:
@mickeyl v300 is fine, you can read battery voltage and control the Blue LED.

mickeyl, 2023-04-26:
Excellent, thanks. I will refer to the ADC code in the firmware.

closed 2023-04-26 by mickeyl