meatpiHQ/wican-fw #41: Multiple MQTT Messages in one

rv10guy, 2023-04-24:
I noticed I was losing some MQTT messages in my application. After looking through the messages in detail, I noticed some sometimes multiple CAN bus messages are being sent as a single MQTT message with multiple frame values. 

Here is an example: 

```
“{"bus":"0","type":"rx","ts":32343,"frame":[{"id":436207005,"dlc":8,"rtr":false,"extd":true,"data":[1,255,20,1,0,0,0,0]},{"id":436207005,"dlc":8,"rtr":false,"extd":true,"data":[1,255,20,1,0,0,0,0]},{"id":436207005,"dlc":8,"rtr":false,"extd":true,"data":[1,255,20,1,0,0,0,0]}]}”
```

Is this the expected way that the MQTT function is supposed to work? It is fine if it is. I can account for it in my code, I just wanted to ensure this was the intent.

meatpiHQ, 2023-04-26:
@rv10guy Yes this is the expected behaver. Notice that the "frame" is a list, so you can loop through it.

It's designed this way to overcome the TCP latency. If you send individual message you'll start losing frames.

rv10guy, 2023-04-26:
Perfect, just wanted to make sure it was expected.

closed 2023-04-26 by rv10guy