meatpiHQ/wican-fw #47: Q: CAN Standby PIN 6

mickeyl, 2023-05-18:
As you may know, I'm using the WiCAN as the "prototype" for our own ESP-based CAN-adapter optimized for reprogramming ECUs with a completely custom firmware. Pretty early in the process I noticed from reading your firmware that I need to set GPIO 6 
to LOW to make it work ­– which is what I do in the board-specific setup, i.e. early in the process, but way before initializing the rest of the CAN subsystem (which gets triggered on the first open of the L2CAP connection).

That worked fine in my test scenarios, but when recently working on a car, I noticed that the WiCAN was interfering with the bus when plugged in ­– as long as as it's not properly initialized. It almost looks like it's sending garbage on a wrong bitrate before the TWAI driver is installed & started. Is that possible and if so, why? I could not notice that in my breadboard-based designs. Is that the reason for you adding the GPIO 6? What does it do exactly? Cut the power off the Transceiver?

It would be nice, if you could shed some light on that.

meatpiHQ, 2023-05-18:
@mickeyl I suggest you keep GPIO6 pulled up until you initialize the TWAI driver. Have a look at the datasheet for SN65HVD230DR, GPIO6 is connected to the RS pin of the transceiver. And it has an external pull up resistor, so you can safely initialize the bus then configure this pin as output and pull it low.

I think what's happening here is the the CAN transceiver TX pin is left floating might be interfering with the bus.

The RS pin puts the transceiver in standby mode, which saves power in sleep mode.

Let me know if this works out for you.

mickeyl, 2023-05-18:
Thanks for your quick answer. Doing this kind of works, apart from one little thing: `TWAI_LISTEN_ONLY` will not work with this setup, as apparently TX is still floating then and interfering with the bus. (It's ok for my purposes anyways, so no worries ­– just want to make sure I understand everything… my profession usually is software and with my work on hardware I'm pretty much leaving the comfort zone ;-))