meatpiHQ/wican-fw #48: Hangs after showing a few packets on a Kona EV 2021

brainstorm, 2023-05-20:
With the following settings on the MeatPi OBD web interface and latest firmware flashed `v1.71`:

[image]

I add a new **network** connection to SavvyCAN:

[image]

It connects, shows some packets flowing and then it stops, here's the output from the "Device Console" on SavvyCAN's "Connection Settings":

```
~GVRetSerial()
~GVRetSerial()
TCP Connection to a GVRET device
Created TCP Socket
Connecting to GVRET Device!
Connecting to GVRET Device!
Write to serial -> e7 e7 f1 c f1 6 f1 7 f1 1 f1 9 
Write to serial -> e7 e7 f1 c f1 6 f1 7 f1 1 f1 9 
Got data from serial. Len = 3
Got data from serial. Len = 3
Write to serial -> f1 d 
Write to serial -> f1 d 
f1 0c 01 
f1 0c 01 
Got data from serial. Len = 26
Got data from serial. Len = 26
f1 06 00 20 a1 07 00 00 00 00 00 00 f1 07 6a 02 20 00 00 00 f1 01 8b 3e b1 0a 
f1 06 00 20 a1 07 00 00 00 00 00 00 f1 07 6a 02 20 00 00 00 f1 01 8b 3e b1 0a 
Got data from serial. Len = 17
Got data from serial. Len = 17
f1 0d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f1 0d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Got data from serial. Len = 20
Got data from serial. Len = 20
f1 09 de ad f1 09 de ad f1 09 de ad f1 09 de ad f1 09 de ad 
f1 09 de ad f1 09 de ad f1 09 de ad f1 09 de ad f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Got data from serial. Len = 8
Got data from serial. Len = 8
f1 09 de ad f1 09 de ad 
f1 09 de ad f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Got data from serial. Len = 20
Got data from serial. Len = 20
f1 09 de ad f1 09 de ad f1 09 de ad f1 09 de ad f1 09 de ad 
f1 09 de ad f1 09 de ad f1 09 de ad f1 09 de ad f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Got data from serial. Len = 8
Got data from serial. Len = 8
f1 09 de ad f1 09 de ad 
f1 09 de ad f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Got data from serial. Len = 8
Got data from serial. Len = 8
f1 09 de ad f1 09 de ad 
f1 09 de ad f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Got data from serial. Len = 4
Got data from serial. Len = 4
f1 09 de ad 
f1 09 de ad 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Write to serial -> f1 9 
Comm validation failed.
Comm validation failed.
```

I'm based in Melbourne if you want to see it for yourself, happy to drive wherever you might be and show this in person ;)

brainstorm, 2023-05-20:
After trying different settings I might have found the problem:

1. AP+STA mode: behavior described above.
2. AP only mode: Works!

Could it be that it's trying to wifi scan and trying to connect to the defined STA and it breaks the connection with GVRET/SavvyCAN?

brainstorm, 2023-05-21:
Perhaps duplicate of #49?

meatpiHQ, 2023-05-21:
@brainstorm 

> I'm based in Melbourne if you want to see it for yourself, happy to drive wherever you might be and show this in person ;)

Don't bother I'll take your word for it :)

> After trying different settings I might have found the problem:
> 
> AP+STA mode: behavior described above.
> AP only mode: Works!
> Could it be that it's trying to wifi scan and trying to connect to the defined STA and it breaks the connection with GVRET/SavvyCAN?
> 

Seems there's a bug in the my SavvyCAN implementation.. enabling AP+STA bricks the device. I'm looking into it.

meatpiHQ, 2023-05-21:
@brainstorm Is your device connected to your home WiFi, did it get an IP?

brainstorm, 2023-05-21:
No I wasn't, I was testing in the car, away from home, so it was not connecting to the defined home network.

meatpiHQ, 2023-05-22:
@brainstorm please try this update. 

SavvyCAN accepts UDP broadcast to detect devices or WiCAN, I think the issue was that it keeps broadcasting even when it's connected. This update should fix it.

[wican-fw_v172b_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/11530817/wican-fw_v172b_hv300.zip)

brainstorm, 2023-05-23:
Thanks for the fast response, @meatpiHQ, unfortunately after flashing, verifying the update got applied in the "About" tab, rebooting and re-enabling the AP+sta mode, away from my home network (and in proximity), it fails in the same way: SavvyCam flags a "Comm validation failed" error :/

meatpiHQ, 2023-05-24:
@brainstorm it's working for me. Are you using MAC os?

brainstorm, 2023-05-24:
> @brainstorm it's working for me. Are you using MAC os?

Yes I am, OSX Ventura on an M1.

brainstorm, 2023-11-21:
Fast forward in the future, this is no longer a problem with firmware 2.90, yay! :D

closed 2023-11-21 by brainstorm