meatpiHQ/wican-fw #49: Web interface unstable reachable

terdt, 2023-05-20:
I am starting using WiCAN and I have problems to reach the web interface.
Sometimes it is working, sometimes and most of the times not.
This is independent from the OS and the browser I am using.
I tried Windows, Mac, IOS, Safari, Firefox and Chrome.
WiCAN ist configured in AP+Station mode. 
WiCAN messages are received via MQTT over the station network.. 
I already updated to v1.71 via esptool.
What is wrong and what can I do ?

meatpiHQ, 2023-05-21:
@terdt 

Can you please post a screenshot of your device configuration page?

Are you trying to reach the web interface using the local WiFi?

terdt, 2023-05-21:
Yes, I tried to reach via local WIfi (AP mode). 
Please find attached some photos. 
[image]
[image]

terdt, 2023-05-21:
Yes, I tried to reach the web interface also via local WiFi (AP mode)

[image]
[image]

meatpiHQ, 2023-05-22:
@terdt Please try this new update.

[wican-fw_v172b_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/11530833/wican-fw_v172b_hv300.zip)

terdt, 2023-05-22:
I am at the moment on tour and have no esptool available. I dont will take the risk to update via the web Interface. What do you think, is it better to erase the ESP befördert flashing. I will come back to you.

turricansi, 2023-05-28:
the issue still exists. 

worse than version 1.71. 

after a few minutes neither reachable via webinterface nor via mqtt

meatpiHQ, 2023-05-28:
@turricansi Can you confirm the update was successful ? the About page should show v1.72

Are you testing with a phone? if so try to disable mobile data.

If you're on a computer do you have another active network connection?

turricansi, 2023-05-28:
yes, the about page show 1.72 and I flashed it via the web interface

i tested it on a computer in lan. so no direct wlan connection to the module
i've Unifi U6-Pro Accesspoints.

after i flashed back to 1.71, mqtt works without problems

meatpiHQ, 2023-05-28:
@meatpiHQ I cannot replicate this issue in v1.71, I made a small change on v1.72 thought it might help.

Just to confirm you're connecting to http and not http**S**? 

http://192.168.XX.XX/ and not http**s**://192.168.XX.XX/ 

Can you try to ping the device?

meatpiHQ, 2023-05-28:
@turricansi another thing, are you using the monitor window?

turricansi, 2023-05-29:
@meatpiHQ  

after powering on, 
the webinterface is not accessible  after 1-2 minutes 
and ping is not possible after 5 - 7 minutes 

direct wifi connection to ap is after 7 minutes also not possible

i'm not using monitor windows, i've only powered on the device

meatpiHQ, 2023-05-29:
@turricansi so WiCAN plugged into your car, with sleep mode off right?

turricansi, 2023-05-29:
No i've plugged it into a Heatpump

[image]
[image]

meatpiHQ, 2023-05-30:
@turricansi are you using WiCAN-OBD or WiCAN-USB?

turricansi, 2023-05-30:
WiCAN-OBD

meatpiHQ, 2023-05-30:
@turricansi can you please share some photos of your setup? 

It would be great if you can try it in your car's OBD2 port and see if you still have the same issue.

turricansi, 2023-05-30:
@meatpiHQ same behaviour...

but after disabling mqtt, there are no problems with the webinterface

terdt, 2023-05-31:
@meatpiHQ v1.72 is for ODB ? I am using USB HW version 1.5
but I recognize the same behavior as turricansi. After disabling mqtt, no problems with the webinterface.
How can I enable BLE ? After I enabled BLE by selection ,submitted the changes and reboot it is still disabled

meatpiHQ, 2023-06-01:
@terdt  please keep using v1.71 for now, v1.72 is only a beta, I'll release a new version soon.

> MeatPi is not listed in the WiFi device list of my router but sending regular MQTT messages

How is that possible? if it's not listed then it's not connected, then it wouldn't be able to send MQTT.

> How can I enable BLE ? After I enabled BLE by selection ,submitted the changes and reboot it is still disabled

You have to disable station mode, you can't have both BLE and station mode active at the same time. Also note that after you connect to BLE WiCAN will stop it's AP and stop broadcasting it's SSID. To enable the AP, you have to disconnect BLE(phone or device) and power cycle WiCAN.

The reason for this is that BLE WiFi coexistence or running at the same time is very unstable.

terdt, 2023-06-06:
@meatpiHQ , any update when the new version will be released ?

terdt, 2023-06-09:
@meatpiHQ the blue LED is not flashing when I try to reach the config page. The blue LED lights up continuously.
I flashed the WiCAN with v1.73, the behaviour has not changed. If MQTT is enabled and the WiCAN is connected
to the MQTT server the config page is not reachable. After a reboot you must be fast to disable MQTT to have
access to the config page

meatpiHQ, 2023-06-10:
@terdt OK this issue is specific to the USB version, I've been testing  on OBD that's why I couldn't see it. v1.74 fixes the unstable web interface. But there is a catch, when mqtt is enabled USB-Serial will be disabled for now.

[wican-fw_v174.zip](https://github.com/meatpiHQ/wican-fw/files/11712077/wican-fw_v174.zip)

terdt, 2023-06-10:
@meatpiHQ sorry to tell you, the web interface issue is still there. If MQTT is enabled and the WiCAN is connected
to the MQTT server the config page is not reachable. How is your test environment. My WiCAN is connected to my 
car with 50k and elm327. Has no other user this problem ?

meatpiHQ, 2023-06-12:
@terdt just to confirm, did you check if it took the update?

My test setup is simple, just a WiCAN-USB connected to another CAN Adapter, blasting it with messages. And I got it connected to my local mqtt server, v1.74  works for me and I can refresh with no issues.

When you say elm327 do you mean a different adapter? or WiCAN set as an elm327?

turricansi, 2023-06-12:
@meatpiHQ 

same issues with v1.74 ... i tested it on WiCAN-OBD.

terdt, 2023-06-12:
@meatpiHQ updates was performed according to about page, WiCAN is set as an elm327

meatpiHQ, 2023-06-12:
@terdt I need to see the console log. I can send you a version with debug output enabled, if you're welling to help out? 

All you need to do is boot up while the usb cable is connected and save the console output.

terdt, 2023-06-12:
@meatpiHQ yes, I can do so. I assume that the WiCAN should used with my configuration. Should it be connected to CAN ? Can it be powered external and USB in parallel ?

meatpiHQ, 2023-06-17:
@terdt sorry for the delay.

First try without connecting the CAN bus, and just power from the USB, and test in the same spot where you've installed it just in case its a wifi signal issue, and see if you can replicate the issue. If not then connect CAN and try again.

[wican-fw_v175_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/11777766/wican-fw_v175_hv300.zip)

terdt, 2023-06-17:
@meatpiHQ is the hv300 version not only for ODB WiCAN, I am using the USB version. Is this version with debug output enabled ?

meatpiHQ, 2023-06-17:
@terdt Yes, this one with debug output enabled. OBD and USB it's the exact same thing the only difference is that the USB version enables CAN output on the UART/serial output. In this case we're printing the console messages, so we're using the OBD firmware with log output enabled.

terdt, 2023-06-17:
@meatpiHQ after update I only powered the WiCAN by USB. After enabling MQTT and app. 4-5 minutes later the same problem as before. I also noticed that if MQTT is disabled the WiCAN sending also tx messages every 10 seconds to the broker. How can I print the log ?

meatpiHQ, 2023-06-17:
@terdt Ok interesting! 

Please download putty. When you plug in USB you will see a COM port come up in the device manager, in the the image below it's "COM50". Then open putty select "serial" and fill in the COM port number speed "115200".

To setup the logging see the screenshot below.

**Download:** https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html

**Device Manager:**

[image]

**Logging:**

[image]

**Putty configuration:**

[image]

terdt, 2023-06-17:
@meatpiHQ  please find attached the log, the only things I changed are my passwords

[putty.log](https://github.com/meatpiHQ/wican-fw/files/11779388/putty.log)

meatpiHQ, 2023-06-18:
@terdt Thanks for the logs! 

The log is showing an error "W (267287) wifi:m f null" this could be related to memory allocation, but the strange thing is that I don't see this error on my router.

I also noticed that you have something publishing to the device? are you running home assistant? 

I'll compile a new firmware for you to test. It might be a couple of tries before we can fix this issue since I cannot replicate this in my setup.

terdt, 2023-06-19:
@meatpiHQ yes, I am running home assistant in the same network

turricansi, 2023-06-19:
@meatpiHQ i'am also running home assistant in the same network!

meatpiHQ, 2023-06-30:
@terdt sorry for the delay, please try this firmware. 

This firmware is also has debug messages enabled and printing the available free memory. So this will tell us if its running out of memory.

[wican-fw_v176_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/11916492/wican-fw_v176_hv300.zip)

terdt, 2023-07-10:
@meatpiHQ sorry I have at the moment no time to test, maybe next weekend

terdt, 2023-07-20:
@meatpiHQ please find attached some logs
[putty.log](https://github.com/meatpiHQ/wican-fw/files/12105434/putty.log)
[putty1.log](https://github.com/meatpiHQ/wican-fw/files/12105435/putty1.log)

meatpiHQ, 2023-07-20:
@terdt thanks for that. 

Btw what router are you using? I noticed that WiCAN wifi is going into power saving mode, this is something I've never seen before on all the routers I've tested on

terdt, 2023-07-20:
@meatpiHQ I also noticed the lost of the wifi connection, therefore I started a 2nd log. I am using the new Gl.Inet Spitz AX 5G router in my RV

meatpiHQ, 2023-08-06:
@terdt Please try [v1.79](https://github.com/meatpiHQ/wican-fw/releases/tag/v1.79) I've disabled the power saving mode, let me know if this helps. This one has the log disabled.

turricansi, 2023-08-06:
The issue still exists! After few minutes,  wican not reachable by ping. 
I have unifi Ap.

meatpiHQ, 2023-08-26:
@turricansi @terdt 

Hey guys sorry it took so long, please try the latest version https://github.com/meatpiHQ/wican-fw/releases/tag/v2.10

I migrated to the latest IDF version hopefully this will fix the stability issue.

terdt, 2023-09-12:
@meatpiHQ  @turricansi , I updated to v2.10, problems is solved, other tests have not be done