meatpiHQ/wican-fw #50: WiCAN causing intermittent issue with stability control

edunhill, 2023-06-08:
On plugging WiCAN in on my Land Rover Discovery 4/LR4 with engine off all functions as expected. Having configured via the web interface and started engine I get intermittent errors from the car saying that the stability control in unavailable. Once started theses errors fire constantly (normally such errors can be dismissed but these re fire every few seconds). On stopping engine and removing WiCAN all errors seem to be cleared.

Does the device broadcast on the CANBus in default config? I had assumed by default it would just be passively listening (but that might betray my lack of knowledge on how such things work).

My use case for WiCAN is first and foremost battery monitoring (reporting to mqtt) but I was also hoping to look at other use cases such as logging button presses to MQTT to trigger automations in Home Assistant/Node Red via MQTT.

Thanks!

meatpiHQ, 2023-06-09:
@edunhill This should not happen if you configure the correct CAN bitrate. 

I suggest that set the protocol to "slcan" and  and CAN Mode to "Silent", as see if you still get the same error.

edunhill, 2023-06-09:
Thanks, for coming back to me. 

Protocol was already slcan but changed mode from normal to silent. Car now ignores ignition button press and does not start when WiCAN is plugged in. Starts fine as soon as WiCAN is unplugged. There is also an audible popping noise which comes through the speakers a number of seconds after WiCAN is plugged in with ignition off. 

I have CAN bitrate set to 125k which I believe is correct for this car.

meatpiHQ, 2023-06-09:
@edunhill 

**This is worrying, please don't plug WiCAN into the car again.** Try to find the OBD2 pin out, sometimes it's next to the OBD2 port.

To answer you previous question 

> Does the device broadcast on the CANBus in default config? 

No it doesn't but it sounds like it's causing a conflict on the bus, it possible that it's not compatible for some reason.

edunhill, 2023-06-09:
Have just found this info on the LR4 OBD pinout:

3 - MS CAN -H
4 - GROUND
5 - GROUND
6 - HS CAN -H
8 - TRANSMISSION CONTROL MODULE
9 - ABS MODULE
11 - MS CAN -L
14 - HS CAN -L
16 - BATTERY 12V

This looks like the HS CAN is actually on pins 6/14 which operates at 500k (and is bridged to MS CAN @125k I believe). Do you think having the speed set to 125k on the HS CAN would cause such a fundamental problem?

meatpiHQ, 2023-06-09:
@edunhill

> 500k (and is bridged to MS CAN @125k I believe)

MS CAN and HS CAN cannot be bridged, you can not have 2 different bitrates on the same bus.

> Do you think having the speed set to 125k on the HS CAN would cause such a fundamental problem?

YES definitely, if you're connected to HS CAN you cannot set the bitrate to 125K it has to be 500K(or whatever HSCAN bitrate is). This will cause errors on the bus.

meatpiHQ, 2023-06-09:
@edunhill 

> Apologies - I had read that the busses were 'interconnected by the Instrument pack which acts as a built in gateway between the two busses'. Perhaps the signals are repeated by the instrument pack on to the respective busses at the correct bitrate?

If there is a gateway then yes that is possible, that they're interconnected.

> I have tried reconnecting with the bitrate set to 500k and the car now starts and has not thrown any errors with the engine running (briefly). I also connected to my network and ran Savvy Can and was able to see data flow.

That's really good results. 

> I then enabled MQTT but this caused the WiCAN to power cycle when the engine is running - couldn't connect to WiCAN AP or via my network and could see DHCP requests from WiCAN every 30seconds or so. Any ideas?

What's your firmware version? 

Please try to update to [v1.73](https://github.com/meatpiHQ/wican-fw/files/11706430/wican-fw_v173.zip) this still a beta version, but it fixes a major mqtt bug.

edunhill, 2023-06-09:
Thanks for all your help - will give 1.73 a try this weekend and report back.

edunhill, 2023-06-18:
Apologies for delay coming back. Now on 1.74 which seems to have fixed the MQTT issue.

It would be great if the variables (especially battery voltage and firmware version) available in the UI were also reported to MQTT. At the moment I just have online/offline and the CANbus frames.