meatpiHQ/wican-fw #52: Too many problems!

luiscaloto, 2023-06-12:
Good morning! Las week I received the wifi Wican. It comes with 1.68 firmware. First thing that I did, was to upgrade firmware to 1.71 from web interface. After that, I tried to connect it to canbus but no way....
After a lot of test of configurations... I decided to upgrade firmware following github instructions from usb and this time I am luck.

I manage to work but not as expected.... When I try it with RealDash or openMonitor, Wicam have a lot of delay of send information... Same thing when I try to get data from MQTT..

Some one can help me....????

Thanks in advance

meatpiHQ, 2023-06-12:
@luiscaloto are you using WiCAN OBD? can you please post a screenshot of your device configuration?

> have a lot of delay of send information... Same thing when I try to get data from MQTT..

What do you mean by delay? do you mean send data from realdash or mqtt to CAN bus?

luiscaloto, 2023-06-12:
RealDash or openmonitor received data with too delay. (refresh data is too slow). For example, RPM motor in the car is 1500rpm, and Realdash show this data about 5 or 6 seconds later.
By the other hand, I have a lot of unstability. As people say on issue49, I can´t reach web interface after a few minutes.
About MQTT, it start works correctly, sometimes it stop to respond messages and I need to reset Wicam-OBD

Thanks for the help!

meatpiHQ, 2023-06-12:
@luiscaloto with RealDash you're setting the protocol as elm327? And the BLE disabled right?

For the MQTT please try v1.74, this version has some bug fixes related to MQTT.

https://github.com/meatpiHQ/wican-fw/files/11712077/wican-fw_v174.zip

luiscaloto, 2023-06-12:
Ok!!! As you say, when I am connected to realDash, I configure elm327 and BLE disabled.
I am on work now, this night I will try new firmware version I post the results.

Thanks in advance

meatpiHQ, 2023-06-12:
@luiscaloto Alright, Just few tips.

-  if you set the mode to "AP+Station" use the station IP(local network ip) to communicate with WiCAN do NOT use the 192.168.80.1. 
- If you set the mode to "AP" only then you can use 192.168.80.1 in realdash or other apps.
- In realdash enable "Request only first reply" and manually select the "OBD2 protocol 500Kboud for example". 
- Make sure sleep mode is disabled for now while you test.