meatpiHQ/wican-fw #54: MeatPI reports ~30V instead of ~12V

eludias, 2023-06-18:
The web-interface reports 28V - 31V as the 'Battery Voltage' on WiCAN-OBD in my Chevy Bolt EV instead of the 'normal' 12V.

Reading out the voltages with an external voltage meter shows around ~12V at all pins of the CAN bus.

Is this a software bug and/or EV related?

RichardPar, 2023-06-20:
[image]

Same ..

meatpiHQ, 2023-06-20:
@RichardPar can you please post a screenshot of the about page?

RichardPar, 2023-06-20:
> @RichardPar can you please post a screenshot of the about page?

[image]

meatpiHQ, 2023-06-20:
@RichardPar @eludias It seems you accidently flashed the "WiCAN-USB" firmware? Please try to flash the OBD version.

WiCAN-USB can operate at higher voltage, so it has a different divider value.

[image]

RichardPar, 2023-06-20:
> @RichardPar @eludias It seems you accidently flashed the "WiCAN-USB" firmware? Please try to flash the OBD version.
> 
> WiCAN-USB can operate at higher voltage, so it has a different divider value.
> 
> [image]

Thanks...

Can the Documentation be made clearer on the different ones? or even an image with the link for firmware next to it?

eludias, 2023-06-20:
Yes, I've flashed the wrong image. New OBD image reports correct voltage.

Thanks for the support!