meatpiHQ/wican-fw #58: WiCAN-USB fails to start WiFi AP until CAN disconnnected

jeffjanz, 2023-07-12:
Under certain conditions the WiCAN USB fails to start the WiFI AP during 12v power on, if CAN Hi/Lo are connected

Tested with Firmware v174

- With AP/Station or AP mode on, MQTT off - WiCAN WiFi AP won't start until CAN Hi/Lo disconnected.
Upon CAN disconnection, WiFi AP starts immediately.

- With AP/Station or AP mode on, MQTT _**on**_ - WiCAN WiFi AP is enabled at power up and functional, regardless of CAN Hi/Lo connection.

WiCAN BLE also pairs when configured, but refuses connections regardless of MQTT, WiFi AP configuration

jeffjanz, 2023-08-05:
Just tested with BLE off, in the WiCAN;
WiFi is much more stable than it was last week - intermittent WiCAN AP WiFi dropouts every 60 seconds or so beforehand - now with BLE disabled the WiFi connection from Android phone to WiCAN AP is persistent.

Still the same issue with CAN Hi/Lo disconnection required post-WiCAN startup unfortunately - the WiCAN AP won't start until CAN Hi/Lo is disconnected.

I did notice just now;
Disabling the CAN Stream on the ECU (Link G4+) and writing the ECU changes - then re-enabling the CAN Stream in the ECU and writing changes to ECU again - has the same effect as physically disconnecting the CAN Hi/Lo cable;

as soon as either of these occur (physical disconnection or CAN Stream disable),  the WiCAN AP starts up as normal..

Keep in mind the CAN Cable is permanently connected to the WiCAN and power to the WiCAN is enabled 30sec after vehicle startup - whether that makes a difference to startup logic?

meatpiHQ, 2023-08-05:
@jeffjanz Try v1.77 it should CAN disconnect issue.

[wican-fw-usb_v177_hv150.zip](https://github.com/meatpiHQ/wican-fw/files/12266622/wican-fw-usb_v177_hv150.zip)

jeffjanz, 2023-08-06:
Flashed the following (as bootloader, etc not available in v178 firmware zip)

1. v178 firmware
2. bootloader.bin ~ _from v174 firmware_
3. ota_data_initial.bin ~ _from v174 firmware_
4. partition-table.bin ~ _from v174 firmware_

WiCAN WiFi AP now starts on 12v power on with CAN Hi/Lo connected and persists.

Still issues with BLE connection however, WiCAN unit will pair to Android phone, however won't allow connection.
Pressing "Connect" in Android OS does nothing.
[image]

[image]

WiCAN won't BLE pair with Windows 10 laptop - Windows must time out connection attempt after roughly 30 seconds and displays "Try connecting your device again".

Updated WiCAN BLE password and rebooted unit - no change in behaviour - WiCAN BLE still refuses connections
[image]

meatpiHQ, 2023-08-06:
@jeffjanz Thanks for the update.

Please try to install "nRF connect" APP, make sure you disconnect from WiCAN AP and try to unpair and then pair again using nRF connect app.

What's the model of your Android phone? 

Keep using your Android phone, I've never tested with Windows.

jeffjanz, 2023-08-07:
Flashed v179 as per Discord chat
[image]

Same CAN reset issue on v179 - requires CAN disconnect on powerup to active AP unfortunately.

Initially flashed v179 via USB to test above,
flashed back down to v178 via web GUI - and CAN disconnect/reconnect issue was resolved again.

Attempted twice now to flash up from v178 -> v179 via web GUI - the WiCAN takes the file and reboots - however remains on v178 still.

I've pulled the unit back out of the car to attempt re-flashing v179 via USB again

jeffjanz, 2023-08-07:
> Please try to install "nRF connect" APP, make sure you disconnect from WiCAN AP and try to unpair and then pair again using nRF connect app.
> 
> What's the model of your Android phone?

Done - nRF Connect successfully connects to the WiCAN BLE and shows "BONDED" - it toggles the WiCAN green LED on when connected/bonded

Phone is a Google Pixel 5

meatpiHQ, 2023-08-07:
@jeffjanz 

> Same CAN reset issue on v179 - requires CAN disconnect on powerup to active AP unfortunately.

ops! pretty sure I got it this time.

[wican-fw_v180.zip](https://github.com/meatpiHQ/wican-fw/files/12279085/wican-fw_v180.zip)

> Done - nRF Connect successfully connects to the WiCAN BLE and shows "BONDED" - it toggles the WiCAN green LED on when connected/bonded

Great thanks for the update. The green led should be solid green when it's connected.