meatpiHQ/wican-fw #60: No data using SavvyCAN but ELM327 does work

WiCanIsCool, 2023-07-31:
I have an Audi A3 8Y 2022 and I'm trying to read the CAN bus data.

Yesterday I received my WiCan OBD but I'm having trouble sniffing using savvycan. When I use the ELM327 protocol and connect Car Scanner Android app it can read most of the sensors which tells me the can bus communication actually works. It seems to be at 500000 baudrate.

When I try protocol savvycan and connect from my Linux laptop I simply don't receive any data in the sniffer. In the connection window everything seems alright because I can see serial communication going back and forth.

After googling a bit I found posts about the CAN bus needing a kind of wake up signal. I think savvycan does not do this but Car Scanner Android app seems to do something like that.

I also enabled the monitor in the webui of the WiCan to maybe find the wakeup signal data Car Scanner uses but it's quite difficult to figure out what is what.

Here is some info about the ECUs from Car Scanner.

```
OBD-II
ECU protocol: 6) ISO 15765-4 CAN (11 bit ID, 500 kbaud)
ECU address/CAN Id: 7E8
ECU name:  _
Calibration ID: 05C
In-use Performance Tracking for spark ignition engines: OBD Monitoring Conditions Encountered Counts: 50505

ECU address/CAN Id: 7E9
Calibration ID: 0CW
In-use Performance Tracking for spark ignition engines: OBD Monitoring Conditions Encountered Counts: 64482

```

Also in the Car Scanner app settings I found an advanced setting to override the ECU initialization mode. Which default seems to do the following:

```
ATZ
ATE0
ATH1
ATSP0
ATS0
ATM0
ATAT1
```

meatpiHQ, 2023-08-01:
@WiCanIsCool 

Some cars don't expose their internal CAN bus on the OBD  connector, I think that is the case with your car. When you use car scanner, you're using the OBD2 protocol, which is a request response protocol. If you want to try it on SavvyCAN you need to send the request manually then you'll see the response message. 

Here are some information about OBD2 protocol and PID request: 
https://en.wikipedia.org/wiki/OBD-II_PIDs
https://www.csselectronics.com/pages/obd2-pid-table-on-board-diagnostics-j1979

tylerharvey, 2026-02-22:
Hi, I'm having the same issue with a WiCAN pro and a Hyundai Ioniq 5 with the OBD port. I set communication protocol to SavvyCAN (not mentioned in [docs](https://meatpihq.github.io/wican-fw/savvycan/usage/)?) and now show one connected bus. In the console, I get:
```
Write to serial -> f1 9 
Got data from serial. Len = 4
f1 09 de ad 
Write to serial -> f1 9 
Got data from serial. Len = 4
f1 09 de ad 
```
which is the same as a [lot of people with ESP32 boards have reported](https://github.com/collin80/ESP32RET/issues/16) (and I know WiCAN is ESP-based), possibly due to data on different pins than SavvyCAN expects? I suspect this is a problem with communication between SavvyCAN and WiCAN.
Edited to add: WiCAN pro is on 4.45 firmware.