meatpiHQ/wican-fw #63: Bluetooth Connectivity with RealDash

martinmain, 2023-08-20:
Hi I seem to be having a problem with Bluetooth connectivity between my Samsung Galaxy S8 Android device which is running RealDash and the MeatPi OBD2 device v1.80 firmware v3.00 hardware. I am operating in realdash66 mode to access canbus data and not in elm327 mode. It seems to make initial connection, requests the passcode and I can even see a burst of canbus data in the RealDash CAN Monitor then it asks for the passcode again and again and again. Any ideas what might be going on?

martinmain, 2023-08-20:
> @martinmain before running realdash, please try to Bond the device using "nrf Connect" APP then try realdash again.

Many thanks for prompt reply, I will give this a try.

martinmain, 2023-08-21:
Hi i tried connecting and although I did manage to get it to connect it would not bond despite several attempts, I am attaching the log file from nrf connect
[Log 2023-08-21 18_13_32.txt](https://github.com/meatpiHQ/wican-fw/files/12397089/Log.2023-08-21.18_13_32.txt)

meatpiHQ, 2023-08-21:
@martinmain from the logs I can see it's connected and receiving data. The frames that start with 0x66 are CAN bus frames. Did you try RealDash again?

martinmain, 2023-08-22:
I followed your advice and tried again after using nrf connect and whilst I had some success with parameters updating on RealDash this was short lived and it stopped communicating on BLE again but I had it operating for circa 5 mins perfectly. I have reverted back to AP Station mode, but perhaps should have mentioned earlier WiFi is also dropping out intermittently. I have no problems getting WiFi connectivity but periodically while I am driving (maybe once or sometimes twice in 30 mins) it loses the connection and re-establishes itself. Possible hardware problem?

meatpiHQ, 2023-08-22:
@martinmain are you using realdash on iOS?

martinmain, 2023-08-22:
No it’s a Samsung S8 running Android

martinmain, 2023-08-22:
Yes, I have sleep mode set at 12.8V with car sitting at 13.4V while running

Pics as attached

[image]
[image]
[image]

meatpiHQ, 2023-08-22:
@martinmain try to set it to AP mode instead of AP+Station mode.

martinmain, 2023-08-22:
The realdash device needs internet access via WiFi in this mode it will only be connected to the MeatPi device?

meatpiHQ, 2023-08-22:
@martinmain RealDash can work without Internet connection. This is only temporary it will help to find the issue.

martinmain, 2023-08-22:
Do you think a clean install of the Android device is worth considering, since I assume BLE should work on Android without issue?

meatpiHQ, 2023-08-22:
@martinmain I don't think it will help. But there are some settings in android related to battery optimization might be causing an issue. 

I also have a couple of ideas what might be causing this in the firmware, I'll need to do some testing, I might send you a new firmware to try tomorrow.

Please try and revert back to v1.71 and try again with the BLE.

https://github.com/meatpiHQ/wican-fw/releases/tag/v1.71

martinmain, 2023-08-23:
Many thanks for your support, I will give this a try later today and let you know how I get on

martinmain, 2023-08-24:
I downgraded the firmware to 1.71 as suggested, but unfortunately it is not any better, it connects momentarily before asking for passcode again repeatedly. Although probably not relevant WiFi is no better either 2 km of my 28 km commute were missed due it reconnecting multiple times.

martinmain, 2023-08-26:
Hi loaded v2.00 as per your guidance and pleased to report BLE connectivity seems to be working fine I was getting circa 750 frames / second in RealDash CAN Monitor. I only tested for 10-15 minutes but observed no data dropouts. Assume the battery sleep function is disabled in firmware?

meatpiHQ, 2023-08-26:
@martinmain Sleep is not disabled, but it's not tested yet. I migrated to IDF5 and I think it's broken because there's some changes to ADC api. I'm still working on it, I'll release a beta version tomorrow.

martinmain, 2023-08-26:
Noted and many thanks for your support

meatpiHQ, 2023-08-26:
@martinmain I released the new [v2.10 beta version.](https://github.com/meatpiHQ/wican-fw/releases/tag/v2.10)

martinmain, 2023-08-27:
Pleased to report BLE is rock solid, have not had the opportunity to try WiFi yet. One thing I did notice and not sure if it's the device or RealDash, but when I turn off the ignition, values are held at their last running value I assume until the device goes to sleep. For most this is okay, but RPM sits at idle speed with the engine off.

martinmain, 2023-08-29:
Is there any possibility of getting the 180 second sleep timer available as a user configurable option?

martinmain, 2023-10-06:
@meatpiHQ just wondering if the recent 2,50 revision contains the same BLE and WiFi mods as 2.10beta?

meatpiHQ, 2023-10-07:
@martinmain Yes, the latest version has some additional features and improvements to MQTT.

martinmain, 2023-10-08:
@meatpiHQ installed 2.50 and was unable to get BLE working, so reinstalled 2.10 beta