meatpiHQ/wican-fw #64: Overwhelming the bus with polls?

robwolff3, 2023-08-21:
I have identified a number of PIDs I want to pull into Home Assistant, and to view the data live, maybe 15 PID's or so. Using the NodeRed example in the documentation I was able to pull in the PID's I desired. Then to continuously pull them to constantly have an updated live value in HA I set an inject node to make the request on an interval and tested 1m, 30s, 10s, 5s, 2s. Seems the WiCAN device locks up, stops responding to requests and the web interface stops responding after sending sending a number or requests. And quicker in accordance to a faster polling time. Then I have to unplug the device to reset it.

Is this an appropriate way to use the WiCAN? When connecting to WiCAN from the Android Torque Pro app over WIFI it seems to have no problem with maintaining an updated live value without crashing the device.

meatpiHQ, 2023-08-23:
@robwolff3 can you post your NodeRed workflow?

The max number of PID requests is 8 per second if I remember correctly.

robwolff3, 2023-08-23:
Sure, here is the flow I am using: https://pastebin.com/mh6F4n6C

I am defiantly well under 8 per second.

robwolff3, 2023-08-24:
I tested it 3 more times today just to make sure. With polling set to 5 seconds requesting 1 PID it consistently froze up 6 minutes later. Which looks like give or take 72 requests.

meatpiHQ, 2023-08-26:
@robwolff3 it might be that ECU is going into low power mode, have a look at this issue. Let me know if it works for you.

https://github.com/meatpiHQ/wican-fw/issues/17#issuecomment-1456751079

robwolff3, 2023-08-28:
I dont think this is the case. I tried the test when the car was on and derivable, so I dont think the ECU would be asleep then.

Its pretty obvious when the ECU is asleep, usually when the car is off and after I have left it for 5 minutes. I did try your suggestion of [trying to initialize the bus when the ECU is sleeping](https://github.com/meatpiHQ/wican-fw/issues/17#issuecomment-1458161566). That didnt seem to work for the Chevy Bolt.

meatpiHQ, 2023-08-28:
@robwolff3 can you please post a screenshot of your device configuration page?

Also please update to the [latest version ](https://github.com/meatpiHQ/wican-fw/releases/tag/v2.10)

robwolff3, 2023-08-28:
Im seeing the same behavior on firmware v2.10.

https://i.ibb.co/vqCBQmK/Screenshot-2023-08-28-at-18-19-37-Wi-CAN.png

aapljosh, 2023-09-05:
I can confirm similar activity with my device. I don't know the exact number of polls that eventually crashes the device but it is in the 70-100 range as rowolff3 states. I poll 2 PIDs on the hour with an initialization command to wake up the ECU (fortunately that works for me - Outlander PHEV). It crashes approximately 16-18 hours after resetting on a consistent basis.

meatpiHQ, 2023-09-07:
@aapljosh Can you confirm that you're using the latest firmware version?

Please post a screenshot of you configuration page.

TNTLarsn, 2023-09-08:
Firmware ver: | v1.80
Hardware ver: | v3.00_obd

Only 2 polls / min over mqtt
Cold restart required every couple of hours.

aapljosh, 2023-09-08:
> @aapljosh Can you confirm that you're using the latest firmware version?
> 
> Please post a screenshot of you configuration page.

I am on the latest beta firmware (2.10)

https://photos.app.goo.gl/dEodvAJqAHPJr5UKA
https://photos.app.goo.gl/MM8HvDtxuMUdGtAVA

meatpiHQ, 2023-09-23:
There was a bug that is causing a memory leak in mqtt. Please try v2.30

**OBD:** [wican-fw_v230_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/12706902/wican-fw_v230_hv300.zip)
**USB:** [wican-fw-usb_v230_hv150.zip](https://github.com/meatpiHQ/wican-fw/files/12706903/wican-fw-usb_v230_hv150.zip)

aapljosh, 2023-09-24:
I think the crashing is fixed. I'm looking into another issue now potentially though. My car's AGM 12V battery dropped all the way to 10.9V over the course of 5 hours. It hasn't done that before with this plugged in. I'm concerned the new firmware is keeping my car's ECU on and not letting it sleep. I'll have to investigate further though. 

Also, the status page of the WiCAN always reads "0.0V" under "Battery Voltage:", which is a new issue.

meatpiHQ, 2023-09-24:
@aapljosh the changes in the firmware are only related to MQTT, nothing else has changed since the latest version. But it's possible you're seeing this issue now because the crashing was fixed. 

I'll check the battery voltage issue on my side.

aapljosh, 2023-09-24:
It's possible that I'm seeing it now because of the fixed crashing, for sure. I'll keep trying things out and check in. I only ask the WiCAN for the AGM voltage and the lithium pack percentage once per hour though if I'm not actively charging. I would be surprised if 10 requests over 5 hours killed the AGM battery and kept the ECU on that whole time. 

Unfortunately my car manufacturer wasn't smart enough to have the electric car charge its 12V battery more than once per day, so if I kill it faster than that, my 12V battery dies.