meatpiHQ/wican-fw #66: Filtering CAN Messages for MQTT on the WICAN?

felixstorm, 2023-09-07:
Are there any plans for filtering CAN messages to be sent via MQTT directly on the device?

My use case is just to get the SOC of my BEV while it’s parked and charging in the driveway to use it with a custom PV surplus power charging algorithm based on Home Assistant and ESPHome. Therefore it would be very handy if I could filter the CAN bus frames on the device already to not flood the MQTT server and to keep the receiving end as simple as possible.

Another option would be to - optionally - include the CAN ID in the MQTT topic and then just subscribe to the ones that are of interest. I understand that it takes less resources to put multiple frames in a single message but if you‘re only interested in very specific ones this might be an acceptable trade-off while not having to manage filters on the device.

Thanks for the great product! And buying it also was a really smooth experience - I ordered it last week through Crowd Supply and had a professional looking product delivered to my door step just a few days later without any hassle about paying customs etc.!

Felix

raphael303, 2023-09-07:
This would be crucial for me. Because right now WiCan is working but practically unusable for me. MQTT on homeassistant is completely flooded from the data of to WiCans. CPU usage jumps from 30% to 50% as soon as data comes in. Power usage from homassistant jumps from 4Watt to 7Watt. Just from that.
But worse! NodeRed get erratic and other flows get out of touch with HA and lose data.

Filtering the data before it gets to MQTT is a MUST. tbh, imho it is crazy to flood the mqtt server with 100'000 of thousands of messages from which I only need 16PIDs. This can't be the proper way to do it. It is wasteful.

felixstorm, 2023-09-08:
@meatpiHQ Thanks for your reply.

And just fyi: I might also try to get ESPHome running directly on the WICAN (it should support ESP32-C3 by now) as this might be a more fitting solution for my personal use case. I will post my ESPHome config here in case of success.

raphael303, 2023-09-08:
> @felixstorm @raphael303 Thanks for the feedback. I see the problem but the fix is not simple. I might introduce an option to pack frames into one or more mqtt messages and send every 1 second also removing redundant frames. I still need to think about this... I'll send you a new firmware to test in a couple of days.

> @felixstorm @raphael303 Thanks for the feedback. I see the problem but the fix is not simple. I might introduce an option to pack frames into one or more mqtt messages and send every 1 second also removing redundant frames. I still need to think about this... I'll send you a new firmware to test in a couple of days.

I'm very happy to hear this. I can imagine, that this might me no so simple to to. 
Maybe an approach would be to have a list of PIDs to forward via MQTT configurable via the webinterface of WiCan. So that it only sends the things to MQTT which interest in the first place. I believe with a list of 10-30 PIDs most people would be satisfied.
It would then be also easier, I believe, to forward those user selected PIDs only when their value change.

I'm no coder, but I guess just handling data-reduction on a unknown number of PIDs is much more challenging, then just handling a user selected number of known PIDs.

felixstorm, 2023-09-08:
Short update: I actually managed to get ESPHome working directly on the WiCAN. My example will only poll for the battery SOC once per minute, but since that's all I need, I will probably continue with ESPHome and not follow the native WiCAN MQTT path for now.  
ESPHome example config: https://gist.github.com/felixstorm/18f8ad9fb31ca71bbf62ba32801c34ae

@meatpiHQ Since MQTT filtering seems to be of interest for others I will leave this issue open - feel free to handle it as you like.

raphael303, 2023-09-14:
Thank you for sharing @felixstorm ! Would you know how to do this without polling? Because I couldn't figure out, how to get anything from our EV with requests. But, it sends data while charging, at least...

Because I may try ESPhome on the wiCan too, if I could get it to just process data which is randomly coming in. This week has been difficult in my smarthome. So much erratic behavior from HomeAssistant, I ended up disabling the WiCans because it was just not funny anymore...

felixstorm, 2023-09-15:
@raphael303 If your car sends data automatically while charging then you could just comment out the interval and the script in the ESPHome config to disable the polling. The CANbus `on_frame` sequence should run for any data on the bus, no matter why it was sent.

raphael303, 2023-09-20:
@felixstorm thank you! Now I have to decide, if I'll invest the time to do it your way.
@meatpiHQ any updates on built in filtering? Will it become a reality?

meatpiHQ, 2023-09-20:
@raphael303 still working on it, I've add the filter in the configuration page still need to implement how it retrieves the filter values on load. It's getting there.

meatpiHQ, 2023-09-23:
I've added the filter, its still a bit rough but it works. If you add an ID to the filter all other ID's will be ignored.

@raphael303 curious to know if you still see high power consumption after adding the filter

**OBD:** [wican-fw_v230_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/12706902/wican-fw_v230_hv300.zip)
**USB:** [wican-fw-usb_v230_hv150.zip](https://github.com/meatpiHQ/wican-fw/files/12706903/wican-fw-usb_v230_hv150.zip)

raphael303, 2023-09-24:
Thank you!
Which format is expected in CAN ID?
Hex or dec. I assume hex.
Hex like so:
0374
or
374
or 
0x374
?

meatpiHQ, 2023-09-24:
@raphael303 it's dec.

Yes the filter function is broken in the monitor.

raphael303, 2023-09-24:
Ok I installed and tried. Right now both cars are fully charged, so I can not test it properly, since they only send data while charging. 
However if I plug it in, it sends data for 20 seconds or so. Within those I can see, that the filter works. Nevertheless MQTT still gets a 2000-3000 messages within those 20 seconds.
I filter 5 IDs and those seem to arrive alright. Tomorrow after work I will charge the car and report on power usage. This is certainly better now. Thank you!
But the amount of messages is still very high, much more than practically necessary.

raphael303, 2023-09-25:
Ok, tested it today. Even though considerably less data came into MQTT, it still crashed node red. Even with just one EV. Power use still spiked, just a little less.
It jumped from 3 Watt to 4 Watt, for one EV connected.

[image]

Also noticeable in CPU Temperature.
[image]

So, while an improvement, still not usable.

raphael303, 2023-09-25:
[image]

meatpiHQ, 2023-09-26:
@raphael303 Thanks for the update. I think it makes sense, it's still about 150 MQTT messages per second that's a lot to handle. 

I'm still working on introducing sampling and aggregation method, probably sampling each ID once every 1 second. 

Eventually I want to implement something like the "ultra-delux-way" you mentioned before, a user defined XML just like in Realdash if you're familiar with that. But I'm not sure if the C3 has enough juice to handle this.

BTW are you processing the messages in anyway? if so maybe add a delay to reduce the processing rate.

raphael303, 2023-09-26:
The input goes directly to the function node, which does the processing. I do not know how to do it differently in node red, since the filter node will not work because the incoming messages all look different to it. how the delay would help I don't understand. How would this reduce the amount of messages?
In any case, any node in node red would have to deal with a lot of messages in some way and the mqtt(mosquitto) is working hard in any case also. so for now, I have disabled it

meatpiHQ, 2023-09-26:
@raphael303 you can try to disconnect the function node just to test. I'm thinking it might be the message processing that's causing the extra power consumption. If that's the case then there's a delay node that can be used to limit the message rate. I need to try this once I get my HA going again.

raphael303, 2023-09-26:
I see, the delay has a rate limiter. Ok I tried to set it to one message every 5 seconds, which is still plenty information density for the use case and this helps to have node red not crash immediately. However powerconsumption and temperature still goes up. But not as pronounced as before. Makes sense, since while the function is not triggered anymore so often, lots of other processes before the delay still deal with the data.

meatpiHQ, 2023-09-29:
Alright, I've add a new CAN to JSON interpreter that filters the CAN messages, makes whatever calculation that's specified in the expression and  send will send it to the MQTT broker in JSON format according to the defined time Cycle. 

```
{"VehicleSpeed": 23.000000}
{"EngineSpeed": 1165.000000}
{"CustomID": 3594.400000}
```

Here's some examples how to use it.

```
CAN ID: Frame ID in DEC

Name: is the JSON Key

PID: PID Number in DEC, If the frame is response to a PID request. Otherwise IT HAS TO BE SET TO -1.

Start Bit: The bit were you want to start extracting the value. Bit 0 starts from Byte 1 most significant bit.
If Byte 0 is 0xA0 in Binary b1010 0000. 'Bit 0' = 1, 'Bit 1' = 0 ..... 'Bit 6' = 0, 'Bit 7' = 0

Bit Length: The number of bits of the value 

Expression: The mathematical operations that will be applied to extracted value. Must include the letter V. 

Cycle: Cycle time between JSON messages sent to MQTT. 100 to 10000 in milliseconds.
```

[image]

## **For PID response frames:**

Name|CAN ID| Byte 0 | Byte 1 | Byte 2 | Byte 3 | Byte 4 | Byte 5 | Byte 6 | Byte 7 
---|--- | --- | --- | --- |--- |--- |--- |--- |---
Engine Speed|2024| 4| 65| **12**| _18_| _52_| 170| 170| 170
| |  |  | | PID| Value 1| Value 2  |  |  

The Engine Speed is PID response so "PID" must be set 12. The value we need to extract is 16bit in Byte 3 and Byte 4. So we set the "Start Bit" to 24 and "Bit Length" to 16. The extracted value will be (11*256)+68 = 4660. Now to get the engine speed in rpm we must multiple by 0.25. So the "Expression" will be V*0.25= 1165

Name|CAN ID| Byte 0 | Byte 1 | Byte 2 | Byte 3 | Byte 4 | Byte 5 | Byte 6 | Byte 7 
---|--- | --- | --- | --- |--- |--- |--- |--- |---
Vehicle Speed|2024| 4| 65| **13**| _23_|170| 170| 170| 170
| |  |  | | PID| Value 1|   |  |  

The Vehicle  Speed is PID response so "PID" must be set 13. The value we need to extract is 8bit in Byte 3. So we set the "Start Bit" to 24 and "Bit Length" to 8. The extracted value will be 23. No need for any operation to get the speed so we set the "Expression" to V, so the value will be 23.

## **Any other CAN Frame**
Name|CAN ID| Byte 0 | Byte 1 | Byte 2 | Byte 3 | Byte 4 | Byte 5 | Byte 6 | Byte 7 
---|--- | --- | --- | --- |--- |--- |--- |--- |---
Custom ID|356| 170| 170| _35_| _16_|170| 170| 170| 170

For this Frame the PID needs to set to -1. The value we need to extract is 16bit in Byte 2 and Byte 3. So we set the "Start Bit" to 16 and "Bit Length" to 16. The extracted value will be (35*256)+16 = 8976. Now to get the value we must add 10 to the Value and divide by 2.5. So the "Expression" will be (V+10)/2.5 = 3594.4

## Download

[OBD v2.50](https://github.com/meatpiHQ/wican-fw/files/12765299/wican-fw_v250_hv300.zip)
[USB v2.50](https://github.com/meatpiHQ/wican-fw/files/12765305/wican-fw-usb_v250_hv150.zip)

raphael303, 2023-10-01:
Wow! This looks very promising. I had just entered about 8 lines at once, then hit "submit", it rebooted, but everything was gone. I have now entered only one line, hit submit, and this has been saved.

I will try again later. Only restriction I see right now are values where I need to calculate something like "Byte(1) * Byte (2) /10" And similar calculations.

raphael303, 2023-10-01:
Soo... I tried again. I hit "submit" after every second filter I added.
After the 10 or 12 filter or so I hit submit and after the reboot all filters were deleted.
Now it is all empty. And I will not try again, because entering all formulas already twice is enough for today ;)

meatpiHQ, 2023-10-01:
@raphael303 sounds like a bug.

raphael303, 2023-10-02:
Ok, so I entered all my filters. I still added only 2 at a time. It first worked.
After around 10 or 12 it stopped sending anything.
This is how it looks now. No data is sent to mqtt at all. When I had only about I think 10 data was being sent. I then added another 3 or 4, I can't remember, then it stopped working. I added all the rest, just for good measure, but still, no data.

The ability to edit filters would be super cool, by the way :)

[image]

meatpiHQ, 2023-10-03:
@raphael303 I cannot replicate this issue. Can you please try to delete "Odometer" ? add try. Expression is limited to 16 characters. So instead of doing "B2*65536+B3*256+" you can simply set the Start bit to 16 and length to 24 and use V.

raphael303, 2023-10-03:
It didn't work without Odometer either. I belief it stopped working after ChargerTemp1 or so.
When I'm home, I will start deleting from the end and let you know if it starts working at some point.

raphael303, 2023-10-03:
Interesting. I arrived home, deleted one entry (OutdoorTemp) and it immediately worked. I readded OutdoorTemp, and it still works. 
Also ODOMETER is in fact: B2*65536+B3*256+B4 (longer than 16)
This also delivers the correct value to MQTT (including +B4)

I will now try this on the second WiCan and let you know if it works.

The above filters are, just in case anyone is looking, for the mitsubishi iMiev or any other of the triplets (iOn, c-zero).
The according code in a node red function node looks like so:

```
// Your variables
var SOC1, SOC2, Battery100Cap, CellMaxTemp, CellMinTemp, BatteryVolts, CellMaxVolt, CellMinVolt, ChargeAmpsAC, ChargeAmpsDC, ChargeVoltsAC, ChargeVoltsDC, ChargerTemp1, ChargerTemp2, Odometer, OutdoorTemp;

// Initialize all variables to null to ensure no data is sent if a variable is not updated
SOC1 = SOC2 = Battery100Cap = CellMaxTemp = CellMinTemp = BatteryVolts = CellMaxVolt = CellMinVolt = ChargeAmpsAC = ChargeAmpsDC = ChargeVoltsAC = ChargeVoltsDC = ChargerTemp1 = ChargerTemp2 = Odometer = OutdoorTemp = null;

// Check if the incoming message payload has one of the expected keys
var key = Object.keys(msg.payload)[0]; // Get the key from the incoming message
var value = msg.payload[key]; // Get the value associated with the key

// Assign the incoming value to the corresponding variable
switch (key) {
    case 'SOC1': SOC1 = value; break;
    case 'SOC2': SOC2 = value; break;
    case 'Battery100Cap': Battery100Cap = value; break;
    case 'CellMaxTemp': CellMaxTemp = value; break;
    case 'CellMinTemp': CellMinTemp = value; break;
    case 'BatteryVolts': BatteryVolts = value; break;
    case 'CellMaxVolt': CellMaxVolt = value; break;
    case 'CellMinVolt': CellMinVolt = value; break;
    case 'ChargeAmpsAC': ChargeAmpsAC = value; break;
    case 'ChargeAmpsDC': ChargeAmpsDC = value; break;
    case 'ChargeVoltsAC': ChargeVoltsAC = value; break;
    case 'ChargeVoltsDC': ChargeVoltsDC = value; break;
    case 'ChargerTemp1': ChargerTemp1 = value; break;
    case 'ChargerTemp2': ChargerTemp2 = value; break;
    case 'Odometer': Odometer = value; break;
    case 'OutdoorTemp': OutdoorTemp = value; break;
    // Add more cases as needed
    default: node.warn("Unexpected key: " + key); // Warn if an unexpected key is received
}

// Return the variables as message payloads for the corresponding outputs
return [SOC1, SOC2, Battery100Cap, CellMaxTemp, CellMinTemp, BatteryVolts, CellMaxVolt, CellMinVolt, ChargeAmpsAC, ChargeAmpsDC, ChargeVoltsAC, ChargeVoltsDC, ChargerTemp1, ChargerTemp2, Odometer, OutdoorTemp].map(function (data) {
    return data !== null ? { payload: data } : null;
});

```

raphael303, 2023-10-03:
Second WiiCan works! No problems this time.
However Odometer doesn't work properly on both. I was mistaken. The last + B4 is missing. I'll have to import that one in two steps.

meatpiHQ, 2023-10-05:
@raphael303 I haven't been able to replicate your issue, but I made some changes to the firmware and added a 'store' button to save the filters without pressing submit. However the filters load on the next reboot. I also increase the length of the expressing from 16 to 32.

I also added FTP so you can edit the CAN filter json file. Here's how to connect to FTP:

1- Download Filezilla client https://filezilla-project.org/
2- Run Filezill, Click File->Site Manager
3- Fill in the details, in Host put in WiCAN IP address. NOT 192.168.80.1, that won't work. Password is like the username 'test'
[image]
4- Click on Transfer Settings, and limit the number of connections to 1
[image]
5- Then Click Connect, and click Allow insecure connection.

[wican-fw-usb_v270_hv150.zip](https://github.com/meatpiHQ/wican-fw/files/12817884/wican-fw-usb_v270_hv150.zip)
[wican-fw_v270_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/12817886/wican-fw_v270_hv300.zip)

raphael303, 2023-10-08:
Thank you. This worked!
I could download and edit the json files. Interestingly it didn't work with "transmit" which is my usual FTP-tool. With filezilla it then worked.

Overall your recent adaptations now made make this very very powerful imho. The ability to preProcess CAN data before sending it to MQTT is really awesome!

My final Json for the mitsubishi iMiev:

```
{
    "can_flt": [
        {
            "CANID": 884,
            "Name": "SOC1",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "(B0-10)/2",
            "Cycle": 5000
        },
        {
            "CANID": 883,
            "Name": "BatteryVolts",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "(B4*256+B5)/10",
            "Cycle": 5000
        },
        {
            "CANID": 883,
            "Name": "CellMaxVolt",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "(B0 + 210) / 100",
            "Cycle": 5000
        },
        {
            "CANID": 883,
            "Name": "CellMinVolt",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "(B1 + 210) / 100",
            "Cycle": 5000
        },
        {
            "CANID": 884,
            "Name": "SOC2",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "(B1-10)/2",
            "Cycle": 5000
        },
        {
            "CANID": 884,
            "Name": "CellMaxTemp",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B4-50",
            "Cycle": 5000
        },
        {
            "CANID": 884,
            "Name": "CellMinTemp",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B5-50",
            "Cycle": 5000
        },
        {
            "CANID": 884,
            "Name": "Battery100Cap",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B6/2",
            "Cycle": 2000
        },
        {
            "CANID": 905,
            "Name": "ChargeVoltsDC",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "2*B0+0.5",
            "Cycle": 4000
        },
        {
            "CANID": 905,
            "Name": "ChargeVoltsAC",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B1",
            "Cycle": 4000
        },
        {
            "CANID": 905,
            "Name": "ChargeAmpsDC",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B2/10",
            "Cycle": 4000
        },
        {
            "CANID": 905,
            "Name": "ChargerTemp1",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B3-50",
            "Cycle": 10000
        },
        {
            "CANID": 905,
            "Name": "ChargerTemp2",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B4-50",
            "Cycle": 10000
        },
        {
            "CANID": 905,
            "Name": "ChargeAmpsAC",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B6/10",
            "Cycle": 4000
        },
        {
            "CANID": 1042,
            "Name": "Odometer",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B2*65536+B3*256+B4",
            "Cycle": 10000
        },
        {
            "CANID": 646,
            "Name": "OutdoorTemp",
            "PID": -1,
            "StartBit": 0,
            "BitLength": 8,
            "Expression": "B3-50",
            "Cycle": 10000
        }
        ]}

```

meatpiHQ, 2023-10-08:
@raphael303 thanks for reporting back.

drcvach, 2023-11-12:
@meatpiHQ - Hello, is the FTP connection available also in fw v2.80? I tried to connect via fileZilla as described but got err: ECONNREFUSE. 
Thanks, Petr

meatpiHQ, 2023-12-03:
@drcvach FTP is not enabled in v2.80, please update to v2.91

[wican-fw-usb_v291_hv150.zip](https://github.com/meatpiHQ/wican-fw/files/13539229/wican-fw-usb_v291_hv150.zip)
[wican-fw_v291_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/13539230/wican-fw_v291_hv300.zip)

drcvach, 2023-12-04:
@meatpiHQ - Great, thank you. With new fw it works also from TotalCommander and other FTP clients.

harkoprinssen, 2024-06-21:
> Ok, so I entered all my filters. I still added only 2 at a time. It first worked. After around 10 or 12 it stopped sending anything. This is how it looks now. No data is sent to mqtt at all. When I had only about I think 10 data was being sent. I then added another 3 or 4, I can't remember, then it stopped working. I added all the rest, just for good measure, but still, no data.
> 
> The ability to edit filters would be super cool, by the way :)
> 
> <img alt="image" width="790" src="https://private-user-images.githubusercontent.com/60788818/272029107-d60580aa-4b29-4d36-ac96-bda865d7026c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg5ODczOTEsIm5iZiI6MTcxODk4NzA5MSwicGF0aCI6Ii82MDc4ODgxOC8yNzIwMjkxMDctZDYwNTgwYWEtNGIyOS00ZDM2LWFjOTYtYmRhODY1ZDcwMjZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjIxVDE2MjQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQxMjgzMTJjOTY4MzFkNDQ3ZGYxZWNjMzdlYmUxNWFkM2NhNjFmNWQ5OTJjYjM2ZjViMzI4MTFmMWU2YWJkNTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MphYsCVxAgXxlNw570oQTxp-6ji47bCLCMT4H49ZYjU">

How did you manage to get this into separate sensors in HA?
I've set up an mqtt sensor, but as a 2nd filter has a different name in the json the other ones are not available. they are emptied in HA.

raphael303, 2024-06-21:
With Node-Red.
Basically a function which parses the stream.

[image]

Full Flow is attached.
[flows.json](https://github.com/user-attachments/files/15930413/flows.json)

```
// Your variables
var SOC1, SOC2, Battery100Cap, CellMaxTemp, CellMinTemp, BatteryVolts, CellMaxVolt, CellMinVolt, ChargeAmpsAC, ChargeAmpsDC, ChargeVoltsAC, ChargeVoltsDC, ChargerTemp1, ChargerTemp2, Odometer, OutdoorTemp;

// Initialize all variables to null to ensure no data is sent if a variable is not updated
SOC1 = SOC2 = Battery100Cap = CellMaxTemp = CellMinTemp = BatteryVolts = CellMaxVolt = CellMinVolt = ChargeAmpsAC = ChargeAmpsDC = ChargeVoltsAC = ChargeVoltsDC = ChargerTemp1 = ChargerTemp2 = Odometer = OutdoorTemp = null;

// Check if the incoming message payload has one of the expected keys
var key = Object.keys(msg.payload)[0]; // Get the key from the incoming message
var value = msg.payload[key]; // Get the value associated with the key

// Assign the incoming value to the corresponding variable
switch (key) {
    case 'SOC1': SOC1 = value; break;
    case 'SOC2': SOC2 = value; break;
    case 'Battery100Cap': Battery100Cap = value; break;
    case 'CellMaxTemp': CellMaxTemp = value; break;
    case 'CellMinTemp': CellMinTemp = value; break;
    case 'BatteryVolts': BatteryVolts = value; break;
    case 'CellMaxVolt': CellMaxVolt = value; break;
    case 'CellMinVolt': CellMinVolt = value; break;
    case 'ChargeAmpsAC': ChargeAmpsAC = value; break;
    case 'ChargeAmpsDC': ChargeAmpsDC = value; break;
    case 'ChargeVoltsAC': ChargeVoltsAC = value; break;
    case 'ChargeVoltsDC': ChargeVoltsDC = value; break;
    case 'ChargerTemp1': ChargerTemp1 = value; break;
    case 'ChargerTemp2': ChargerTemp2 = value; break;
    case 'Odometer': Odometer = value; break;
    case 'OutdoorTemp': OutdoorTemp = value; break;
    // Add more cases as needed
    default: node.warn("Unexpected key: " + key); // Warn if an unexpected key is received
}

// Return the variables as message payloads for the corresponding outputs
return [SOC1, SOC2, Battery100Cap, CellMaxTemp, CellMinTemp, BatteryVolts, CellMaxVolt, CellMinVolt, ChargeAmpsAC, ChargeAmpsDC, ChargeVoltsAC, ChargeVoltsDC, ChargerTemp1, ChargerTemp2, Odometer, OutdoorTemp].map(function (data) {
    return data !== null ? { payload: data } : null;
});

```

harkoprinssen, 2024-06-21:
ah, used Node Red :)
Will look into that.

Thanks!