meatpiHQ/wican-fw #68: VSCAN_MSG.Id is wrong when writing to CAN BUS

wasouthpnt, 2023-09-20:
Hello
I am using the mpUSB-CAN V2.0 adapter and coding in C# example TestDrive.cs.
The program uses the vs_can_api_wrapper.cs and the vs_can_api.dll.

The issue is with the CanDevice.Write() which calls the Write() function in the wrapper.
The VSCAN_MSG.Id is set to 0x215 and the bus sees 0x2051.
My guess is the Id nibbles are reversed somewhere in the DLL or the driver.

Thanks for taking my issue
-Wyatt

meatpiHQ, 2023-09-21:
@wasouthpnt  it works for me. I'm not sure which firmware version you got, try to update to the latest version.

https://github.com/meatpiHQ/mpUSB-CAN#flash-new-firmware
[socketcan_vscom_v110.zip](https://github.com/meatpiHQ/wican-fw/files/12686151/socketcan_vscom_v110.zip)

wasouthpnt, 2023-09-22:
I am using version 1.10.2
How are you reading what's on the CAN bus?
Can you monitor the bus with a different program that does not use VSCOM?

My approach was to hook up two mpUSB-CAN devices back to back and transmit / receive for monitoring.
Please let me know your findings.

wasouthpnt, 2023-09-22:
I also used the Can-Analyzer (USB-CAN(V7.20).exe) to monitor CAN messages from the mpUSB-CAN device 
It also confirmed the wrong frame ID was on the BUS, e.g. 0x215 appears as 0x2051
Just FYI

meatpiHQ, 2023-09-22:
@wasouthpnt

Did you try BUSMaster? Technically BUSMaster uses VSCOM, but that's also working for me. 

I tested using PCAN adapter connected to the same bus as mpCAN-USB. And I used C# test code to send/receive CAN frames using mpCAN-USB.

wasouthpnt, 2023-09-23:
I can try BUSMaster. I don't think the issue with mpCAN-USB is receiving CAN frames.  It is the Write function of VSCOM I believe.
Are you receiving CAN frames on the bus with any other hardware that does not use VSCOM? 
Please do not use mpCAN-USB loopback feature.  Send CAN frames out on actual CH CL wires to another device that can monitor the frames.

meatpiHQ, 2023-09-23:
@wasouthpnt  

> Are you receiving CAN frames on the bus with any other hardware that does not use VSCOM?
Yes I'm using PCAN..

I found the problem. 

Try setting the Flags to VSCAN_FLAGS_STANDARD as below instead of extended.

```
  msgs[0].Id = 0x100;
  msgs[0].Size = 2;
  msgs[0].Data = new byte[8];
  msgs[0].Data[0] = 0xde;
  msgs[0].Data[1] = 0xad;
  msgs[0].Flags = VSCAN.VSCAN_FLAGS_STANDARD;
```

wasouthpnt, 2023-09-23:
I think we are getting closer.  I set the Flags to VSCAN.VSCAN_FLAGS_STANDARD.

Now here is what I get.
Sending 0x215  - Received 0x251
Sending 0x308  - Received 0x380
So nibbles still seem swapped.

Thanks for the help.

wasouthpnt, 2023-09-25:
I did try BUSMASTER version 3.2.2 Vscom 1.0
The problem still exists.
Here is what I get sending from BUSMASTER.
Sending 0x215  -  Received on bus 0x251
Sending 0x308  -  Received on bus 0x280

So BUSMASTER has the same issue.

meatpiHQ, 2023-09-25:
@wasouthpnt Ok I'm able to replicate the issue. I'll send you a fix soon.

meatpiHQ, 2023-09-26:
@wasouthpnt Please flash the new firmware, this should fix the issue.

[socketcan_vscom_v120.zip](https://github.com/meatpiHQ/wican-fw/files/12727968/socketcan_vscom_v120.zip)

wasouthpnt, 2023-09-26:
The issue is fixed with your firmware **socketcan_vscom_v120**! 
Thanks for your excellent support Ali.

closed 2023-09-27 by wasouthpnt