meatpiHQ/wican-fw #70: Filtering for mpCAN-USB not working

wasouthpnt, 2023-10-13:
Hi Ali,
I use .Net C# and here is my code that fails at the designated line below.
As you know the C# code uses the wrapper file: vs_can_api_wrapper.cs

                   // set one filters
                    VSCAN_FILTER[] filter = new VSCAN_FILTER[1];

                    filter[0].Size = 1;
                    filter[0].Id = 0x200;
                    filter[0].Mask = 0xff;
                    filter[0].Extended = 0;
                    
                    CanDevice.SetFilterMode(VSCAN.VSCAN_FILTER_MODE_SINGLE);
                    CanDevice.SetFilter(filter);  <==== This line fails

Can you assist?
Thanks,
Wyatt

meatpiHQ, 2023-10-15:
Hey @wasouthpnt yes sure. Filters are not yet implemented.. I will send you a firmware update soon.

meatpiHQ, 2023-10-17:
@wasouthpnt Please try this new firmware.

[socketcan_vscom_v130.zip](https://github.com/meatpiHQ/wican-fw/files/12928009/socketcan_vscom_v130.zip)

wasouthpnt, 2023-10-19:
Thanks for the new firmware, 
I loaded the new firmware.
Now the the code for setting a filter does not crash.  Thank you.
However, I expect that setting a filter for 0x200 like shown below,  then reading the bus with the command 
CanDevice.Read(ref Readmsgs, 1, ref Read), 
would only return 0x200 on the bus.  But instead all commands are returned on the bus.  Am I misunderstanding something?

                    // set one filters
                    VSCAN_FILTER[] filter = new VSCAN_FILTER[1];

                    filter[0].Size = 2;
                    filter[0].Id = 0x200;
                    filter[0].Mask = 0xff;
                    filter[0].Extended = 0;

                    CanDevice.SetFilterMode(VSCAN.VSCAN_FILTER_MODE_SINGLE);
                    CanDevice.SetFilter(filter);

meatpiHQ, 2023-10-19:
@wasouthpnt here's the how to set the filter, its very impotent to set your filter size correctly otherwise you might send a mask 0x00000000 which will allow undesired messages. If you want 2 filters the size should be 2 : VSCAN_FILTER[2]

```
//This will allow only 0x200 and 0x12345678
                VSCAN_FILTER[] filter = new VSCAN_FILTER[2];
                filter[0].Size = 2;
                filter[0].Id = 0x200;
                filter[0].Mask = 0xFFFFFFFF;
                filter[0].Extended = 0;
                
                filter[1].Size = 2;
                filter[1].Id = 0x12345678;
                filter[1].Mask = 0xFFFFFFFF;
                filter[1].Extended = 1;

                CanDevice.SetFilterMode(VSCAN.VSCAN_FILTER_MODE_SINGLE);
                CanDevice.SetFilter(filter);
```

wasouthpnt, 2023-10-19:
My Mistake.
I changed the Mask to 0xFFFF and the filtering works as expected.  Thank you!

meatpiHQ, 2023-10-19:
@wasouthpnt perfect!

wasouthpnt, 2023-10-19:
Please bear with me. When I set the filter and for 0x200., CanDevice.Read(ref Readmsgs, 1, ref Read) does filter IDs 0x200.
But now the issue is for all other IDs CanDevice.Read(ref Readmsgs, 1, ref Read), still returns 0x200.
It appears that after the filter is applied to the read it does not clear the read buffer, thus all other commands found that are not  ID 0x200, CanDevice.Read(ref Readmsgs, 1, ref Read) still returns ID 0x200.  I hope that makes sense?

meatpiHQ, 2023-10-19:
@wasouthpnt I'm not sure if I understand what you mean. It would be easier if you can send me your code so I can replicate the issue.

wasouthpnt, 2023-10-19:
// set one filters
                    VSCAN_FILTER[] filter = new VSCAN_FILTER[1];

                    filter[0].Size = 1;
                    filter[0].Id = 0x200;
                    filter[0].Mask = 0xFFFF;
                    filter[0].Extended = 0;

                    CanDevice.SetFilterMode(VSCAN.VSCAN_FILTER_MODE_DUAL);
                    CanDevice.SetFilter(filter);

                 while (bContinue == true)
                 {      
                        // read CAN frames
                        CanDevice.Read(ref Readmsgs, 1, ref Read); // **This line continues to return ID 0x200 even other CMDS on bus**

                        for (int i = 0; i < Read; i++)
                        {
                            VSCAN_MSG m = new VSCAN_MSG();
                            m.Id = Readmsgs[i].Id;
                            m.Size = Readmsgs[i].Size;
                            m.Data = new byte[m.Size];
                            m.TimeStamp = Readmsgs[i].TimeStamp;

                            for (int j = 0; j < Readmsgs[i].Size; j++)
                            {
                                m.Data[j] = Readmsgs[i].Data[j];

                            }
                            UpdateLog(m, LogMsgType.Incoming);
                     }
              }      

I expect **ref Read** to be 0 until the next 0x200 appears on the bus.  Am I correct?
I hope this helps, thank you!

meatpiHQ, 2023-10-20:
@wasouthpnt Alright, i fixed the issue. Please try the new firmware.

 
[socketcan_vscom_v140.zip](https://github.com/meatpiHQ/wican-fw/files/13053908/socketcan_vscom_v140.zip)

wasouthpnt, 2023-10-20:
Excellent!  New firmware works! 
I set up 2 filters like below and CanDevice.Read(ref Readmsgs, 1, ref Read) filters correctly.  Thank you!

                    // set filters
                    VSCAN_FILTER[] filter = new VSCAN_FILTER[2];

                    filter[0].Size = 2;
                    filter[0].Id = 0x200;
                    filter[0].Mask = 0xFFFF;
                    filter[0].Extended = 0;

                    filter[1].Size = 2;
                    filter[1].Id = 0x215;
                    filter[1].Mask = 0xFFFF;
                    filter[1].Extended = 0;

                    CanDevice.SetFilterMode(VSCAN.VSCAN_FILTER_MODE_SINGLE);
                    CanDevice.SetFilter(filter);

closed 2023-10-29 by wasouthpnt