meatpiHQ/wican-fw #74: WiCan connecting to undesired AP

puni2k, 2023-11-07:
Hi,

i use my WiCan device to monitor the SOC of my EV during charging.
I have a meshed wifi with multiple APs all with the same SSID. Since the wifi signal in my garage normally isn't very good I tried to improve it with a repeater.

'Normal' devices are working fine with this setup but I found that the WiCan device will almost never connect to the near (strong signal) AP but will more likely connect to a far (weak signal) AP which result in a rather unstable connection.

This seems to be a common problem with many ESP32 Wifi implementations and I found several suggestions on how to improve this.
- AP Sorting by Signal https://www.esp32.com/viewtopic.php?t=18979#p101541
- WifiMulti https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiMulti/WiFiMulti.ino
- Support for 802.11k / v https://github.com/espressif/esp-idf/blob/b4268c874a4/examples/wifi/roaming/README.md

I'd appreciate it very much if the wifi code could somehow be improved to work better with my environment.

meatpiHQ, 2023-11-10:
@puni2k Thanks. I will look into this.

puni2k, 2023-11-10:
@meatpiHQ 
Great timing :) , just managed to get a working build environment and build a firmware binary myself with WIFI_ALL_CHANNEL_SCAN, WIFI_CONNECT_AP_BY_SIGNAL and 802.11 k / v support.
From a first test it seemed to work fine.

I can create a pull request if you want.

meatpiHQ, 2023-11-10:
@puni2k Yes please that would be great!

puni2k, 2023-11-10:
opened #76

closed 2023-11-10 by puni2k