meatpiHQ/wican-fw #77: 2.80 fw update failing

badgerwithagun, 2023-11-11:
Hi, just got my WICAN and trying to update to the latest firmware. Mine arrived with v1.69.

I tried upgrading to v2.80 OTA, but this just caused the device to reset to factory settings without changing the version number.

Not wanting to go straight to the USB option, I tried upgrading to v1.71 OTA. That worked fine. I then carried on all the way up to v2.50 OTA with no problems.  None of these upgrades caused a factory reset.

Then I tried the upgrade from v2.50 to v2.80 and got the same problem as originally.  The upgrade seemed to work OK, but the device was reset to factory settings and it still shows v2.50.

Any suggestions (other than the USB option - I want to avoid it if I can, since the car isn't nearby)!

meatpiHQ, 2023-11-12:
@badgerwithagun I'm aware that there's an issue with the OTA, I'm still trying to find a solution. Please see the comment below.

https://github.com/meatpiHQ/wican-fw/discussions/71#discussioncomment-7451940

badgerwithagun, 2023-11-12:
Thanks @meatpiHQ

meatpiHQ, 2023-11-15:
@badgerwithagun Please try v2.90. After the OTA is complete the page will refresh and goes back to status page, you'll need to refresh again and check the about page if it was successful.

[wican-fw-usb_v290_hv150.zip](https://github.com/meatpiHQ/wican-fw/files/13365760/wican-fw-usb_v290_hv150.zip)
[wican-fw_v290_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/13365761/wican-fw_v290_hv300.zip)

brainstorm, 2023-11-21:
@meatpiHQ This updated failed for me over cable as well anyway (tried OTA and didn't work either) 🤷🏻 

I apparently have a v1.00 WiCAN dongle and I was trying to update to `wican-fw_v290_hv300.bin` (I saw it on https://github.com/meatpiHQ/wican-fw/issues/77#issuecomment-1812614826) via USB port instead of OTA:

Seems like it went alright:

```
% esptool.py -c esp32c3 -p /dev/cu.usbmodem1244301 --baud 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader.bin 0x10000 wican-fw_v290_hv300.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin
esptool.py v4.6.2
Serial port /dev/cu.usbmodem1244301
Connecting....
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 10:91:a8:17:ae:2c
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00171fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Erasing flash...
Took 0.23s to erase flash block
Wrote 22528 bytes at 0x00000000 in 0.5 seconds (342.0 kbit/s)...
Hash of data verified.
Erasing flash...
Took 6.68s to erase flash block
Wrote 1446912 bytes at 0x00010000 in 34.0 seconds (340.8 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.05s to erase flash block
Wrote 3072 bytes at 0x00008000 in 0.1 seconds (361.9 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.08s to erase flash block
Wrote 8192 bytes at 0x0000d000 in 0.2 seconds (366.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

The example Readme.txt is quite wrong in a few ways and doesn't work with today's esptool syntax nor firmware paths, btw:

```
esptool.py esp32c3 -p COMx -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 wican-fw_vxxx_hv300.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin
```

But when re-connecting to the dongle, it still states 1.00?:

[image]

meatpiHQ, 2023-11-21:
@brainstorm wow! Never seen anything like this before.    There's no v1.00. Which browser are you using?

brainstorm, 2023-11-21:
Hey, sorry for deleting the post, I'm using firefox with uBlock Lite (AdBlocker) and after you mentioned that I disabled it and it now shows 2.90 and a different Hardware ver!:

[image]

Anyway, reposting it here since I thought I was doing sth wrong because I didn't short the two vias you show in the docs, but same result... although now we know it was a browser issue (sorry to hijack this issue I'll now try OTA updates and report back when there's a newer version)!:

Address   File 
0x0       bootloader.bin 
0x10000   wican-fw_vxxx_hv300.bin 
0x8000    partition-table.bin 
0xd000    ota_data_initial.bin

Using esptool:
esptool.py esp32c3 -p COMx -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 wican-fw_vxxx_hv300.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin

@meatpiHQ This updated failed for me over cable as well anyway (tried OTA and didn't work either) 🤷🏻 

I apparently have a v1.00 WiCAN dongle and I was trying to update to `wican-fw_v290_hv300.bin` (I saw it on https://github.com/meatpiHQ/wican-fw/issues/77#issuecomment-1812614826) via USB port instead of OTA:

Seems like it went alright:

```
% esptool.py -c esp32c3 -p /dev/cu.usbmodem1244301 --baud 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader.bin 0x10000 wican-fw_v290_hv300.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin
esptool.py v4.6.2
Serial port /dev/cu.usbmodem1244301
Connecting....
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 10:91:a8:17:ae:2c
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00171fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Erasing flash...
Took 0.23s to erase flash block
Wrote 22528 bytes at 0x00000000 in 0.5 seconds (342.0 kbit/s)...
Hash of data verified.
Erasing flash...
Took 6.68s to erase flash block
Wrote 1446912 bytes at 0x00010000 in 34.0 seconds (340.8 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.05s to erase flash block
Wrote 3072 bytes at 0x00008000 in 0.1 seconds (361.9 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.08s to erase flash block
Wrote 8192 bytes at 0x0000d000 in 0.2 seconds (366.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

The example Readme.txt is quite wrong in a few ways and doesn't work with today's esptool syntax nor firmware paths, btw:

```
esptool.py esp32c3 -p COMx -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 wican-fw_vxxx_hv300.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin
```

But when re-connecting to the dongle, it still states 1.00?:

[image]

badgerwithagun, 2023-11-26:
@meatpiHQ - 2.90 upgrade worked fine OTA 👍🏻. Thanks.

I'll leave this open for @brainstorm

brainstorm, 2023-11-26:
All good on my side, feel free to close!

closed 2023-11-28 by badgerwithagun