meatpiHQ/wican-fw #79: Hard reset and/or debug - how to?

pcnetzwerke, 2023-11-19:
Hi,

I changed the sleep voltage of my v2.50 WiCAN-OBD2 from 13.1 to 13.0.
The device restarted and did not connect to WLAN any more.

So I connected to the WICAN AP, but could not reach the webserver. Trying to ping 192.168.80.1. was not successful either.

So I think I somehow bricked my device.

I connected via USB and flashed the v2.90 as described. But still the connection kept impossible, trying both: my AP and the WICAN-AP.

I would like to debug and maybe hardreset the device but could not find any description.
Could You please describe or tell me where to find it?

Thanks in advance,
Martin

meatpiHQ, 2023-11-21:
@pcnetzwerke did you have BLE enabled? Is the blue led flashing? Changing the sleep voltage should not break things. 

To hardreset please flash the firmware using the USB interface, make sure you erase the flash first.

https://github.com/meatpiHQ/wican-fw#2-usb-flash

meatpiHQ, 2023-11-22:
@pcnetzwerke yes of course, for WiCAN-OBD the USB connector is connected to the ESP32-C3 USB interface. For WiCAN-USB it has a USB to serial chip. To print out the debug logs you have to rebuild the project and enable output.

One thing to note that WiCAN-OBD needs to powered by a 12V supply through its OBD2 connector if you power it up from the USB connector only it might not work properly. WiCAN-USB doesn't have this issue.

meatpiHQ, 2024-11-27:
@cvalentine99 are you using WiCAN-OBD? is it only powered from USB?