meatpiHQ/wican-fw #83: Longer CAN-Bus Messages than 8 Byte on MQTT

ggtimtom, 2024-01-10:
Hello,

is it possible to get data from the canbus which is longer than 8 Bytes over MQTT?

Regards, c

ggtimtom, 2024-01-21:
I want to get the following date from the VW E-Up:

22 02 BD -->
aa = central aging counter
(bb*2^16+cc*2^8+dd) = Odometer [km]
ee = reserved

so i send the following...
TX:
{ "bus": "0", "type": "tx", "frame": [{ "id": 2021, "dlc": 8, "rtr": false, "extd": false, "data": [3, 34, 2, 189, 170, 170, 170, 170] }] };

and get back this...
RX:
{"bus":"0","type":"rx","ts":21635,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[16,13,98,2,189,169,0,28]}]}

i think in data 169 = aa, 0 = bb, 28 = cc, dd and ee is missing.

Is there a way to extend the messages?

meatpiHQ, 2024-01-21:
> Hello,
> 
> is it possible to get data from the canbus which is longer than 8 Bytes over MQTT?
> 
> Regards, c

Hey @ggtimtom, each CAN frame can only be 8 bytes, so the mqtt will only have 8 bytes max.

To address your issue, try sending a frame with data bytes: "data":[48,0,10,170,170,170,170,170] after receiving the first Rx message, this is a flow control message telling the ECU that your ready to receive multiple frames.

Here are some useful links, if have an app that reads the this parameter you can log the request and response, see the wiki link below.

The discussion link, is someone also trying to handle multiple frame response.

https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

https://github.com/meatpiHQ/wican-fw/discussions/75

Let me know if it works for you

ggtimtom, 2024-01-21:
Hello @meatpiHQ ,

i sent the dataarray as recommended an got back

[33,240,0,0,55,174,172,240] 
the first 240 seem to be the missing byte. I drove two kilometers and it show the correct data.

Thank you!