meatpiHQ/wican-fw #88: Custom MQTT topics

MarcosAPMT, 2024-02-09:
Hi, I have been testing the WiCAN-OBD2 and I have seen that it is not possible to manually configure the MQTT topic to which the device publishes. My idea of use is for fleets of vehicles managed by third parties and not being able to configure the topics makes this implementation very difficult for several reasons. Would it be possible to add this feature? Thank you.

MarcosAPMT, 2024-02-09:
Yes, I am aware of this limitation. In principle this is not a major constraint for us, at least for some tests. The equipment is located within a closed, secured network. 

As I said, being able to configure the topics is very important for us.

I appreciate your help and quick response. Best regards.

meatpiHQ, 2024-02-09:
Now there's only one receive and transmit topics, do you want to make those configurable?

MarcosAPMT, 2024-02-09:
Yes, and also the status topic. Right now, both CAN and status topics are located in wican/xxxxxxxxxxxx. My request is to at least be able to change the xxxxxxxxxxxxxxx part to an arbitrary value. This would make it easier for us to know from which vehicle we are receiving data.

meatpiHQ, 2024-02-09:
xxxxxxxxxxxx is the device ID and it unique, not a random value.

MarcosAPMT, 2024-02-09:
Oh, I know this. But as I mentioned, for our use case, these devices are operated by third-party workers. Having them take note of the ID of the device installed in each vehicle and send it to us is rather unreliable. It is important for us to know exactly which vehicle in the fleet a device is installed in, and as I said, if the topics could be configured, this would be more than enough for us.

meatpiHQ, 2024-02-12:
@MarcosAPMT I'll add this feature soon.

meatpiHQ, 2024-02-29:
@MarcosAPMT Please try this firmware, all 3 topics are now configurable. 

[image]

[wican-fw_v297_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/14448508/wican-fw_v297_hv300.zip)

MarcosAPMT, 2024-02-29:
@meatpiHQ Thanks a lot. Will try this tomorrow and revert.

MarcosAPMT, 2024-03-04:
@meatpiHQ Hello. I've been able to test and it's working well. Looks like the CAN message filtering is different is this version, so the old config didn't work for me.

meatpiHQ, 2024-03-05:
@MarcosAPMT set the index in the filter to 2. You'll get the same behaviour as before.

MarcosAPMT, 2024-03-05:
This seems to be working well. I think the issue can be closed once this is implemented in a final release. Thanks a lot!

closed 2024-04-12 by MarcosAPMT