meatpiHQ/wican-fw #89: No TCP connection to torque or carscanner possible

atze09, 2024-02-09:
I successfully connected my wican to my wifi and can access it via PC in the browser, but my phone can not access it with the torque nor the car scanner app:
Those are my settings (FW 2.9):
[image]

meatpiHQ, 2024-02-09:
@atze09 can you please screenshot your CAR scanner app configuration? 

Also I assume you submitted the configuration and it is stored successfully?

atze09, 2024-02-09:
[image]
This is car scanner settings. Yes I saved the setting in the wican

meatpiHQ, 2024-02-09:
@atze09 please try to set the ecu protocol, 500K or 250k. 

https://github.com/meatpiHQ/wican-fw/issues/17#issuecomment-1454783099

atze09, 2024-02-09:
Unfortunately none of the settings is working.
I also turned off mobile data and mqtt, but without success

meatpiHQ, 2024-02-09:
Please try changing the CAN bitrate to 500K instead of Auto

[image]

atze09, 2024-02-09:
Pixel 8 phone was not working, but with Samsung Tab i could connect to ELM, but not to ECU. I think i need to start the engine to connect to ECU.
I will try that later.

atze09, 2024-02-15:
Now i have bought the cable, needed to switch the ground pin for WICAN to work.
But i can not get access to ECU. i have tried with same setting with another bluetooth OBD adapter and that works flawlessly.

Here is the car scanner log:
`
===== DISCONNECTED AT: 15.02.2024 15:33:52
===== LOG STARTED: 15.02.2024 15:35:18=====
=====Version=1.104.0 Build=401040 P=True NoDelayELM327Init=False
====Connection profile=Opel OBD-II / EOBD Electric / Hybrids (PSA/Stellantis based)
[Connection type: WIFI]
[WiFi name: <unknown ssid>]

[Connecting to 192.168.17.139:3333]
TCPConnectionV2
Start binding WiFi network
Wifi networks found:1

[Connected to 192.168.17.139:3333]
ATZ
ATE0

[ELM RESET]
ELM327 v1.3a

>
Initialize(initMode=Default)

InitializeCustomInitString(initMode=Default)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATZ
ELM327 v1.3a

>ATE0
OK

>ATH1
OK

>ATSPA6
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>ATSH6B4
OK

>ATFCSH6B4
OK

>ATFCSD300000
OK

>ATFCSM1
OK

>ATCRA694
OK

>22D816
BUS INIT: ...ERROR

>22D816
BUS INIT: ...ERROR

>ATDPN
a

>22D816
BUS INIT: ...ERROR

>22D816
BUS INIT: ...ERROR

>ATDPN
a

>
InitializeCustomInitString(initMode=Default)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATZ
ELM327 v1.3a

>ATE0
OK

>ATH1
OK

>ATSPA6
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>ATSH6B4
OK

>ATFCSH6B4
OK

>ATFCSD300000
OK

>ATFCSM1
OK

>ATCRA694
OK

>22D816
BUS INIT: ...ERROR

>22D816
BUS INIT: ...ERROR

>ATDPN
a

>22D816
BUS INIT: ...ERROR

>22D816
BUS INIT: ...ERROR

>ATDPN
a

>
InitializeCustomInitString(initMode=Default)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATZ
ELM327 v1.3a

>ATE0
OK

>ATH1
OK

>ATSPA6
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>ATSH6B4
OK

>ATFCSH6B4
OK

>ATFCSD300000
OK

>ATFCSM1
OK

>ATCRA694
OK

>22D816
BUS INIT: ...ERROR

>22D816
BUS INIT: ...ERROR

>ATDPN
a

>22D816
BUS INIT: ...ERROR

>22D816
BUS INIT: ...ERROR

>ATDPN
a

>
Stop(Disconnect:UserClickbtnDisconnect)

[SetRunning(False,From Stop #3640)]

Stopped`

atze09, 2024-02-15:
Is there another easy way to sniff the right can requests?

meatpiHQ, 2024-02-15:
@atze09 The issue here is the command ATSPA6, it's trying to set the protocol as automatic. Try setting it manually in carscanner app to "11bit 500k"

atze09, 2024-02-16:
> @atze09 The issue here is the command ATSPA6, it's trying to set the protocol as automatic. Try setting it manually in carscanner app to "11bit 500k"

Unfortunately the same result (even with engine on). Any other setting which could make the difference?

meatpiHQ, 2024-02-16:
@atze09 can you please post the log when you set it to 11bit 500k

atze09, 2024-02-16:
Of course, here is the log:
`===== LOG STARTED: 16.02.2024 07:07:45=====
=====Version=1.104.0 Build=401040 P=True NoDelayELM327Init=True
====Connection profile=Opel OBD-II / EOBD Electric / Hybrids (PSA/Stellantis based)
[Connection type: WIFI]
[WiFi name: <unknown ssid>]

[Connecting to 192.168.17.139:3333]
TCPConnectionV2
Start binding WiFi network
Wifi networks found:1
Start binding WiFi network
Wifi networks found:1

[Connected to 192.168.17.139:3333]
ATZ
ATE0

[ELM RESET]
ELM327 v1.3a

>
Initialize(initMode=Default)

InitializeDefaultInitString(ProtocolNumber=6,initMode=Default,useATST96=False)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATD
OK

>ATD0
OK

>ATE0
OK

>ATH1
OK

>ATSP6
OK

>ATE0
OK

>ATH1
OK

>ATM0
OK

>ATS0
OK

>ATAT1
OK

>ATAL
?

>0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]

InitializeDefaultInitString(ProtocolNumber=6,initMode=Default,useATST96=False)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATD
OK

>ATD0
OK

>ATE0
OK

>ATH1
OK

>ATSP6
OK

>ATE0
OK

>ATH1
OK

>ATM0
OK

>ATS0
OK

>ATAT1
OK

>ATAL
?

>0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]

InitializeDefaultInitString(ProtocolNumber=0,initMode=Default,useATST96=False)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATD
OK

>ATD0
OK

>ATE0
OK

>ATH1
OK

>ATSP0
OK

>ATE0
OK

>ATH1
OK

>ATM0
OK

>ATS0
OK

>ATAT1
OK

>ATAL
?

>0100
BUS INIT: ...ERROR

>
[cmd_reply=] BUS INIT: ...ERROR

>ATDPN
0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
BUS INIT: ...ERROR

>
[cmd_reply=] BUS INIT: ...ERROR

>ATDPN
0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
BUS INIT: ...ERROR

>
[cmd_reply=] BUS INIT: ...ERROR

>ATDPN
0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
BUS INIT: ...ERROR

>
[cmd_reply=] BUS INIT: ...ERROR

>ATDPN
0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
BUS INIT: ...ERROR

>
[cmd_reply=] BUS INIT: ...ERROR

>ATDPN
0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
BUS INIT: ...ERROR

>
[cmd_reply=] BUS INIT: ...ERROR

>ATDPN
0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
BUS INIT: ...ERROR

>
[cmd_reply=] BUS INIT: ...ERROR

>ATDPN
0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
BUS INIT: ...ERROR

>
[cmd_reply=] BUS INIT: ...ERROR

>ATDPN
0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]

Stop(Disconnect:UserClickbtnDisconnect)

[SetRunning(False,From Stop #3640)]

Stopped

===== DISCONNECTED AT: 16.02.2024 07:08:22`

meatpiHQ, 2024-02-16:
@atze09 can you please also post the logs from the other OBD adaptor that you mentioned. Just to compare, also leave it at 11bit 500k

atze09, 2024-02-16:
> @atze09 can you please also post the logs from the other OBD adaptor that you mentioned. Just to compare, also leave it at 11bit 500k

If i do so, i get the same errer see the log `===== LOG STARTED: 16.02.2024 12:36:19=====
=====Version=1.104.0 Build=401040 P=True NoDelayELM327Init=True
====Connection profile=Opel OBD-II / EOBD Electric / Hybrids (PSA/Stellantis based)
[Connection type: Bluetooth; DeviceName=OBD BLE]

[Connecting to 66:1E:11:0E:3C:AB]
bt_socket.Connected
got streams from client

[Connected to 66:1E:11:0E:3C:AB]
ATZ
ATE0

[ELM RESET]
ATZ

OBDII  v1.5

>
Initialize(initMode=Default)

InitializeDefaultInitString(ProtocolNumber=6,initMode=Default,useATST96=False)
ATE0
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATD

OK

>ATD0
ATD0
OK

>ATE0
ATE0
OK

>ATH1
OK

>ATSP6
OK

>ATE0
OK

>ATH1
OK

>ATM0
OK

>ATS0
OK

>ATAT1
OK

>ATAL
OK

>0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]

InitializeDefaultInitString(ProtocolNumber=6,initMode=Default,useATST96=False)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATD

OK

>ATD0
ATD0
OK

>ATE0
ATE0
OK

>ATH1
OK

>ATSP6
OK

>ATE0
OK

>ATH1
OK

>ATM0
OK

>ATS0
OK

>ATAT1
OK

>ATAL
OK

>0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]
0100
NO DATA

>
[cmd_reply=] NO DATA

>
[Trying to decode:
OK

>
ELMFormat=CAN11bit]

[DecodeResult=False]

InitializeDefaultInitString(ProtocolNumber=0,initMode=Default,useATST96=False)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATD

OK

>ATD0
ATD0
OK

>ATE0
ATE0
OK

>ATH1
OK

>ATSP0
OK

>ATE0
OK

>ATH1
OK

>ATM0
OK

>ATS0
OK

>ATAT1
OK

>ATAL
OK

>0100
SEARCHING...
UNABLE TO CONNECT

>
[cmd_reply=] SEARCHING...
UNABLE TO CONNECT

>ATDPN
A0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
SEARCHING...
UNABLE TO CONNECT

>
[cmd_reply=] SEARCHING...
UNABLE TO CONNECT

>ATDPN
A0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
SEARCHING...
UNABLE TO CONNECT

>
[cmd_reply=] SEARCHING...
UNABLE TO CONNECT

>ATDPN
A0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
SEARCHING...
UNABLE TO CONNECT

>
[cmd_reply=] SEARCHING...
UNABLE TO CONNECT

>ATDPN
A0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
SEARCHING...
UNABLE TO CONNECT

>
[cmd_reply=] SEARCHING...
UNABLE TO CONNECT

>ATDPN
A0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
SEARCHING...
UNABLE TO CONNECT

>
[cmd_reply=] SEARCHING...
UNABLE TO CONNECT

>ATDPN
A0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
SEARCHING...
UNABLE TO CONNECT

>
[cmd_reply=] SEARCHING...
UNABLE TO CONNECT

>ATDPN
A0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]
0100
SEARCHING...
UNABLE TO CONNECT

>
[cmd_reply=] SEARCHING...
UNABLE TO CONNECT

>ATDPN
A0

>
ProtocolFromATDPN=0

[Trying to decode:
OK

>
ELMFormat=Unknown]

[DecodeResult=False]

Stop(Disconnect:UserClickbtnDisconnect)

[SetRunning(False,From Stop #3640)]

Stopped

===== DISCONNECTED AT: 16.02.2024 12:37:36`

atze09, 2024-02-16:
But if i chose custom under ecu initialization:
Then it is working with the following log
`===== LOG STARTED: 16.02.2024 12:47:33=====
=====Version=1.104.0 Build=401040 P=True NoDelayELM327Init=True
====Connection profile=Opel OBD-II / EOBD Electric / Hybrids (PSA/Stellantis based)
[Connection type: Bluetooth; DeviceName=OBD BLE]

[Connecting to 66:1E:11:0E:3C:AB]
bt_socket.Connected
got streams from client

[Connected to 66:1E:11:0E:3C:AB]
ATZ
ATE0

[ELM RESET]

OBDII  v1.5

>
Initialize(initMode=Default)

InitializeCustomInitString(initMode=Default)
ATE0
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATZ

OBDII  v1.5

>ATE0
ATE0
OK

>ATH1
OK

>ATSPA6
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>ATSH6B4
OK

>ATFCSH6B4
OK

>ATFCSD300000
[... 1599 lines cut ...]
6940762D81600012D05

>22D816
6940762D81600012D02

>22D816
6940762D81600012D06

>22D816
6940762D81600012D05

>22D816
6940762D81600012CFA

>22D816
6940762D81600012D0A

>
Stop(Disconnect:UserClickbtnDisconnect)

[SetRunning(False,From Stop #3640)]

Stopped

===== DISCONNECTED AT: 16.02.2024 12:48:15`

meatpiHQ, 2024-02-16:
@atze09 thanks. I'll try to replicate this issue, this will take a while.

meatpiHQ, 2024-02-17:
@atze09 Please try this new firmware, select "custom under ecu initialization".

[wican-fw_v293_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/14317274/wican-fw_v293_hv300.zip)

atze09, 2024-02-18:
> @atze09 Please try this new firmware, select "custom under ecu initialization".
> 
> [wican-fw_v293_hv300.zip](https://github.com/meatpiHQ/wican-fw/files/14317274/wican-fw_v293_hv300.zip)

Awesome this solved the issue, i can successfully connect to ECU and see requests on MQTT (RX and TX)
Thanks you very much.

meatpiHQ, 2024-02-18:
@atze09 thanks for confirming

atze09, 2024-08-14:
> @atze09 thanks for confirming

How can i transfer this to new automate tab?
i tried with wican-fw_v320_hv300 firmware but i cannot get acces to carscanner app once again.

For documentation this was working for me:
[image]

meatpiHQ, 2024-09-22:
@atze09 What is your car's make and model? I can help you make a car profile

atze09, 2024-09-22:
> @atze09 What is your car's make and model? I can help you make a car profile

It is an Opel electro corsa, found the information on another forum.

meatpiHQ, 2024-09-22:
@atze09 that's great! can you please share the link to the forum?

If you have already made a car profile please make a pull request.