meatpiHQ/wican-fw #90: Home assistant NODE RED Mqtt

atze09, 2024-02-12:
I successfully connected my wican to my wifi and can access it via PC and also home assistant.
Those are my settings (FW 2.9):
I see a lot of traffic on the mqtt, when charging my car.
If it is not charging there is neraly no traffic:
[image]

I also installed the Node Red example and this one is also connected (green dots), but it does not publish CAR1/fuelstatus to mqtt.
[image]

Can you help me publish my fuel level or soc to mqtt?

meatpiHQ, 2024-02-12:
@atze09 check out this tutorial it might be easier if you don't use NodeRED

https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

atze09, 2024-02-12:
> @atze09 check out this tutorial it might be easier if you don't use NodeRED
> 
> https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

The problem is I cannot access ECU. I have tried all combinations in car Scanner App and also in meatpi.

meatpiHQ, 2024-02-12:
@atze09 so your car is Opel Corsa-e ?

atze09, 2024-02-13:
> @atze09 so your car is Opel Corsa-e ?

Yes

meatpiHQ, 2024-02-13:
@atze09 ok here's the issue, you need a special cable for this to work. The reason is that the Corsa-e diagnostics CAN bus is located on pins 3 and 8. WICAN-OBD like almost all adapters CAN bus pins are on 6 and 14. See the how comments https://github.com/EVNotify/EVNotify/issues/205#issuecomment-810540420

That's the reason you cannot connect to the ECU #89 

You could make a cable like this one https://github.com/EVNotify/EVNotify/issues/205#issuecomment-822307374

atze09, 2024-02-13:
> @atze09 ok here's the issue, you need a special cable for this to work. The reason is that the Corsa-e diagnostics CAN bus is located on pins 3 and 8. WICAN-OBD like almost all adapters CAN bus pins are on 6 and 14. See the comments [EVNotify/EVNotify#205 (comment)](https://github.com/EVNotify/EVNotify/issues/205#issuecomment-810540420)
> 
> That's the reason you cannot connect to the ECU #89
> 
> You could make a cable like this one [EVNotify/EVNotify#205 (comment)](https://github.com/EVNotify/EVNotify/issues/205#issuecomment-822307374)

I have ordered it and give you a feedback. Thank you!

atze09, 2024-02-15:
> I have ordered it and give you a feedback. Thank you!

See answer in different issue.

atze09, 2024-02-18:
As i now can successfully get connected to ECU i reqested som data on the carscanner app:
`soc
{"bus":"0","type":"tx","ts":22377,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,16,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":34216,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,103,10,0,0]}]}
soh
{"bus":"0","type":"tx","ts":43224,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,216,96,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":50140,"frame":[{"id":1684,"dlc":7,"rtr":false,"extd":false,"data":[6,98,216,96,1,5,186,0]}]}
obd V
{"bus":"0","type":"rx","ts":11202,"frame":[{"id":1684,"dlc":8,"rtr":false,"extd":false,"data":[7,98,216,22,0,1,45,19]}]}
{"bus":"0","type":"tx","ts":33312,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,216,22,170,170,170,170]}]}
dc capacity
{"bus":"0","type":"tx","ts":23782,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,216,101,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":42007,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,216,101,4,225,0,0]}]}
`

atze09, 2024-02-19:
> @atze09 ok here's the issue, you need a special cable for this to work. The reason is that the Corsa-e diagnostics CAN bus is located on pins 3 and 8. WICAN-OBD like almost all adapters CAN bus pins [...]

Even with engine started, i do not receive any rx message, is my JSON Example the right one?

atze09, 2024-02-19:
> @atze09 the filter and JSON looks ok. Are you able to request using mqtt explorer? Do you have the "elm327 log" disabled or enabled?

ELM327 log is disabled.
The request is published to mqqt by home assistant under tx, but i do not receive rx.
Do you mean to manually publish the tx request vie mqtt explorer?
Engine can also be off?

atze09, 2024-02-19:
> @atze09 the filter and JSON looks ok. Are you able to request using mqtt explorer? Do you have the "elm327 log" disabled or enabled?

I have tried it also manually but to rx. Engine was off at this time.

meatpiHQ, 2024-03-07:
@atze09 sorry for the delay. 

> Engine can also be off?

This depends on the car, usually you can request some data while the car is off but it has to be unlocked if im not mistaken 

> The request is published to mqqt by home assistant under tx, but i do not receive rx.

can you please post here the request you're sending from Home assistant 

> Do you mean to manually publish the tx request vie mqtt explorer?

Yes. There is one thing to keep in mind, once you add a filter all RX frame will be ignored unless it matches the filter. So only the PID in the filter will be sent to Home assistant

atze09, 2024-04-30:
@meatpiHQ 👍 Maybe you can now help me get it working, i already upgraded to latest firmware.
My issue is quite the same as:
https://github.com/meatpiHQ/wican-fw/issues/119

meatpiHQ, 2024-05-02:
@atze09 can you post mqtt logs for the SoC, with the SoC value at that moment ?

Make sure to click on the SoC sensor in carscanner.

https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

atze09, 2024-05-03:
> @atze09 can you post mqtt logs for the SoC, with the SoC value at that moment ?
> 
> Make sure to click on the SoC sensor in carscanner.
> 
> https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

Great i did it for some points, should be quite similar to the other thread?
SOC 73,56
{"bus":"0","type":"rx","ts":36548,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,147,33,0,0]}

SOC 73,54
{"bus":"0","type":"rx","ts":10853,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,147,19,0,0]}]}

SOC 73,48 or 73,49
{"bus":"0","type":"rx","ts":27902,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,146,247,0,0]}]}

SOC 73,46
{"bus":"0","type":"rx","ts":18365,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,146,236,0,0]}]}

But i am struggling in setting the Filter and getting some RX data out of the car.

atze09, 2024-05-03:
This was the TX log
`
{"bus":"0","type":"tx","ts":22377,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,16,170,170,170,170]}]}
{"bus":"0","type":"tx","frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,91,170,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":22377,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,16,170,170,170,170]}]}
{"bus":"0","type":"tx","frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,91,170,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":22377,"frame":[{"id":1716,"dlc":8,"rtr":false,"extd":false,"data":[3,34,212,16,170,170,170,170]}]}
{"bus":"0","type":"tx","frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,91,170,170,170,170,170]}]}`

And this the RX LOG:
`{"bus":"0","type":"rx","ts":41744,"frame":[{"id":851,"dlc":5,"rtr":false,"extd":false,"data":[0,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":41794,"frame":[{"id":851,"dlc":5,"rtr":false,"extd":false,"data":[0,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":41844,"frame":[{"id":851,"dlc":5,"rtr":false,"extd":false,"data":[0,0,0,0,0,0,0,0]}]}
And the obove mentioned id1684, but i couldn*t find them in the mqtt log, i just copied them live.

`

meatpiHQ, 2024-05-04:
@atze09 I think the SoC response is clear, but we need more data points to confirm, other the 73%.

For the TX log its not clear 
 

> SOC 73,56
> {"bus":"0","type":"rx","ts":36548,"frame":[{"id":1684,"dlc":6,"rtr":false,"extd":false,"data":[5,98,212,16,**147**,33,0,0]}

**SoC=147/2=73.5%**

atze09, 2024-08-13:
Now it works with the informations from here:
https://github.com/meatpiHQ/wican-fw/issues/119

closed 2024-08-13 by atze09