meatpiHQ/wican-fw #92: Transfer/Logic: From the PID Car-Scanner shows to the "right" TX-Phrase

HaareistNase, 2024-02-19:
Dear Sirs,
as a non-Programmer its not easy to understand the connection between what CarScanner shows, and how I get this data.

I have a e-up, and accidentially stumbled about the tx-phrase I have to send to the car to get the bms. This is working so far.
But I want to understand what I do there, and I have a e-Niro, who is not willing to reveal his battery-soc to me.

I connected car scanner to wican and press the button for the soc.

I get this pid.

{"bus":"0","type":"rx","ts":42141,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[6,65,0,128,0,0,1,170]}]}

So I understand that the PID I have to get from the car ist 2028. And I learned, that the current soc is in bit 4 and I have to divide it by two, to get the right number for my soc.

But what do I have to send to the car, to get this? And how do I calculate it? Its seem like a bunch of bits, but I do not understand what bit have to has what value, and where I find or calculate it.

Please help me to understand.

Thank you.

meatpiHQ, 2024-02-20:
@HaareistNase you're on the right track.

> I connected car scanner to wican and press the button for the soc.

Where is the tx frame? 

Can you please post a screenshot of your configuration page.

HaareistNase, 2024-02-20:
@meatpiHQ 

The Car Scanner generates a TX-Frame? This is the magic? I just copy the TX-Frame from Car-Scanner?
I will send as soon I get back from work.

HaareistNase, 2024-02-20:
Thank you for the Link.
I did read it before, but I never realized that the case is so simple, and everyone works like this. Maybe I did not read close enough, or it has to be mentioned more prominent, that the magic is just copy the tx from Carscanner^^
I thought, that there must be a deeper understanding, what all you cracks and hackers do.
Thank you very much. Thats works perfect for me.

meatpiHQ, 2024-02-21:
@HaareistNase it would great if you post here the MQTT tx and rx frames that worked for each car. It might help other users.

HaareistNase, 2024-02-21:
First the Logs.

Car Scanner is Sending:

{"bus":"0","type":"tx","ts":27332,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,1,1,170,170,170,170]}]}

the Car respond with
{"bus":"0","type":"rx","ts":27526,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[16,62,98,1,1,255,247,231]}]}

After That Car Scanner is sending 

{"bus":"0","type":"tx","ts":27343,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,10,170,170,170,170,170]}]}

And After that I get the 7 Frames in wich the SOC is Coded [actual 53%]
{"bus":"0","type":"rx","ts":27356,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[33,255,103,0,0,0,0,3]}]}
{"bus":"0","type":"rx","ts":27371,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[34,0,3,14,22,11,11,11]}]}
{"bus":"0","type":"rx","ts":27386,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[35,10,11,11,0,0,11,184]}]}
{"bus":"0","type":"rx","ts":27401,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[36,11,183,69,0,0,147,0]}]}
{"bus":"0","type":"rx","ts":27416,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[37,3,112,208,0,3,106,119]}]}
{"bus":"0","type":"rx","ts":27431,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,1,67,139,0,1,55]}]}
{"bus":"0","type":"rx","ts":27446,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[39,39,0,110,159,240,13,1]}]}
{"bus":"0","type":"rx","ts":27461,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[40,104,0,0,0,0,3,232]}]}

meatpiHQ, 2024-02-21:
@HaareistNase alright thanks for that. What is the formula for the SoC?

you can use home assistant to do the calculation or use the firmware mentioned in the discussion. This is still pre release version.

https://github.com/meatpiHQ/wican-fw/discussions/75#discussioncomment-8335407

HaareistNase, 2024-02-22:
Im Referencing to https://github.com/meatpiHQ/wican-fw/discussions/86
Kona 2019 and my e-Niro 2019 share the same basis.

New Day, new Data.

Im Questioning the e-Niro by

alias: MQTT Niro Soc Abfrage
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.niro_status_mqtt
    to: online
condition: []
action:
  - repeat:
      sequence:
        - condition: state
          entity_id: sensor.niro_status_mqtt
          state: online
        - service: mqtt.publish
          data:
            qos: 0
            retain: false
            topic: wican/5432048f58a1/can/tx
            payload: >-
              {"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,1,1,170,170,170,170]}]}
        - delay:
            hours: 0
            minutes: 0
            seconds: 0
            milliseconds: 200
        - service: mqtt.publish
          data:
            qos: 0
            retain: false
            topic: wican/5432048f58a1/can/tx
            payload: >-
              {"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,10,170,170,170,170,170]}]}
        - delay:
            hours: 0
            minutes: 1
            seconds: 0
            milliseconds: 0
      until:
        - condition: state
          entity_id: sensor.niro_status_mqtt
          state: offline
mode: single

 {"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,1,1,170,170,170,170]}]}
 {"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,10,170,170,170,170,170]}]}

Today I get the "right" Data.

{"bus":"0","type":"rx","ts":20657,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[33,255,86,16,118,1,74,3]}]}
{"bus":"0","type":"rx","ts":20667,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[34,254,231,14,19,15,14,15]}]}
{"bus":"0","type":"rx","ts":20677,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[35,14,15,15,0,0,13,183]}]}
{"bus":"0","type":"rx","ts":20687,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[36,20,183,69,0,0,131,0]}]}
{"bus":"0","type":"rx","ts":20697,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[37,3,113,198,0,3,108,15]}]}
{"bus":"0","type":"rx","ts":20707,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,1,67,226,0,1,55]}]}
{"bus":"0","type":"rx","ts":20717,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[39,181,0,110,216,254,9,1]}]}
{"bus":"0","type":"rx","ts":20727,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[40,104,0,0,0,0,3,232]}]}

My Soc is coded in Frame 33.
Its Bit Two and its Coded V/2 (Here 86/2)

Now I try this to set as a filter in Wican.

[image]

But the Data I recive did not Change. Its still all Frames from 33 to 40.

meatpiHQ, 2024-02-22:
@HaareistNase this is great. Please update to V2.92 and set 
PID: 33
Index: 2

[V2.92](https://github.com/meatpiHQ/wican-fw/discussions/75#discussioncomment-8335407) introduces index for PID.

HaareistNase, 2024-02-22:
And Leave Start Bit end Bit-Length to "0"?

meatpiHQ, 2024-02-22:
Sorry I made a mistake. **The index should be 0** no 2. 

And the expression, you can set it to B2/2.  Referring to byte 2. Or you can keep your current expression, it is correct.

If you use B2, it will ignore the start and end but.

HaareistNase, 2024-02-22:
OMG: Its Working.
Thank you so much for your endless Effort to get a Non-Checker like me to get this DONE.

meatpiHQ, 2024-02-22:
Glad to hear that! You did great!

I planing to use this info to create a database, to make it easier for everyone.

Mathopenevse, 2024-03-08:
> Im Referencing to #86 Kona 2019 and my e-Niro 2019 share the same basis.
> 
> New Day, new Data.
> 
> Im Questioning the e-Niro by
> 
> alias: MQTT Niro Soc Abfrage description: "" trigger:
> 
> * platform: state
>   entity_id:
>   
>   * sensor.niro_status_mqtt
>     to: online
>     condition: []
>     action:
> * repeat:
>   sequence:
>   - condition: state
>   entity_id: sensor.niro_status_mqtt
>   state: online
>   - service: mqtt.publish
>   data:
>   qos: 0
>   retain: false
>   topic: wican/5432048f58a1/can/tx
>   payload: >-
>   {"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,1,1,170,170,170,170]}]}
>   - delay:
>   hours: 0
>   minutes: 0
>   seconds: 0
>   milliseconds: 200
>   - service: mqtt.publish
>   data:
>   qos: 0
>   retain: false
>   topic: wican/5432048f58a1/can/tx
>   payload: >-
>   {"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,10,170,170,170,170,170]}]}
>   - delay:
>   hours: 0
>   minutes: 1
>   seconds: 0
>   milliseconds: 0
>   until:
>   - condition: state
>   entity_id: sensor.niro_status_mqtt
>   state: offline
>   mode: single
> 
> {"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,1,1,170,170,170,170]}]} {"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,10,170,170,170,170,170]}]}
> 
> Today I get the "right" Data.
> 
> {"bus":"0","type":"rx","ts":20657,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[33,255,86,16,118,1,74,3]}]} {"bus":"0","type":"rx","ts":20667,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[34,254,231,14,19,15,14,15]}]} {"bus":"0","type":"rx","ts":20677,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[35,14,15,15,0,0,13,183]}]} {"bus":"0","type":"rx","ts":20687,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[36,20,183,69,0,0,131,0]}]} {"bus":"0","type":"rx","ts":20697,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[37,3,113,198,0,3,108,15]}]} {"bus":"0","type":"rx","ts":20707,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,1,67,226,0,1,55]}]} {"bus":"0","type":"rx","ts":20717,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[39,181,0,110,216,254,9,1]}]} {"bus":"0","type":"rx","ts":20727,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[40,104,0,0,0,0,3,232]}]}
> 
> My Soc is coded in Frame 33. Its Bit Two and its Coded V/2 (Here 86/2)
> 
> Now I try this to set as a filter in Wican.
> 
> ![grafik](https://private-user-images.githubusercontent.com/135537446/307177262-024d018f-aee4-419f-8375-2604a7f52b2b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk4OTA0NjQsIm5iZiI6MTcwOTg5MDE2NCwicGF0aCI6Ii8xMzU1Mzc0NDYvMzA3MTc3MjYyLTAyNGQwMThmLWFlZTQtNDE5Zi04Mzc1LTI2MDRhN2Y1MmIyYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwOFQwOTI5MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YTVlNGZkM2NlMzIwOGQyMDFjYWM3ZDA2MmJkMTJhMDE2MzBlYWVjZTY3NGU1ZmY0YjcxOWJiNmEyYmNkM2ViJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Tv8AC7Q9zG1xnBZEGv_MX3AsBTyY71359OL2_C1Mq8U)
> 
> But the Data I recive did not Change. Its still all Frames from 33 to 40. And I do not know how to filter the Frame 33 out of the RX-Data (without Using Node-Red).
> 
> I try to use your Beta Firmware ![grafik](https://private-user-images.githubusercontent.com/135537446/307178675-e07afd2e-b32d-4b7a-959d-3e3b89b2da10.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk4OTA0NjQsIm5iZiI6MTcwOTg5MDE2NCwicGF0aCI6Ii8xMzU1Mzc0NDYvMzA3MTc4Njc1LWUwN2FmZDJlLWIzMmQtNGI3YS05NTlkLTNlM2I4OWIyZGExMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwOFQwOTI5MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MzQ3ZjZjNTI1YTE3MjkwNDMwM2Q3MzIzY2YyNTY2YzdlYjdiZjQyNjg1YzU4NGFjOTM5YjdjNGQ5ODZhYzAzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.BNqsPLFzB2VjvpUB_tb6Hjv_KkPCVhJ1ncdKtD27PVM)

hi mister, could you confirm that you have a E-Niro?
i have tried to send this two frames with delay 200ms but I receive nothing...

`{"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,1,1,170,170,170,170]}]}
{"bus":"0","type":"tx","frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,10,170,170,170,170,170]}]}

`

HaareistNase, 2024-03-08:
Yes. E-Niro in Germany, Nero EV in other Countries.

Mathopenevse, 2024-03-09:
> Yes. E-Niro in Germany, Nero EV in other Countries.

Thanks for your reply, i will try again