meatpiHQ/wican-fw #95: SoC auslesen Fiat 500e

michi741, 2024-02-23:
Hallo.

Möchte gern den SoC eines Fiat 500e auslesen. Ich weiß nicht genau, was ich in den Filter des Wican eintragen muss.
[image]

Ich habe die Daten für tx und rx mit Car Scanner und MQTT Explorer ausgelesen.
Diese Werte sind angekommen.

{"bus":"0","type":"rx","ts":53691,"frame":[{"id":417001794,"dlc":6,"rtr":false,"extd":true,"data":[5,98,1,158,3,93,0,0]}]}
{"bus":"0","type":"tx","ts":53683,"frame":[{"id":416957169,"dlc":8,"rtr":false,"extd":true,"data":[3,34,1,158,170,170,170,170]}]}

{"bus":"0","type":"rx","ts":52793,"frame":[{"id":417001796,"dlc":8,"rtr":false,"extd":true,"data":[33,198,197,195,0,0,0,0]}]}
{"bus":"0","type":"tx","ts":52788,"frame":[{"id":416957681,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}

{"bus":"0","type":"rx","ts":52786,"frame":[{"id":417001796,"dlc":8,"rtr":false,"extd":true,"data":[16,9,98,160,16,197,198,197]}]}
{"bus":"0","type":"tx","ts":52783,"frame":[{"id":416957681,"dlc":8,"rtr":false,"extd":true,"data":[3,34,160,16,170,170,170,170]}]}

meatpiHQ, 2024-02-23:
@michi741 if you want to read the SoC, i suggest that in carscanner click on "All Sensors", then click on Battery SoC. This way it will continuously request the SoC. This will make it easier to find the tx and rx frame.

Post the frame here and ill will try to help you set the filter. You'll also need to note the SoC when you are doing the reading so we can try and find how to calculate it.

lmuser22, 2024-02-23:
Which "Car Scanner" App is the right one?

michi741, 2024-02-23:
Das ist angekommen.
Aktueller SoC 77,25%

[image]
[image]
[image]
[image]
[image]

lmuser22, 2024-02-23:
> Das ist angekommen. Aktueller SoC 77,25%
> 
> <img alt="image" width="773" src="https://private-user-images.githubusercontent.com/120442108/307332529-ac60d5ab-73fe-4972-ba68-ba487b2fa746.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2OTU0MzcsIm5iZiI6MTcwODY5NTEzNywicGF0aCI6Ii8xMjA0NDIxMDgvMzA3MzMyNTI5LWFjNjBkNWFiLTczZmUtNDk3Mi1iYTY4LWJhNDg3YjJmYTc0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyM1QxMzMyMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMGQxMzQ0ZmYwMzllNzRhNDNkYmEwNjcwNGM2ZTVlNjBlOTJmMTY1MzZhY2Q4YmMyOWMxOTcyMWVkMDgxYzE0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qPMoLnRE7KbW_E0dnqJ9MmBr92lF76USpxb0pnGKk9k"> <img alt="image" width="757" src="https://private-user-images.githubusercontent.com/120442108/307332951-4de96447-4986-4d54-a07d-35a05075db46.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2OTU0MzcsIm5iZiI6MTcwODY5NTEzNywicGF0aCI6Ii8xMjA0NDIxMDgvMzA3MzMyOTUxLTRkZTk2NDQ3LTQ5ODYtNGQ1NC1hMDdkLTM1YTA1MDc1ZGI0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyM1QxMzMyMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NzlmMjY5YTU5NGY1NDE3MzE1Zjk0NmU4YWIzZGMxNTJjNTcyMTAwMTEzOWZiYzE4ZGI4NjZiMTIwOTU0MGJjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iceTaH_1m-XtcBEv9RYO6MG54f-1Cnoy9GnJzVmu9T8"> <img alt="image" width="759" src="https://private-user-images.githubusercontent.com/120442108/307333079-4dfbc161-9d5a-4888-b676-f46dab678b65.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2OTU0MzcsIm5iZiI6MTcwODY5NTEzNywicGF0aCI6Ii8xMjA0NDIxMDgvMzA3MzMzMDc5LTRkZmJjMTYxLTlkNWEtNDg4OC1iNjc2LWY0NmRhYjY3OGI2NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyM1QxMzMyMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZmE5YmFhNTZmYzQ2OGRmZDk2YWVhMDU0MTExNDM4NDAzYzg0N2JiNWJmMDdjNDIxYWEwZTNjYWFkMWUwYjRhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-cS1h9Hhaitrsi_UdDteCYnaexcrfq8WR4V6YsPi6dA"> <img alt="image" width="754" src="https://private-user-images.githubusercontent.com/120442108/307333264-64f92804-8fff-48a1-ab37-c826093504ed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2OTU0MzcsIm5iZiI6MTcwODY5NTEzNywicGF0aCI6Ii8xMjA0NDIxMDgvMzA3MzMzMjY0LTY0ZjkyODA0LThmZmYtNDhhMS1hYjM3LWM4MjYwOTM1MDRlZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyM1QxMzMyMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kM2ZlMzlhNzUwNDY4YmI4OTgyYjNmM2M2NjFiMjVkZjhmYjIwODMwNjBjYWFhOGJkZjM3ZmFjMThhNTZlODU2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.dB_79SyyDyX0UpMfw5hsfEWmVBIfICVZwr_ZZVrcRNY"> <img alt="image" width="757" src="https://private-user-images.githubusercontent.com/120442108/307333449-a386270f-b401-4d37-98fb-6957113b7a09.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2OTU0MzcsIm5iZiI6MTcwODY5NTEzNywicGF0aCI6Ii8xMjA0NDIxMDgvMzA3MzMzNDQ5LWEzODYyNzBmLWI0MDEtNGQzNy05OGZiLTY5NTcxMTNiN2EwOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyM1QxMzMyMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yODEwODRiZWE3ODVlOTgxMjIxNjMxOTJmNGQ5ZTJhM2MwMDVlZTQ5MDNiZWZjMThhMTM5YWMxNDI5NGRhYzRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.YFjrTM17UysriE9SqFkWz7s2kjkeswhixx7877Ua00Q">

womit liest Du die Daten aus? Handy App / Programm am PC? Habe den Dongle auch, er ist auch bereits im WLAN erreichbar aber mehr schaff ich noch nicht ohne Hilfe

michi741, 2024-02-23:
Schau dir die Anleitung in folgenden Link an.
https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

lmuser22, 2024-02-23:
bei 3. komm ich nicht weiter.

Using "Car Scanner" connect to WiCAN

Was für ein Car Scanner?

michi741, 2024-02-23:
Ich hab dafür die Android App "Car Scanner" verwendet.

lmuser22, 2024-02-23:
> Which "Car Scanner" App is the right one?

Diese App?
https://play.google.com/store/apps/details?id=com.ovz.carscanner&hl=de&gl=US&pli=1

meatpiHQ, 2024-02-23:
> Ja, diese App.

Yes that's correct

michi741, 2024-02-27:
Gibt es schon Informationen zu den Filtereinstellungen?

michi741, 2024-03-01:
I've tried everything, but reading the SOC doesn't work,

meatpiHQ, 2024-03-02:
@michi741 can you please capture more SoC data from car scanner ?

I can try to help you set up the filter.

michi741, 2024-03-23:
Is it possible that reading out the OBD data, such as the charge level, only works when the ignition is switched on?
Is there a way for it to work without ignition?

meatpiHQ, 2024-03-24:
@michi741 It is possible on some cars, have you tried to leave the car unlocked?

michi741, 2024-03-24:
Yes. Doesn't work, go to sleep anyway.

meatpiHQ, 2024-03-26:
@michi741 can you please try with car scanner, while unlocked and charging? See if you can get any data. Car scanner sends an init frame to the bus, if that works we can send the same frame from WiCAN.

michi741, 2024-04-01:
When I connect to the car scanner, this command is sent several times,

{"bus":"0","type":"tx","ts":5274,"frame":[{"id":416957169,"dlc":8,"rtr":false,"extd":true,"data":[3,34,1,158,170,170,170,170]}]}

nothing comes back from the car.

meatpiHQ, 2024-04-07:
@michi741 How about when the car is charging?

puni2k, 2024-04-20:
My Fiat 500e (Type 332) was delivered a few days ago and i found to have look at making it work with the wican dongle.

When requesting the SoC values carscanner send the following additional messages
```
{"bus":"0","type":"tx","ts":59058,"frame":[{"id":416957681,"dlc":8,"rtr":false,"extd":true,"data":[3,34,160,16,170,170,170,170]}]}

{"bus":"0","type":"tx","ts":38212,"frame":[{"id":416957681,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}
```
When you transmit the messages individually you however only get a reply for the first message, so for now I'm ignoring the second.

So here's the reply for the first message:
```
{"bus":"0","type":"rx","ts":57681,"frame":[{"id":417001796,"dlc":8,"rtr":false,"extd":true,"data":[16,9,98,160,16,172,173,172]}]}
```

After driving a bit around the block I got:
```
{"bus":"0","type":"rx","ts":20303,"frame":[{"id":417001796,"dlc":8,"rtr":false,"extd":true,"data":[16,9,98,160,16,169,173,169]}]}
```
So Byte 6 and 8 look promising. Not 100% sure though since using Byte*100/255 is close to the value in carscanner but not identical.
Also not the same as on the dash, but i didn't expect it to be. I adapted a Node-RED flow i've been previously using for my Citigo and will give the 500e a small charge today and check the results.

After successfully verification i can share my flow if anybody's interested. Just be warned that it contains some error handling and workarounds that were neccessary for the strange wakeup and sleep behaviour of the Citigo which might not be neccessary for the 500e

meatpiHQ, 2024-04-20:
@puni2k please do share, I'm sure someone out there will find it very useful.

puni2k, 2024-04-20:
Some good and some bad news.

My flow seems to be still good for the 500e.
Once the charger is unlocked wican starts up and requests and receives can data. Also the above mentioned value increases as charging progresses.

Here comes the bad news: After 20 minutes the car stops responding to requests. I assume it goes back to sleep after this time.
Have to think how i‘m gonna work around this issue.

puni2k, 2024-04-20:
As requested, my Node-RED flow:
```
[{"id":"24d2bae83e216617","type":"tab","label":"Wican 500e","disabled":false,"info":"","env":[]},{"id":"b315be78d912d7d5","type":"mqtt in","z":"24d2bae83e216617","name":"","topic":"wican/dc5475934095/can/status","qos":"2","datatype":"json","broker":"54787fa7a5e2bbba","nl":false,"rap":true,"rh":0,"inputs":0,"x":150,"y":60,"wires":[["1c03a269e59ce8e0"]]},{"id":"36442b567d9047e7","type":"inject","z":"24d2bae83e216617","name":"Timer","props":[],"repeat":"300","crontab":"","once":false,"onceDelay":"","topic":"","x":100,"y":380,"wires":[["c88fd84b155e98e3"]]},{"id":"9b993b42e1be27d3","type":"mqtt out","z":"24d2bae83e216617","name":"","topic":"wican/dc5475934095/status_retain","qos":"","retain":"true","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"54787fa7a5e2bbba","x":1160,"y":60,"wires":[]},{"id":"f786585b28a73fd5","type":"mqtt out","z":"24d2bae83e216617","name":"","topic":"wican/dc5475934095/can/tx","qos":"","retain":"false","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"54787fa7a5e2bbba","x":1140,"y":380,"wires":[]},{"id":"1e74e071190d8218","type":"mqtt in","z":"24d2bae83e216617","name":"","topic":"wican/dc5475934095/can/rx","qos":"2","datatype":"json","broker":"54787fa7a5e2bbba","nl":false,"rap":true,"rh":0,"inputs":0,"x":140,"y":620,"wires":[["a392908412e4838f","087f5335b1141ede"]]},{"id":"833ffd94d8dbbc87","type":"mqtt out","z":"24d2bae83e216617","name":"","topic":"wican/dc5475934095/SOC","qos":"","retain":"true","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"54787fa7a5e2bbba","x":1140,"y":620,"wires":[]},{"id":"3d15abae90b18fa5","type":"rbe","z":"24d2bae83e216617","name":"","func":"rbe","gap":"","start":"","inout":"out","septopics":true,"property":"payload","topi":"topic","x":370,"y":260,"wires":[["5d4aa57b53fd37d2"]]},{"id":"20631e93a0363f7f","type":"change","z":"24d2bae83e216617","name":"extract status","rules":[{"t":"set","p":"wicanStatus","pt":"flow","to":"payload.status","tot":"msg"},{"t":"set","p":"payload","pt":"msg","to":"payload.status","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":560,"y":60,"wires":[["9b993b42e1be27d3","3d15abae90b18fa5"]]},{"id":"1c03a269e59ce8e0","type":"trigger","z":"24d2bae83e216617","name":"debounce","op1":"","op2":"","op1type":"nul","op2type":"payl","duration":"10","extend":true,"overrideDelay":false,"units":"s","reset":"","bytopic":"all","topic":"topic","outputs":1,"x":380,"y":60,"wires":[["20631e93a0363f7f"]]},{"id":"461b06505c8c81d8","type":"switch","z":"24d2bae83e216617","name":"if noAck","property":"receive","propertyType":"flow","rules":[{"t":"false"}],"checkall":"true","repair":false,"outputs":1,"x":540,"y":440,"wires":[["c88fd84b155e98e3"]]},{"id":"35693bd07e359962","type":"change","z":"24d2bae83e216617","name":"Ack","rules":[{"t":"set","p":"receive","pt":"flow","to":"true","tot":"bool"}],"action":"","property":"","from":"","to":"","reg":false,"x":550,"y":620,"wires":[["ca9346ca9b0d0fd0"]]},{"id":"087f5335b1141ede","type":"trigger","z":"24d2bae83e216617","name":"","op1":"","op2":"","op1type":"nul","op2type":"date","duration":"15","extend":true,"overrideDelay":false,"units":"min","reset":"","bytopic":"all","topic":"topic","outputs":1,"x":400,"y":540,"wires":[["a2a2e5a31cabfd11"]]},{"id":"a2a2e5a31cabfd11","type":"change","z":"24d2bae83e216617","name":"set unresponsive","rules":[{"t":"set","p":"unresponsive","pt":"flow","to":"true","tot":"bool"}],"action":"","property":"","from":"","to":"","reg":false,"x":770,"y":540,"wires":[[]]},{"id":"c98d6965d5cfdaec","type":"mqtt out","z":"24d2bae83e216617","name":"","topic":"wican/dc5475934095/can/rx","qos":"","retain":"false","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"54787fa7a5e2bbba","x":1140,"y":260,"wires":[]},{"id":"11faa238a6eeee17","type":"change","z":"24d2bae83e216617","name":"initialize responsive","rules":[{"t":"set","p":"unresponsive","pt":"flow","to":"false","tot":"bool"},{"t":"set","p":"payload","pt":"msg","to":"{}","tot":"json"}],"action":"","property":"","from":"","to":"","reg":false,"x":750,"y":260,"wires":[["c98d6965d5cfdaec","c88fd84b155e98e3"]]},{"id":"5d4aa57b53fd37d2","type":"switch","z":"24d2bae83e216617","name":"if online","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"online","vt":"str"}],"checkall":"true","repair":false,"outputs":1,"x":540,"y":260,"wires":[["11faa238a6eeee17"]]},{"id":"c88fd84b155e98e3","type":"function","z":"24d2bae83e216617","name":"query SOC","func":"msg.payload = \"\";\n\nif (flow.get(\"wicanStatus\") === \"online\" && flow.get(\"unresponsive\") === false) {\n    msg.payload = {\"bus\":\"0\",\"type\":\"tx\",\"ts\":59058,\"frame\":[{\"id\":416957681,\"dlc\":8,\"rtr\":false,\"extd\":true,\"data\":[3,34,160,16,170,170,170,170]}]};\n    flow.set(\"receive\", false);\n}\n\nreturn msg;","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":550,"y":380,"wires":[["9c5cedc68658d173"]]},{"id":"9c5cedc68658d173","type":"switch","z":"24d2bae83e216617","name":"hasPayload","property":"payload","propertyType":"msg","rules":[{"t":"nempty"}],"checkall":"true","repair":false,"outputs":1,"x":730,"y":380,"wires":[["f786585b28a73fd5","fae9b114cbf2c0e9"]]},{"id":"fae9b114cbf2c0e9","type":"trigger","z":"24d2bae83e216617","name":"","op1":"","op2":"","op1type":"nul","op2type":"payl","duration":"5","extend":false,"overrideDelay":false,"units":"s","reset": [... cut ...] ba","type":"mqtt-broker","name":"home","broker":"mqtt.tsteger.org","port":"1883","clientid":"","autoConnect":true,"usetls":false,"protocolVersion":"4","keepalive":"60","cleansession":true,"autoUnsubscribe":true,"birthTopic":"","birthQos":"0","birthRetain":"false","birthPayload":"","birthMsg":{},"closeTopic":"","closeQos":"0","closeRetain":"false","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willRetain":"false","willPayload":"","willMsg":{},"userProps":"","sessionExpiry":""}]
```

DerHerrW, 2024-05-05:
> When requesting the SoC values car scanner sends the following additional messages
> {"bus":"0","type":"tx","ts":59058,"frame":[{"id":416957681,"dlc":8,"rtr":false,"extd":true,"data":[3,34,160,16,170,170,170,170]}]}

This here:
 
> {"bus":"0","type":"tx","ts":38212,"frame":[{"id":416957681,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}

is a "continue" command - a request to continue with data after receiving the first message of the answer. The answer

> {"bus":"0","type":"rx","ts":57681,"frame":[{"id":417001796,"dlc":8,"rtr":false,"extd":true,"data":[16,9,98,160,16,172,173,172]}]}

says it is a first frame (Bit 4 of Byte 0 is 1) of a multi part message and the message contains a total of 9 data bytes (Bits 3..0 of byte 0 and byte 1 form a 12 bit length information). 98,160,16 is the answered request (34, offset +64 is added to the answer).
After receiving the first part car scanner  sends a "continue" request [48,0,10,170,170,170,170,170]. I think the second part of the message was also sent but not recognized by you.

puni2k, 2024-05-06:
Thanks for the clarification. I might want to have a second look at it when I find some time.

puni2k, 2024-05-31:
Just stumbled upon this:
https://gist.github.com/MrPoofy/8450b725ec45988c27e5e59bb99be383

puni2k, 2024-06-01:
I reformatted the data that was kindly provided by @MrPoofy and upload them to this spreadsheet:
https://docs.google.com/spreadsheets/d/1JGLoPidFT_5bXdlT0SeV3P_QbHm-R1WkVDd2DqIv4vs

F1DD (Frame 1, Data Byte D), F1DF, F2DA, F2DB, F2DC all seem to represent Real SOC about equally. 
Note: 
F1DD is data[5] in the wican response in frame 1
F1DF is data[7] in the wican response for frame 1
F2DA is data[2] in the wican response for frame 2
...

Also from the data I managed to come up with a formula to convert the Real SOC value to one that should match the one on the dash.
First you convert your choosen HEX value to a percent value via this formula 
`Dx / 255 * 100`
You then take the result and put it into this
`MIN(100; [Percent] - (( 40 - [PERCENT] ) / 7))`

With 40 and 7 being magical numbers. 40 because it seems to be the point where Real and dash SOC are equal, and 7 because of try-and-error until the lines in the graph matched.

I updated my Node-RED flow to make us of this and removed all the Skoda Citigo stuff that doesn't seem to be neccessary for the 500e
```
[{"id":"a5e59add4bb83757","type":"tab","label":"Wican 500e Test","disabled":false,"info":"","env":[]},{"id":"c4b58391981dd8e5","type":"mqtt in","z":"a5e59add4bb83757","name":"","topic":"wican/dc5475934095/can/status","qos":"2","datatype":"json","broker":"54787fa7a5e2bbba","nl":false,"rap":true,"rh":0,"inputs":0,"x":150,"y":60,"wires":[["d9db4d59ec0ee798"]]},{"id":"fcd7fa4ab088e265","type":"inject","z":"a5e59add4bb83757","name":"Timer","props":[],"repeat":"30","crontab":"","once":false,"onceDelay":"","topic":"","x":100,"y":180,"wires":[["1a86a7d1a3496857"]]},{"id":"271cdc051c767678","type":"mqtt out","z":"a5e59add4bb83757","name":"","topic":"wican/dc5475934095/status_retain","qos":"","retain":"true","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"54787fa7a5e2bbba","x":1400,"y":60,"wires":[]},{"id":"9a4eede45bed8871","type":"mqtt out","z":"a5e59add4bb83757","name":"","topic":"wican/dc5475934095/can/tx","qos":"","retain":"false","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"54787fa7a5e2bbba","x":1340,"y":180,"wires":[]},{"id":"278a7cf05feb346e","type":"mqtt in","z":"a5e59add4bb83757","name":"","topic":"wican/dc5475934095/can/rx","qos":"2","datatype":"json","broker":"54787fa7a5e2bbba","nl":false,"rap":true,"rh":0,"inputs":0,"x":140,"y":420,"wires":[["7d670ef6d4dca3b1"]]},{"id":"73c5c30ac57a2184","type":"mqtt out","z":"a5e59add4bb83757","name":"","topic":"wican/dc5475934095/SOC","qos":"","retain":"true","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"54787fa7a5e2bbba","x":1600,"y":440,"wires":[]},{"id":"f8238d73d71d774c","type":"change","z":"a5e59add4bb83757","name":"extract status","rules":[{"t":"set","p":"wicanStatus","pt":"flow","to":"payload.status","tot":"msg"},{"t":"set","p":"payload","pt":"msg","to":"payload.status","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":560,"y":60,"wires":[["271cdc051c767678"]]},{"id":"d9db4d59ec0ee798","type":"trigger","z":"a5e59add4bb83757","name":"debounce","op1":"","op2":"","op1type":"nul","op2type":"payl","duration":"10","extend":true,"overrideDelay":false,"units":"s","reset":"","bytopic":"all","topic":"topic","outputs":1,"x":380,"y":60,"wires":[["f8238d73d71d774c"]]},{"id":"1a86a7d1a3496857","type":"function","z":"a5e59add4bb83757","name":"query SOC","func":"msg.payload = \"\";\n\nif (flow.get(\"wicanStatus\") === \"online\") {\n    msg.payload = {\"bus\":\"0\",\"type\":\"tx\",\"ts\":59058,\"frame\":[{\"id\":416957681,\"dlc\":8,\"rtr\":false,\"extd\":true,\"data\":[3,34,160,16,170,170,170,170]}]};\n}\n\nreturn msg;","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":570,"y":180,"wires":[["9a4eede45bed8871"]]},{"id":"7d670ef6d4dca3b1","type":"switch","z":"a5e59add4bb83757","name":"not empty","property":"payload","propertyType":"msg","rules":[{"t":"nempty"}],"checkall":"true","repair":false,"outputs":1,"x":360,"y":420,"wires":[["f889b6938c144cc0","ab15d387e5d78037"]]},{"id":"f889b6938c144cc0","type":"debug","z":"a5e59add4bb83757","name":"debug 18","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":560,"y":340,"wires":[]},{"id":"3d9f313a2594cebb","type":"trigger","z":"a5e59add4bb83757","name":"","op1":"","op2":"","op1type":"pay","op2type":"nul","duration":"5","extend":false,"overrideDelay":false,"units":"min","reset":"","bytopic":"all","topic":"topic","outputs":1,"x":1370,"y":440,"wires":[["73c5c30ac57a2184"]]},{"id":"ab15d387e5d78037","type":"switch","z":"a5e59add4bb83757","name":"SOC response","property":"payload.frame[0].id","propertyType":"msg","rules":[{"t":"eq","v":"417001796","vt":"num"}],"checkall":"true","repair":false,"outputs":1,"x":540,"y":420,"wires":[["b628a14d3822e34f"]]},{"id":"b628a14d3822e34f","type":"function","z":"a5e59add4bb83757","name":"get frameid","func":"msg.frameid = msg.payload.frame[0].data[0] >> 4;\n\nreturn msg;","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":730,"y":420,"wires":[["8345bdbba876b380"]]},{"id":"59d4131ec0321f78","type":"rbe","z":"a5e59add4bb83757","name":"","func":"rbe","gap":"","start":"","inout":"out","septopics":true,"property":"payload","topi":"topic","x":1210,"y":440,"wires":[["3d9f313a2594cebb"]]},{"id":"d36e41c1285517d6","type":"function","z":"a5e59add4bb83757","name":"CAN continue", [... cut ...] cleansession":true,"autoUnsubscribe":true,"birthTopic":"","birthQos":"0","birthRetain":"false","birthPayload":"","birthMsg":{},"closeTopic":"","closeQos":"0","closeRetain":"false","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willRetain":"false","willPayload":"","willMsg":{},"userProps":"","sessionExpiry":""}]
```

Much more compact.
Also note: Query time is now down to thirty seconds, but MQTT update interval is limited to 5 minutes. Fits my use-case best, your mileage may vary.

DerHerrW, 2024-06-07:
Maybe FIDD...F2DA are SoC values for 4 Battery blocks...

for soc_helper I will use the formula min(Dx*0.45-6.4;100)

puni2k, 2024-06-07:
From my understanding the small battery consists of 5 packs and the large one of 9

https://www.researchgate.net/figure/Stellantis-Fiat-500e-MY-2022-Low-Range-LR-version-Samsung-SDI-battery-pack-before_fig1_372262493
https://evclinic.eu/2022/11/06/fiat-e500-2022-42kwh-battery/

MrPoofy, 2024-06-07:
And the readings that I did on the file that @puni2k shared is for the big battery one (42kWh)

swissrene, 2025-09-13:
Hallo

Der Thread ist zwar schon etwas älter aber ich versuche trotzdem mal ihn zu reaktivieren. 

Inzwischen habe ich es geschafft, den SOC meines Fiat 500e zu lesen. Massgebend sind die ersten 3 Positionen im data-array. Wenn die 16,9,98 aufweisen, dann ist die letzte Position der SOC. Will man den Wert in % dann berechnet man ihn so:

data[7] * 100 / 255

In folgendem Beispiel also 71*100/255= 27.8%

`{"bus":"0","type":"rx","ts":50272,"frame":[{"id":417001796,"dlc":8,"rtr":false,"extd":true,"data":[16,9,98,160,16,74,74,71]}]}`

Dies habe ich mehrmals mit verschiedenen Werten geprüft. Car Scanner zeigt das gleiche, das Auto stets ca. 2 Prozentpunkte weniger.

Das senden der Daten kann ich aber nicht selbst via mqtt triggern. Nur wenn ich Car Scanner verwende wird der Wert auch gesendet.

Was muss ich via mqtt senden, um den SOC zu erhalten?

Grüsse
René

puni2k, 2025-09-13:
> 
> Was muss ich via mqtt senden, um den SOC zu erhalten?
> 
> Grüsse René

Ich schick immer das JSON Objekt:
```
{"bus":"0","type":"tx","ts":59058,"frame":[{"id":416957681,"dlc":8,"rtr":false,"extd":true,"data":[3,34,160,16,170,170,170,170]}]}
```

Wenn du dir meinen Kommentar weiter oben anschaust findest du auch eine andere Formel die weitestgehend der SOC Anzeige des Autos entspricht. Weicht nur gelegentlich um einen Prozentpunkt ab.
https://github.com/meatpiHQ/wican-fw/issues/95#issuecomment-2143365965