meatpiHQ/wican-fw #96: Inconsistent battery alerts

Byter09, 2024-02-26:
Hi!

I originally set up the WiCAN to sleep at about 12.8V, so its only active when the alternator is running. Additionally, I set the battery alert to go off at 11.8V. However, I started to receive MQTT messages saying the voltage is at 12.4V. Which is weird. I mean the reading was correct but why would I get that message if the voltage isn't below the alert value?

For now, I have disabled the alert and set the sleep to 12.3V and I use HTTP polling every 10 minutes to get the voltage instead. Probably a bit worse in terms of energy usage, but I guess I get reports of its status more consistently, so I can charge it if necessary.

Anyway, I was curious why this behaviour differs from what the readme says. Does the battery alert just send you the battery level in the configured interval, regardless of actual voltage? Preferably I'd use the first described setup and get a value every hour over MQTT, and the remaining time the device sleeps. I have not yet played with actual requests on the CAN to get back the battery voltage or so, which is why I was hoping this would work as described.

Anyway, device works as expected. I barely drive so getting my current battery voltage is very nice. I will see how long the battery lasts with this active polling method I'm using right now.

meatpiHQ, 2024-02-29:
@Byter09 sounds like a bug. I will look into it.

> Does the battery alert just send you the battery level in the configured interval, regardless of actual voltage?

No, only if the voltage is lower than the threshold.

Byter09, 2024-02-29:
In the meantime I changed it to trigger below 15V to "abuse" that system to get voltage reports. Works as expected. So no clue why I got reports with values higher than the threshold on my first day of using.

Byter09, 2024-03-01:
Okay so the above also does not seem to work. Sleep is at 12.6V, alert is set to 15V. I do get the online and offline status message over MQTT, but no voltage reports. Honestly I'm so confused why. I swear this worked for a day but then suddenly stopped the moment it was out of reach of the WiFi for a day. After it was in range again I never received another message.

Oh and on a side note. I set the channel to 14, which seems not to be available in Germany or something and managed to brick it. Had to re-flash, so I can confirm this behaviour even after completely erasing all of the memory on the chip.

meatpiHQ, 2024-03-02:
@Byter09 Do you know if its connected to wifi but no messages are coming in? 

> Oh and on a side note. I set the channel to 14, which seems not to be available in Germany or something and managed to brick it. Had to re-flash, so I can confirm this behaviour even after completely erasing all of the memory on the chip.

That's interesting, so you were unable to connect to the AP after setting it to channel 14?

Byter09, 2024-03-02:
Regarding the channel:
[image]
I don't see any other WiFi in the area actually use it. I've set my stuff to "auto" so I'd expect it to use that one if free, but looking at the chart nobody does. I assume it's some kind of channel restriction based on location. I don't really know. When I selected channel 14 on the WiCAN it booted back up and was blinking like every second using the green and orange LEDs, but never getting to a constant blue.

And for the connection of the WiCAN, I can see DHCP leases going out:
[image]
Mind you everything on the first of March is me setting stuff up again, and then if becomes not really frequent but more or less every two hours or so?

Mind you my car is far away from my WiFi and it can barely connect to it. Like 20 to 40m if I had to guess. I'm on the third floor and the car is parked on the other side of the road directly in front of my window. But this worked flawlessly a while ago when it was even further away.

Oh and I should note that the DHCP request at 9:05 correlates with a voltage alert, but no online/offline status update. That is the first and only one since plugging it in yesterday. So it seems they come really infrequently, and maybe it's the bad WiFi connection. I am not sure how to debug this.

meatpiHQ, 2024-03-07:
@Byter09 It's very likely it's bad WiFi signal.

> When I selected channel 14 on the WiCAN it booted back up and was blinking like every second using the green and orange LEDs, but never getting to a constant blue.

This also looks like a bug.. seems like you have a talent in finding bugs :) . I will look into this also.

I'm working on improving the battery voltage reading, I will send you a new firmware to try in the next few days.

Byter09, 2024-03-07:
Awesome. No worries. Love the WiCAN. Worth every penny so far :D