meatpiHQ/wican-fw #97: Reading from Honda Clarity Plugin

alexsahka, 2024-03-01:
I completed everything per Wiki: https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

Car Scanner set to read SOC Voltage

{"bus":"0","type":"tx","ts":29165,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,11,170,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":29098,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,7,170,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":29022,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,6,170,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":23895,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,0,170,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":23153,"frame":[{"id":417018865,"dlc":8,"rtr":false,"extd":true,"data":[2,1,0,170,170,170,170,170]}]}

{"bus":"0","type":"rx","ts":23907,"frame":[{"id":417001750,"dlc":8,"rtr":false,"extd":true,"data":[6,65,0,152,24,128,3,85]}]}
{"bus":"0","type":"rx","ts":23906,"frame":[{"id":417001770,"dlc":8,"rtr":false,"extd":true,"data":[6,65,0,152,24,128,1,85]}]}
{"bus":"0","type":"rx","ts":23905,"frame":[{"id":417001749,"dlc":8,"rtr":false,"extd":true,"data":[6,65,0,152,24,128,3,85]}]}
{"bus":"0","type":"rx","ts":23905,"frame":[{"id":417001749,"dlc":8,"rtr":false,"extd":true,"data":[6,65,0,152,24,128,3,85]}]}
{"bus":"0","type":"rx","ts":23898,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[6,65,0,152,24,128,3,85]}]}
{"bus":"0","type":"rx","ts":23160,"frame":[{"id":417001742,"dlc":8,"rtr":false,"extd":true,"data":[6,65,0,182,60,168,19,85]}]}
{"bus":"0","type":"rx","ts":23159,"frame":[{"id":417001730,"dlc":8,"rtr":false,"extd":true,"data":[6,65,0,152,24,128,3,85]}]}
{"bus":"0","type":"rx","ts":23157,"frame":[{"id":417001734,"dlc":8,"rtr":false,"extd":true,"data":[6,65,0,152,24,128,3,85]}]}

Car Scanner set to read Odometer.

{"bus":"0","type":"tx","ts":1916,"frame":[{"id":416964849,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":1907,"frame":[{"id":416964849,"dlc":8,"rtr":false,"extd":true,"data":[3,34,112,34,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":1686,"frame":[{"id":416964849,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}

{"bus":"0","type":"rx","ts":1684,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[16,57,98,112,34,236,0,0]}]}
{"bus":"0","type":"rx","ts":1606,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[40,0,0,85,85,85,85,85]}]}
{"bus":"0","type":"rx","ts":1594,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[39,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":1575,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[37,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":1534,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[33,0,0,0,1,100,236,0]}]}
{"bus":"0","type":"rx","ts":41601,"frame":[{"id":417001824,"dlc":8,"rtr":false,"extd":true,"data":[16,57,98,112,34,236,0,0]}]}

How do I convert this data into a human-readable format?

I spent the past week trying to find a solution, but I'm still unsure what to do with this data. Can anyone guide me in the right direction?

meatpiHQ, 2024-03-02:
@alexsahka 

Your mqtt log should look like this, tx then rx frame. It would be great if you can post more mqtt logs for the SoC, and what is the SoC showing in car scanner, so we can try and guess how it's calculated, 
[image]

alexsahka, 2024-03-02:
> @alexsahka
> 
> Your mqtt log should look like this, tx then rx frame. It would be great if you can post more mqtt logs for the SoC, and what is the SoC showing in car scanner, so we can try and guess how it's calculated, <img alt="image" width="739" src="https://private-user-images.githubusercontent.com/94690098/309471086-2a176e43-d58d-4225-8af7-c0ca6b20c8b6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk0MDI2NzAsIm5iZiI6MTcwOTQwMjM3MCwicGF0aCI6Ii85NDY5MDA5OC8zMDk0NzEwODYtMmExNzZlNDMtZDU4ZC00MjI1LThhZjctYzBjYTZiMjBjOGI2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzAyVDE3NTkzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5OTU1MGMyZWE3ZmY3OWNhYWIzYTQ0YzdmYjUxZDFiNzg2N2QwYTM5MTliY2M1NTkwZjA0MGQ3OGExOTgxZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OUDhoR8rlHRaM053y2F6hU7xqVNWh_H_wsOjENJTTVI">

There isn't always a TX frame followed immediately by an RX frame. Additionally, I'm seeing a lot of frames in MQTT Explorer (http://mqtt-explorer.com/). How can I capture these frames, and is there a limit on the number I can capture?

meatpiHQ, 2024-03-03:
@alexsahka When you're logging from carscanner do you click on the SoC? 

When you click on the SoC or other sensors in carscanner, it will only request that specific sensor.

alexsahka, 2024-03-04:
While I stay on carscanner SoC sensor only
[image]

the MQTT topic wican/f412fa33xxxx/can/rx  got bombarded with 4-5 messages per second, question what im doing not right?
[image]

[image]

alexsahka, 2024-03-06:
> @alexsahka When you're logging from carscanner do you click on the SoC?
> 
> When you click on the SoC or other sensors in carscanner, it will only request that specific sensor.

When using only the carscanner SoC sensor, the MQTT topic "wican/f412fa33xxxx/can/rx" receives a high volume of messages (multiple per second). Is this behavior expected, or is there an issue with my setup?
[image]

[image]

[image]

alexsahka, 2024-03-07:
> @alexsahka When you're logging from carscanner do you click on the SoC?
> 
> When you click on the SoC or other sensors in carscanner, it will only request that specific sensor.

If we're unable to resolve this simple question regarding product support, I'll need to return the three MeatPis units to Mouser due to their lack of support availability. Thanks anyway, good looking product.

meatpiHQ, 2024-03-07:
> When using only the carscanner SoC sensor, the MQTT topic "wican/f412fa33xxxx/can/rx" receives a high volume of messages (multiple per second).

Yes that's normal. On some car's you might see a multi-line response for a PID request. If you post the frame sequence here i can help you look analyze it.

https://github.com/meatpiHQ/wican-fw/discussions/86
[image]

> If we're unable to resolve this simple question regarding product support

For your reference it usually takes me 24 to 72 hours to respond to issues, I try to respond to every user to my best ability.

> I'll need to return the x3 MeatPis units to Mouser due to their lack of support availability.

You can reach out to Mouser customer service and they will help you out.

alexsahka, 2024-03-07:
> > When using only the carscanner SoC sensor, the MQTT topic "wican/f412fa33xxxx/can/rx" receives a high volume of messages (multiple per second).
> 
> Yes that's normal. On some car's you might see a multi-line response for a PID request. If you post the frame sequence here i can help you look analyze it.
> 
> #86 <img alt="image" width="845" src="https://private-user-images.githubusercontent.com/94690098/310857625-fd93b8a4-2b84-41b1-9a23-592968dd9aa7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk4NTA1NjAsIm5iZiI6MTcwOTg1MDI2MCwicGF0aCI6Ii85NDY5MDA5OC8zMTA4NTc2MjUtZmQ5M2I4YTQtMmI4NC00MWIxLTlhMjMtNTkyOTY4ZGQ5YWE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA3VDIyMjQyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYxZjM3M2Y1NDIxZWE0ZDhmNWVjZTJmZjljNDgyNzYxMzU3MTFhMmQ0N2M1Yzg1OWQ0ZjU2NmZmOTIxZmM0ODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bdA2M7pVTK2Yi4UxrZAJp7EAVpuDNDIZIiWK1qwlxVI">
> 
> > If we're unable to resolve this simple question regarding product support
> 
> For your reference it usually takes me 24 to 72 hours to respond to issues, I try to respond to every user to my best ability.
> 
> > I'll need to return the x3 MeatPis units to Mouser due to their lack of support availability.
> 
> You can reach out to Mouser customer service and they will help you out.

Ok, my bad! I finally figured out how to capture the MQTT data stream.

Here SOC data:

{"bus":"0","type":"tx","ts":46533,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[3,34,32,42,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":46536,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[16,246,98,32,42,255,255,63]}]}
{"bus":"0","type":"tx","ts":46538,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":46540,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[33,255,255,255,255,255,255,231]}]}
{"bus":"0","type":"rx","ts":46543,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[34,255,252,3,255,242,179,63]}]}
{"bus":"0","type":"rx","ts":46545,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[35,255,51,51,51,51,255,225]}]}
{"bus":"0","type":"rx","ts":46547,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[36,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":46549,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[37,0,50,0,45,0,0,93]}]}
{"bus":"0","type":"rx","ts":46551,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[38,204,21,58,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":46554,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[39,129,0,193,0,3,95,32]}]}
{"bus":"0","type":"rx","ts":46556,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[40,83,0,23,26,134,0,31]}]}
{"bus":"0","type":"rx","ts":46558,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[41,7,6,26,132,255,196,21]}]}
{"bus":"0","type":"rx","ts":46559,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[42,18,254,162,49,254,252,36]}]}
{"bus":"0","type":"rx","ts":46560,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[43,36,76,36,25,78,252,78]}]}
{"bus":"0","type":"rx","ts":46561,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[44,227,0,0,109,196,0,0]}]}
{"bus":"0","type":"rx","ts":46562,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[45,109,196,1,32,2,5,170]}]}
{"bus":"0","type":"rx","ts":46564,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[46,0,120,0,80,255,28,3]}]}
{"bus":"0","type":"rx","ts":46566,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[47,2,0,0,32,12,55,100]}]}
{"bus":"0","type":"rx","ts":46567,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[32,56,194,49,1,243,255,251]}]}
{"bus":"0","type":"rx","ts":46568,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[33,0,31,25,25,3,131,0]}]}
{"bus":"0","type":"rx","ts":46570,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[34,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":46572,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[35,26,167,17,0,39,16,56]}]}
{"bus":"0","type":"rx","ts":46574,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[36,14,127,0,127,0,127,0]}]}
{"bus":"0","type":"rx","ts":46575,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[37,0,0,1,0,1,0,39]}]}
{"bus":"0","type":"rx","ts":46576,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[38,16,0,0,1,244,0,0]}]}
{"bus":"0","type":"rx","ts":46577,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[39,246,0,240,12,99,0,8]}]}
{"bus":"0","type":"rx","ts":46578,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[40,155,0,14,18,118,0,72]}]}
{"bus":"0","type":"rx","ts":46579,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[41,0,0,0,65,0,0,26]}]}
{"bus":"0","type":"rx","ts":46580,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[42,134,0,0,0,37,0,0]}]}
{"bus":"0","type":"rx","ts":46582,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[43,78,255,0,0,78,227,0]}]}
[... cut ...]
{"bus":"0","type":"rx","ts":46587,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[47,1,4,46,4,46,0,0]}]}
{"bus":"0","type":"rx","ts":46588,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[32,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":46590,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[33,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":46591,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[34,0,0,0,0,0,0,0]}]}
{"bus":"0","type":"rx","ts":46592,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[35,0,0,85,85,85,85,85]}]}

meatpiHQ, 2024-03-08:
@alexsahka there's a lot of Rx frame, as suspected it multi frame response. What was the exact SoC value for this log?

Another thing were you able to find any info on reading the SoC for this car?

alexsahka, 2024-03-08:
> @alexsahka there's a lot of Rx frame, as suspected it multi frame response. What was the exact SoC value for this log?
> 
> Another thing were you able to find any info on reading the SoC for this car?

While experimenting with the ignition ON, I tried to capture the MQTT data stream. Unfortunately, I don't remember the exact number, but it should be between 96 and 99. I searched for information on how to read the SoC on the Honda Clarity PHEV, but haven't found anything yet.

meatpiHQ, 2024-03-09:
@alexsahka you should record the exact SoC while logging, this will help when trying to guess the SoC.

I also found the link below, I haven't gone through it all yet but seems helpful.

https://www.insideevsforum.com/community/index.php?threads/budget-battery-capacity-readout.10531/page-3

alexsahka, 2024-03-09:
> @alexsahka you should record the exact SoC while logging, this will help when trying to guess the SoC.
> 
> I also found the link below, I haven't gone through it all yet but seems helpful.
> 
> https://www.insideevsforum.com/community/index.php?threads/budget-battery-capacity-readout.10531/page-3

Thanks for the link! I actually saw this before, but I missed the part where MrFixit posted a config file for CarScanner. I downloaded that zip file, and inside it, I found a file named "claritypid.csp." I opened it with Notepad++ and found the line with SOC info. Here's the line:

"NM":"[Clarity]SOC","SNM":"SOC","CMD":"22202A","RL":0,"MAX":100.0,"MIN":0.0,"Id":2038,"VIS":true,"SkipCycles":0},{"SBI":104,"DL":1,"MUL":1.0,"DIV":1.0,"OFS":0.0,"SIG":false,"TP":1,"BIT":0,"FHID":false,"RBS":false,"HDR":"DA01F1","FR":"/1#104#1","UN":0,"BCM":"","ACM":"","ACT":false,"TRNS":[]

alexsahka, 2024-03-10:
> @alexsahka When you load claritypid.csp into carscanner are you able to use it to read the SoC?

Three years ago, CarScanner did not support the Honda Clarity PHEV. However, the latest version of CarScanner offers full support for this vehicle.  No claritypid.csp needed.

meatpiHQ, 2024-03-11:
@alexsahka it's weird because I cannot find the CAN IDs in the mqtt log you posted.

alexsahka, 2024-03-11:
What is this then?  "id":416940529   or   "id":417001729
[image]

> @alexsahka it's weird because I cannot find the CAN IDs in the mqtt log you posted.

meatpiHQ, 2024-03-13:
@alexsahka id 416940529 in Hex  is 18DA01F1. I don't see that in the file.

alexsahka, 2024-03-14:
> @alexsahka id 416940529 in Hex is 18DA01F1. I don't see that in the file.

The file contains many hexadecimal numbers that appear similar "HDR":"DA01F1"  , but there seems to be a missing two numbers 18. Are there any conversion methods we can use to fill in the missing value? 

On that forum, MrFixit does some kind of conversion.
https://www.insideevsforum.com/community/index.php?threads/budget-battery-capacity-readout.10531/page-8#post-125374
[image]

meatpiHQ, 2024-03-16:
@alexsahka I think the best way forward is to try crack the car scanner logs. 

Let's try to log 2 readings, at 2 different SoC levels. Note the SoC value you read from car scanner and log the can tx Rx frames like you've done before, do this for both SoC levels.

alexsahka, 2024-03-18:
> @alexsahka I think the best way forward is to try crack the car scanner logs.
> 
> Let's try to log 2 readings, at 2 different SoC levels. Note the SoC value you read from car scanner and log the can tx Rx frames like you've done before, do this for both SoC levels.

Here are 3 different SOC%:

SOC  94%

{"payload":{"bus":"0","type":"tx","ts":39081,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[3,34,32,42,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":39083,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[16,246,98,32,42,255,255,63]}]}}
{"payload":{"bus":"0","type":"tx","ts":39085,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":39089,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[33,255,255,255,255,255,255,231]}]}}
{"payload":{"bus":"0","type":"rx","ts":39091,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[34,255,252,3,255,242,179,63]}]}}
{"payload":{"bus":"0","type":"rx","ts":39092,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[35,255,51,51,51,51,255,225]}]}}
{"payload":{"bus":"0","type":"rx","ts":39094,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[36,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39095,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[37,0,68,0,45,0,0,94]}]}}
{"payload":{"bus":"0","type":"rx","ts":39096,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[38,174,11,224,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39097,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[39,129,0,193,0,3,94,32]}]}}
{"payload":{"bus":"0","type":"rx","ts":39099,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[40,81,0,163,26,101,1,27]}]}}
{"payload":{"bus":"0","type":"rx","ts":39101,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[41,19,18,26,94,255,196,22]}]}}
{"payload":{"bus":"0","type":"rx","ts":39103,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[42,180,253,2,50,28,250,160]}]}}
{"payload":{"bus":"0","type":"rx","ts":39104,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[43,35,192,35,147,78,155,78]}]}}
{"payload":{"bus":"0","type":"rx","ts":39106,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[44,127,0,0,76,5,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39107,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[45,76,5,1,32,6,5,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":39108,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[46,1,34,0,180,255,28,3]}]}}
{"payload":{"bus":"0","type":"rx","ts":39109,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[47,2,0,0,21,25,55,200]}]}}
{"payload":{"bus":"0","type":"rx","ts":39110,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[32,57,38,49,1,243,1,16]}]}}
{"payload":{"bus":"0","type":"rx","ts":39111,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[33,1,27,25,25,3,131,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39113,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[34,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39114,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[35,26,131,17,0,39,16,56]}]}}
{"payload":{"bus":"0","type":"rx","ts":39115,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[36,134,127,0,127,0,127,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39116,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[37,0,0,1,0,1,0,39]}]}}
{"payload":{"bus":"0","type":"rx","ts":39117,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[38,16,0,0,1,244,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39118,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[39,246,0,240,12,99,0,8]}]}}
{"payload":{"bus":"0","type":"rx","ts":39119,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[40,207,0,96,18,118,0,185]}]}}
{"payload":{"bus":"0","type":"rx","ts":39120,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[41,0,0,0,176,0,0,26]}]}}
{"payload":{"bus":"0","type":"rx","ts":39121,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[42,100,0,0,0,249,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39122,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[43,78,156,0,0,78,127,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":39124,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[44,0,35,209,0,0,35,155]}]}}

SOC  96%

{"payload":{"bus":"0","type":"tx","ts":49627,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[3,34,32,42,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":49630,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[16,246,98,32,42,255,255,63]}]}}
{"payload":{"bus":"0","type":"tx","ts":49632,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}
{"payload":{"bus":"0","type":"rx","ts":49635,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[33,255,255,255,255,255,255,231]}]}}
{"payload":{"bus":"0","type":"rx","ts":49636,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[34,255,252,3,255,242,179,63]}]}}
{"payload":{"bus":"0","type":"rx","ts":49638,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[35,255,51,51,51,51,255,225]}]}}
[... cut ...]
{"payload":{"bus":"0","type":"rx","ts":46374,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[32,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":46375,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[33,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":46376,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[34,0,0,0,0,0,0,0]}]}}
{"payload":{"bus":"0","type":"rx","ts":46377,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[35,0,0,85,85,85,85,85]}]}}

alexsahka, 2024-03-22:
Could you help me further?
 I sending 2 messages to wican/device/can/tx
msg.payload = { "bus": "0", "type": "tx", "frame": [{ "id": 416940529, "dlc": 8, "rtr": false, "extd": true, "data": [3, 34, 32, 42, 170, 170, 170, 170] }] };

delay 200ms

msg.payload = { "bus": "0", "type": "tx", "frame": [{ "id": 416940529, "dlc": 8, "rtr": false, "extd": true, "data": [48, 0, 10, 170, 170, 170, 170, 170] }] };

and nothing rx back to MQTT, what im doing wrong?
Here is screen shoot for settings:
[image]

meatpiHQ, 2024-03-24:
@alexsahka Do you get a response when the car is running?

Can you try to leave the car open while it charging see if the ECU goes to sleep?

alexsahka, 2024-03-24:
> @alexsahka Do you get a response when the car is running?
> 
> Can you try to leave the car open while it charging see if the ECU goes to sleep?

Sorry for the confusion. I actually had two questions combined in one. Please disregard the question about how to wake the CAN bus. All the tests I performed were done with the ignition ON.

meatpiHQ, 2024-03-26:
@alexsahka so if i understand correctly, the issue here is that you're not getting any rx data?

Can you please try to remove all the filters, and just keep the first one?

[image]

alexsahka, 2024-04-30:
> {"bus":"0","type":"tx","ts":39085,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}

Posting to:
wican/xxxxxxx/can/tx

This payloads:

msg.payload = { "bus": "0", "type": "tx", "frame": [{ "id": 416940529, "dlc": 8, "rtr": false, "extd": false, "data": [3, 34, 32, 42, 170, 170, 170, 170] }] };

or:

msg.payload = { "bus": "0", "type": "tx", "frame": [{ "id": 416940529, "dlc": 8, "rtr": false, "extd": false, "data": [48, 0, 10, 170, 170, 170, 170, 170] }] };

Nothing received back
Ignition is ON.

alexsahka, 2024-04-30:
> @alexsahka so if i understand correctly, the issue here is that you're not getting any rx data?
> 
> Can you please try to remove all the filters, and just keep the first one?
> 
> <img alt="image" width="358" src="https://private-user-images.githubusercontent.com/94690098/316852175-37f0b40d-79c8-4c94-97e4-4deb829a777e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0NjI2MDAsIm5iZiI6MTcxNDQ2MjMwMCwicGF0aCI6Ii85NDY5MDA5OC8zMTY4NTIxNzUtMzdmMGI0MGQtNzljOC00Yzk0LTk3ZTQtNGRlYjgyOWE3NzdlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDMwVDA3MzE0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFkODFmMDhiNTA4Y2U2YzlmNGI0NWNkNWZiN2JhMjJlYWU4MTc0MWQ2ZmVhYzY4ZmRhZTZkNGNjYjM1ZjZjZWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dX3WuRfWuSxY_NSEdC4-up70YLUa1G9lKyN0sKberyw">

Posting to:
wican/xxxxxxx/can/tx

This payloads:

msg.payload = { "bus": "0", "type": "tx", "frame": [{ "id": 416940529, "dlc": 8, "rtr": false, "extd": false, "data": [3, 34, 32, 42, 170, 170, 170, 170] }] };

or:

msg.payload = { "bus": "0", "type": "tx", "frame": [{ "id": 416940529, "dlc": 8, "rtr": false, "extd": false, "data": [48, 0, 10, 170, 170, 170, 170, 170] }] };

Nothing received back
Ignition is ON.

alexsahka, 2024-05-02:
> @alexsahka so if i understand correctly, the issue here is that you're not getting any rx data?
>
> Can you please try to remove all the filters, and just keep the first one?
>
> <img alt="image" [...]

Can you help me with the issue of not receiving any RX back to MQTT? If I can gain traction with this, I'll be able to understand how to get other sensors to read in the future. Is the PID 39 is correct one?

meatpiHQ, 2024-05-02:
@alexsahka I suggest that you remove all filters first. And test with some standard PID like Engine speed. https://www.csselectronics.com/pages/obd2-pid-table-on-board-diagnostics-j1979

Note that if you have 1 filter added, all other CAN IDs will be disabled.

alexsahka, 2024-05-02:
> @alexsahka I suggest that you remove all filters first. And test with some standard PID like Engine speed. https://www.csselectronics.com/pages/obd2-pid-table-on-board-diagnostics-j1979
> 
> Note that if you have 1 filter added, all other CAN IDs will be disabled.

Yes I left the only one with PID 39

meatpiHQ, 2024-05-02:
@alexsahka ok. If you remove the PID 39 filter, do you receive a CAN frame on MQTT?

alexsahka, 2024-05-02:
> @alexsahka ok. If you remove the PID 39 filter, do you receive a CAN frame on MQTT?

Thanks, I learn the troubleshooting step!
[image]

meatpiHQ, 2024-05-02:
@alexsahka try this

1-  Send this frame first
> {"payload":{"bus":"0","type":"tx","ts":49627,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[3,34,32,42,170,170,170,170]}]}}

2- wait for this frame:
> {"payload":{"bus":"0","type":"rx","ts":49630,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[16,246,98,32,42,255,255,63]}]}}

3- After you receive the frame in step 2 l, send this frame.
> {"payload":{"bus":"0","type":"tx","ts":49632,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}

alexsahka, 2024-05-02:
> 1- Send this frame first
> 
> > {"payload":{"bus":"0","type":"tx","ts":49627,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[3,34,32,42,170,170,170,170]}]}}
> 
> 2- wait for this frame:
> 
> > {"payload":{"bus":"0","type":"rx","ts":49630,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[16,246,98,32,42,255,255,63]}]}}
> 
> 3- After you receive the frame in step 2, send this frame.
> 
> > {"payload":{"bus":"0","type":"tx","ts":49632,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}

1- Sent this frame 

{"payload":{"bus":"0","type":"tx","ts":49627,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[3,34,32,42,170,170,170,170]}]}}

2- recived this frame:

{"payload":{"bus":"0","type":"rx","ts":49630,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[16,246,98,32,42,255,255,63]}]}}

3- Sent this frame

{"payload":{"bus":"0","type":"tx","ts":49632,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}

4- Nothing recived

meatpiHQ, 2024-05-02:
@alexsahka are you doing this manually or using Home assistant or a script?

Do you send the frame in step 2 without any delay?

If there's a big enough delay the ECU might ignore the frame.

alexsahka, 2024-05-02:
100 msec
[image]

alexsahka, 2024-05-02:
> @alexsahka are you doing this manually or using Home assistant or a script?
> 
> Do you send the frame in step 2 without any delay?
> 
> If there's a big enough delay the ECU might ignore the frame.

Sometimes when I quickly and randomly press the inject node with the 100 ms delay node I can see a very long RX frame.

[image]

alexsahka, 2024-05-02:
> @alexsahka try to use a subscribe node to the Rx topic, and the Rx node should trigger your "TEST2 req"
> 
> This way the 2nd frame is triggered after the first RX, without any delay.

Now it is consistently responding with different long RX frames, similar to what I received during the carscanner capture section.

alexsahka, 2024-05-02:
> @alexsahka try to use a subscribe node to the Rx topic, and the Rx node should trigger your "TEST2 req"
> 
> This way the 2nd frame is triggered after the first RX, without any delay.

But I haven't seen this RX frame starting with 39 yet.
[image]

meatpiHQ, 2024-05-02:
@alexsahka what I meant is that you should make a mqtt subscription node that subscribe to the Rx topic and waits for this frame 

> {"payload":{"bus":"0","type":"rx","ts":49630,"frame":[{"id":417001729,"dlc":8,"rtr":false,"extd":true,"data":[16,246,98,32,42,255,255,63]}]}}

Then the subscription node should trigger another node that sends this frame:

> {"payload":{"bus":"0","type":"tx","ts":49632,"frame":[{"id":416940529,"dlc":8,"rtr":false,"extd":true,"data":[48,0,10,170,170,170,170,170]}]}}

alexsahka, 2024-05-02:
> @alexsahka what I meant is that you should make a mqtt subscription node that subscribe to the Rx topic and waits for this frame
>
>
> Then the subscription node should trigger another node that sends [...]

[image]

alexsahka, 2024-05-02:
> @alexsahka looks good! You need to update to the latest firmware version. There's a new PID index feature. You'll need it for the filter.

Is this correct?
[image]

meatpiHQ, 2024-05-02:
@alexsahka Great! But it's better if you use 

PID: 129
index: 1

alexsahka, 2024-05-02:
> @alexsahka Great! But it's better if you use
> 
> PID: 129 index: 1

Can you explain? Just curious, for future reference.

meatpiHQ, 2024-05-02:
@alexsahka you might want to add another PID request with multiple frame response, the 34 won't be unique. Anyway it's not that critical.

alexsahka, 2024-05-04:
> @alexsahka Do you get a response when the car is running?
> 
> Can you try to leave the car open while it charging see if the ECU goes to sleep?

I left the car unlocked while it was charging, and the ECU is not responding. Is there a way to wake the ECU up?

meatpiHQ, 2024-05-07:
@alexsahka I don't know if there is a way to wake up the ECU. Is there any CAN activity while charging?

alexsahka, 2024-05-12:
> @alexsahka I don't know if there is a way to wake up the ECU. Is there any CAN activity while charging?
Here's an idea: The adapter could record parameters every minute or more frequently. When it disconnects from the ECU, it would then start publishing to MQTT upon connecting to home Wi-Fi.

alexsahka, 2024-05-15:
> @alexsahka I don't know if there is a way to wake up the ECU. Is there any CAN activity while charging?

Any feedback on this approach?

alexsahka, 2024-05-16:
> @alexsahka This is not currently implemented, but that's a good idea I think I will add it in the future. What I think you mean, is that WiCAN will continuously poll the ECU and record the response if it's not connected the WiFi. And once it connect it will upload all the record.

Yes, you understand everything correctly.  This will benefit all cars with ECU  that enter sleep mode immediately after the ignition is turned off.