meatpiHQ/wican-fw #98: SavvyCAN: Getting 3000 frames per second...

keerttttt, 2024-03-02:
Hi. I'm new to WiCAN, i'm trying to sniff CAN data via SavvyCAN. I'm getting 1 message looped, that's the CAN Gateway on my VW E-Golf. So the data seems ok. But, i'm getting 3000 messages per second, it's flooding my SavvyCAN... Is that normal? I Read it should give 1 message like this per 500ms...

Am i doing anything wrong?

keerttttt, 2024-03-02:
[image]
This also shows it. I stopped it after a second or two, the count flys up like crazy.

meatpiHQ, 2024-03-02:
looks like it receiving a message every 5ms, in SavvyCan enable overwrite mode and check the cycle time. 

But it's weird only one CAN ID...

[image]

keerttttt, 2024-03-02:
Thanks for the quick reply. The only one CAN id seems normal, it's the VW CAN Gateway. It should report only 1 message every 500ms. And when i put the car on, it does that. So same message, only then 2 per second.

But, when it's off, and i mean really off, so locked for a few minutes. The message rampages.

keerttttt, 2024-03-03:
Ok, don't know why but I'm getting more normal behavior now. I get silence on the CAN bus If the vehicle is off. When I try to poll a pid I get 10 messages (all the same) back from the gateway, with 500ms interval and then it stops (goes to sleep again?)

Any idea how to query the gateway itself? I'm not really experienced with can, but I would like to try to read all possible values belonging to the Can gateway, which seems to be the only module that is awake when the car is off, and also should store last known values of a lot of car stuff.

Any ideas?

typxxi, 2024-08-07:
Have you found a solution to read the SOC and current range from your car while ignition is turned of ?

I am only able to get the SOC if the ignition is on and the car scanner app is running and i have opened the all sensors menu on the upper right corner and have selected via filter for "state" only and gotten the 3 sensors in an overview .
But only from the second when the value appear in the app behind the sensors mqtt explorer gets the messages.

And that is unuseable state. I have used e-Up settings so far cause I did not find better settings I could use.

thanks