meatpiHQ/wican-fw #103: stationary SOC from Lexus

Sx54A, 2024-03-06:
Hello,

I fail at the very first step of connecting to the existing Wi-Fi. My Wi-Fi password contains numbers, letters, as well as the following special characters: "([*\\. 

What character set is allowed?

Sx54A, 2024-03-07:
pw is 16 characters
ssid is 8 characters

meatpiHQ, 2024-03-07:
@Sx54A works ok for me, I used password: Hello12345"([*\.

How about your ssid anything special there?

Also what is your wifi encryption, WPA3?

meatpiHQ, 2024-03-08:
@Sx54A did you manage to get it to connect?

Sx54A, 2024-03-09:
I checked my connections. Created a new Ad-hoc wifi from my mobile with nearly same ssid & pw like may stationary one. --> wican connected. --> its not about the characters. It must be a problem with my stationary wifi hardware. 

Because I have a direct connection to meatpi now. I will concentrate on get SOC first. When I find out later, why my Fritzbox Wifi make problems and how to solve, I will update.

Sx54A, 2024-03-10:
update: 

station wifi network:
I think the Fritzbox mesh caused connection problems. Now I created a new wifi-network (only for the meatpi)... set meatpi to AP+station and it works.

Car scanner to meatpi connection:
installed car scanner on mobile; 
connected to wifi ap network of meatpi; 
setup car sacnner (192.168.80.1) and port (i my case 3333)
connection profile: Lexus bZ4X/Solterra/Lexus RZ
Press: Connect in the App
When I change view to sensors, I get informations... 
unfortunately SOC and all battery relevant data is (and stays) empty.

But I have a "remaining usage time of hybrid battery"... with percentage (max was 93%....)
I think, this is the SOC.

I setup mqtt but dont know, how I go the last steps:

-> If I have a sensor in "car scanner" like "remaining usage time of hybrid battery" how can I get this information published by mqtt?

meatpiHQ, 2024-03-10:
@Sx54A 

> But I have a "Hybrid/EV battery pack remaining charge"... with percentage (max was 93%....)
> I think, this is the SOC. This value I want to get.

You might be lucky! Try to request PID 91.

[image]

> Now: How to get specific sensor information (car scanner found & displayed) published by Wican?

Pressing on a specific sensor doesn't work?

>  If I have a sensor in "car scanner" like "Hybrid/EV battery pack remaining charge" how can I get this information published by mqtt?

Sorry I'm not sure I understand your question. Like you mentioned enable MQTT elm327 logging...

Sx54A, 2024-03-10:
I'm trying to express myself clearly (not a native English speaker, sorry): I searched for the term "Hybrid/EV battery pack remaining charge" in the OBD2 calculator and got PID91 as a result.
[image]
I now have to somehow process this information and enter it into Wican.
[image]

As I understand it, I enter in the line what WiCan is supposed to SEND. So I take the 'Response' information from the calculator:"
[image]

From that, I now take the '2024'.
Name: I can think of any one. Lets take "Battery"
PID: is 91

Start Bit: how to get this information? (I take 0 because I saw it often)
End Bit:  how to get this information? (I take 1 because I saw it often)

Expression: this is, how to calculate - can I take it from the OBD2Calculator, too? In my case it could be 0+1+18 ... 18 seems to be a value (an example) --> I take: 0+1+V

[image]

With all information I get this:
[image]

And now, either through the MQTT Explorer or via ioBroker or any other home automation system, I can send a message to WiCAN, and WiCAN will return the battery status to me via MQTT, correct? As soon as I have set up WiCAN properly, I'll inquire about how to make that request.

meatpiHQ, 2024-03-10:
@Sx54A 

An example of the mqtt frames is mentioned in the readme: https://github.com/meatpiHQ/wican-fw/tree/3bb50a431f4812e1a0c3c1604e29e036c89245e3?tab=readme-ov-file#6-request-battery-soc-mqtt-example

> Expression: this is, how to calculate - can I take it from the OBD2Calculator, too? In my case it could be 0+1/2.55+12 ... 12 seems to be a value (an example) --> I take: 0+1/2.55+V

I suggest you use B instead of V to reference the Byte, so the expression should be: B3/2.55
If you use B you can ignore the "start bit" and the "bit length"

Sx54A, 2024-03-10:
If I change the letter V into a B I get:  0+1/2.55+B

In your expression, where did you get the "3" from? And how do you know, that it should look like B3/2.55?

Sx54A, 2024-03-10:
Ok, I see my mistake in the calculation.Thanks.

You wrote: "If you want to use V, the you have to set then "start bit: 24 and bit length: 8", then your expression will be V/2.55"

I I try to save the new created line without setting start bit & bit lenght, it doesn`t work --> It look, that I need these startbit&bit lenght.
[image]

Can I use these ones (I can´t find your "start bit 24"):
[image]

meatpiHQ, 2024-03-10:
@Sx54A use any values for start bit and length, it will be ignored.

Sx54A, 2024-03-10:
ok, short update:

I found out, that I get nearly all values with the car scanner only, if the car is "ready" (toyota owners knows what I mean)... its equivalent to "ignition on".
When I switch the car of, Car scanner only shows "OBD Module Voltage" and "Current time". Disappointing. I did not expect, that mqtt will deliver more when car is switched off.