meatpiHQ/wican-fw #105: Only rx -no tx while loggin Carscanner OBD2 Request To MQTT

Gics78, 2024-03-10:
Hi,

I am trying to setup my Genesis GV70e in Homeassistant. I'm using this [guide](https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example). The Genesis should be close or identical to the Hyundai Ionic 5.

When continuously reading SoC in CarScanner App MQTT-Explorer only shows me rx-topics - there is no tx logged so I can't figure out the correct PID.
[image]

The MQTT messages look like this:
`{"bus":"0","type":"rx","ts":12489,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[16,62,98,1,1,239,251,231]}]}
{"bus":"0","type":"rx","ts":12424,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[40,18,0,0,0,0,7,38]}]}
{"bus":"0","type":"rx","ts":12414,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[39,216,0,78,110,102,0,3]}]}
{"bus":"0","type":"rx","ts":12404,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,0,98,146,0,0,92]}]}
{"bus":"0","type":"rx","ts":12404,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,0,98,146,0,0,92]}]}`

The only thing constant is the CAN ID 2028. So I'm kind of stuck. Is CarScanner IOS at fault? Might there be a wrong setting? Why are only rx messages captured?

I'm confused and looking for help!

Thank you

Gics78, 2024-03-17:
Okay, found my mistake: forgot to enable MQTT to elm327 log BEFORE Carscanner...
now I got a tx frame:
`{"bus":"0","type":"tx","ts":11333,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,0,170,170,170,170,170]}]}`
 and response:
`{"bus":"0","type":"rx","ts":11331,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[16,62,98,1,1,239,251,231]}]}`

So Can ID 2028, PID 98 What do I chose for index, start, length and expression? Seem like it is different from Hyundai after all...

meatpiHQ, 2024-03-17:
@Gics78 can you please post the rest of the frames? The tx frame show that it's a multi frame response, so its similar to what this user got in his log.

https://github.com/meatpiHQ/wican-fw/discussions/75#discussioncomment-7996872

Gics78, 2024-03-17:
`{"bus":"0","type":"tx","ts":26520,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,1,5,170,170,170,170]}]}`

response:

`{"bus":"0","type":"rx","ts":26527,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[16,46,98,1,5,255,251,116]}]}`
`{"bus":"0","type":"rx","ts":26538,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[33,15,1,44,1,1,44,12]}]}`
`{"bus":"0","type":"rx","ts":26547,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[34,12,11,11,11,11,12,50]}]}`
`{"bus":"0","type":"rx","ts":26547,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[34,12,11,11,11,11,12,50]}]}`
`{"bus":"0","type":"rx","ts":26557,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[35,22,153,232,0,0,100,11]}]}`
{`"bus":"0","type":"rx","ts":26557,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[35,22,153,232,0,0,100,11]}]}`
`{"bus":"0","type":"rx","ts":26567,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[36,0,3,232,108,128,0,0]}]}`
{"bus":"0","type":"rx","ts":26577,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[37,181,0,0,0,0,0,0]}]}
`{"bus":"0","type":"rx","ts":26587,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,11,12,11,12,170,170]}]}`
`{"bus":"0","type":"rx","ts":26587,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,11,12,11,12,170,170]}]}`

meatpiHQ, 2024-03-17:
@Gics78 is this for the SoC request?

Gics78, 2024-03-17:
This is for "[BMS] State of Charge BMS", I am wondering if I should have rather choosen "[BMS] State of Charge Display". Both should be 87-90%.

Gics78, 2024-03-17:
Soc Display: 91,5

{"bus":"0","type":"tx","ts":26860,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[3,34,1,5,170,170,170,170]}]}
`{"bus":"0","type":"rx","ts":26871,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[16,46,98,1,5,255,251,116]}]}`

{"bus":"0","type":"tx","ts":26873,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,0,170,170,170,170,170]}]}

`{"bus":"0","type":"rx","ts":26882,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[33,15,1,44,1,1,44,12]}]}`
`{"bus":"0","type":"rx","ts":26891,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[34,12,12,11,12,11,12,48]}]}`
`{"bus":"0","type":"rx","ts":26901,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[35,37,153,232,0,0,80,13]}]}`
`{"bus":"0","type":"rx","ts":26911,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[36,0,3,232,108,129,53,0]}]}`
`{"bus":"0","type":"rx","ts":26921,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[37,183,0,0,0,0,0,0]}]}`
`{"bus":"0","type":"rx","ts":26931,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,11,12,12,12,170,170]}]}`

meatpiHQ, 2024-03-17:
@Gics78 I suspect it might be this frame B1/2. Can you please post more frames logs at different SoC levels?

>{" bus":"0","type":"rx","ts":26921,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[37,**183**,0,0,0,0,0,0]}]}

Gics78, 2024-03-18:
@meatpiHQ thank you for your support -I will certain post more frames.
You are most certainly correct with the expression.

At least I found [this ](https://github.com/Esprit1st/Hyundai-Ioniq-5-Torque-Pro-PIDs/blob/main/TorqueIONIQ5AWD77kWh.csv) for the Ioniq 5 (which should be a close relative):

> Name,ShortName,ModeAndPID,Equation,Min Value,Max Value,Units,Header
000_State of Charge BMS,SOC BMS,0x220101,**e/2,0**,100,%,7E4
000_State of Charge Display,SOC,0x220105,**af/2,0**,100,%,7E4
000_State of Health,SOH,0x220105,((z<8)+aa)/10,0,100,%,7E4

Gics78, 2024-03-18:
Soc Display 85,5%: -> *2=**171**,   (State of Charge BMS 83,5%=**167**)
CarScanner for Soc Display continuously:
{"bus":"0","type":"tx","ts":29200,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,0,170,170,170,170,170]}]}

`{"bus":"0","type":"rx","ts":29208,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[33,15,1,44,1,1,44,12]}]}`
`{"bus":"0","type":"rx","ts":29218,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[34,12,12,12,12,12,12,65]}]}`
`{"bus":"0","type":"rx","ts":29228,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[35,193,153,232,0,0,80,12]}]}`
`{"bus":"0","type":"rx","ts":29238,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[36,0,3,232,108,119,235,0]}]}`
{"bus":"0","type":"rx","ts":29248,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[37,**171**,0,0,0,0,0,0]}]}
`{"bus":"0","type":"rx","ts":29258,"frame":[{"id":2028,"dlc":8,"rtr":false,"extd":false,"data":[38,0,12,12,12,12,170,170]}]}`

@meatpiHQ  So there it is, in the sceond from last frame!

Gics78, 2024-03-18:
Filters, MQTT elm327 is disabled now
[image]

Published this frame to wican/5432048f59f1/can/tx:
`{"bus":"0","type":"tx","ts":29200,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,0,170,170,170,170,170]}]}`
[image]

Unfortunately no answer in MQTT Explorer...

Gics78, 2024-03-18:
multi request?

meatpiHQ, 2024-03-19:
@Gics78 yes. Similar to this https://github.com/meatpiHQ/wican-fw/discussions/86

Note that this frame that you mentioned is not the actual PID request, it's a response from WiCAN that tells the ecu that you're ready for muli-frame response

> {"bus":"0","type":"tx","ts":29200,"frame":[{"id":2020,"dlc":8,"rtr":false,"extd":false,"data":[48,0,0,170,170,170,170,170]}]}

Gics78, 2024-03-19:
Okay, so will go back to logging in CarScanner and find the actual PID request. Then I will try to send the two requests from a script and hopefully will get a reasonable answer back.

BTW: as it seems that I get the whole of the BMS via the multiframe I might as well set up some other filters to get the most of it.
Thank you again- I could not succeed without your great support!