meatpiHQ/wican-fw #106: Rebooting WiCan by MQTT command

beaune33, 2024-03-13:
Hi,
in some special situations it makes sense to reeboot the wican adapter. This is possible already by using the web interface. But is it also possible to call reboot by using an MQTT command? That would be great!

meatpiHQ, 2024-03-13:
@beaune33 will be available in the next release. Also adding "get battery voltage" command.

savikko, 2024-08-18:
> @beaune33 will be available in the next release. Also adding "get battery voltage" command.

Hi, is this "get battery voltage" already on the current (beta) release?

meatpiHQ, 2024-08-18:
@savikko no not yet, I'll add it now.

meatpiHQ, 2024-08-18:
Ok done, I've added 2 new commands to get the battery voltage and reboot over MQTT. Keep an eye on the next beta or release.