meatpiHQ/wican-fw #107: Car scanner can't connect to ECU

chrismelba, 2024-03-13:
Not sure if this is the best place to post, or if possibly I'm facing a hardware issue.

I have plugged my WiCAN into the OBDII port of my XC40 electric. 

I can log in to the captive portal. I have connected to my home wifi.
I've set up the MQTT broker and set the read mode to ELM327
I can connect to the ELM server using car scanner, but car scanner says it can't get the ECU connection.

MQTT shows that the WiCAN is online, but i never get any RX messages, I've tried sending some TX which log properly.

Going to the monitor tab shows 3 different frames, though I don't understand them

Using a cheap o bluetooth OBDII scanner I can connect immediately and get values no problem.

I have turned bluetooth off

I have power save on

Gics78, 2024-03-16:
Make sure you have CAN Mode on normal and not silent.

meatpiHQ, 2024-03-16:
@chrismelba can you please post a screenshot of your device configuration webpage?

I suggest you disable mqtt and power saving mode at first.

> Going to the monitor tab shows 3 different frames, though I don't understand them

Is the engine ON or OFF?

chrismelba, 2024-03-17:
Turned MQTT off and tried to connect again and it worked!

It's an EV, and what I think was actually happening is that I turned it on, put it in drive, then tried to connect. But as soon as I took my foot off the brake it would go back to park. If I left my foot on the brake and tried to connect, then carscanner would work fine with or without MQTT.

Thanks for the prompts, I'll try my rubber ducky a little harder next time!

Really appreciate the project!

closed 2024-03-17 by chrismelba