meatpiHQ/wican-fw #108: Reading Odometer from 2019 Volkswagen e-Golf

MrSaiclops, 2024-03-21:
Hello! I'm so pleased with this product and feel like I'm just scraping the surface of it's usefulness so far. I'm successfully able to pull the SoC from my vehicle while its charging which is essential for reminding myself and others to plug it in if forgotten. 

The other stat I would like to pull is the odometer reading, for basic charting and insights into usage. CarScanner wasn't able to provide this for either profile tuned towards the e-golf, but it did get a perfect readout by using their profile for the newest get (id.4 id.5 etc.). By perfect I mean my dash showed **44353 mi** and CarScanner reads: **44353.48 mi**

Parsing the MQTT elm327 log when only querying this information yielded these messages:
`{"bus":"0","type":"tx","frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[48,0,0,170,170,170,170,170]}]}
{"bus":"0","type":"tx","frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[3,34,2,189,170,170,170,170]}]}
{"bus":"0","type":"rx","frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[16,13,98,2,189,147,1,22]}]}
{"bus":"0","type":"rx","frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[33,212,1,88,96,234,116,52]}]}`

I've tried a variety of parsing methods and can't determine how these responses show 71380 km (Assuming it's converted)

So here are the questions:
1. Does anyone know what calculation I need to use with these responses to get the milage?
2. Does anyone know why these frame ids are so much longer than the typical ones? 
3. Will I need to build a combo transmission that sends two messages, as seen in the elm327 log?

<details>

<summary>For reference, here are codes I know work with my car:</summary>

### SoC

`{ "bus": "0", "type": "tx", "frame": [{ "id": 2021, "dlc": 8, "rtr": false, "extd": false, "data": [3, 34, 2, 140, 170, 170, 170, 170] }] };`

```js
var obj = Object(); 
for (let i = 0; i < msg.payload.frame.length; i++) {
    if (msg.payload.frame[i].id === 2029 && msg.payload.frame[i].data[1] === 98 && msg.payload.frame[i].data[2] === 2) {
        obj.soc = (msg.payload.frame[0].data[4] / 2.55).toFixed(2);
        msg.payload = obj;
        msg.topic = "CAR1/SoC";
        return msg;
    }
}
```

### Ambient Temp (In Fahrenheit) 
`{ "bus": "0", "type": "tx", "frame": [{ "id": 1862, "dlc": 8, "rtr": false, "extd": false, "data": [3, 34, 38, 19, 170, 170, 170, 170] }] };`
```js

var obj = Object(); 
for (let i = 0; i < msg.payload.frame.length; i++) {
    if (msg.payload.frame[i].id === 1968 && msg.payload.frame[i].data[1] === 98 && msg.payload.frame[i].data[2] === 38) {
        obj.int_temp = (msg.payload.frame[0].data[4] * 1.8) + 32;
        msg.payload = obj;
        msg.topic = "CAR1/Interior_Temp";
        return msg;
    }
}
```
[image]

</details>

pyrousg, 2024-03-29:
> Hello! I'm so pleased with this product and feel like I'm just scraping the surface of it's usefulness so far. I'm successfully able to pull the SoC from my vehicle while its charging which is essential for reminding myself and others to plug it in if forgotten.
> 
> The other stat I would like to pull is the odometer reading, for basic charting and insights into usage. CarScanner wasn't able to provide this for either profile tuned towards the e-golf, but it did get a perfect readout by using their profile for the newest gen (EV MEB ID.3, ID.4, etc.). By perfect I mean my dash showed **44353 mi** and CarScanner reads: **44353.48 mi**
> 
> Parsing the MQTT elm327 log when only querying this information yielded these messages: `{"bus":"0","type":"tx","frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[48,0,0,170,170,170,170,170]}]} {"bus":"0","type":"tx","frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[3,34,2,189,170,170,170,170]}]} {"bus":"0","type":"rx","frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[16,13,98,2,189,147,1,22]}]} {"bus":"0","type":"rx","frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[33,212,1,88,96,234,116,52]}]}`
> 
> I've tried a variety of parsing methods and can't determine how these responses show 71380 km (Assuming it's converted)
> 
> So here are the questions:
> 
> 1. Does anyone know what calculation I need to use with these responses to get the milage?
> 2. Does anyone know why these frame ids are so much longer than the typical ones?
> 3. Will I need to build a combo transmission that sends two messages, as seen in the elm327 log?
> 
> For reference, here are codes I know work with my car:

Hello, i would like to readout data from my egolf 2018 using wican. Could i ask you about help with codes for reading SOC, odometer, etc? Tomas

MrSaiclops, 2024-03-30:
> it's hard to tell from a single data point can you please get another read, it would make it easier to guess.
> What are the rx and tx frame before this frame:
> _{"bus":"0","type":"tx","frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[48,0,0,170,170,170,170,170]}]}_

Here you go:

<details>

<summary>Car Scanner reading 44541.75 mi:</summary>

{"bus":"0","type":"tx","ts":39534,"frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[48,0,0,170,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":39542,"frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[33,3,0,0,96,252,129,176]}]}
{"bus":"0","type":"tx","ts":39570,"frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[3,34,2,189,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":39582,"frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[16,13,98,2,189,166,1,24]}]}
{"bus":"0","type":"tx","ts":39584,"frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[48,0,0,170,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":39591,"frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[33,3,0,0,96,252,129,176]}]}
{"bus":"0","type":"tx","ts":39627,"frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[3,34,2,189,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":39632,"frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[16,13,98,2,189,166,1,24]}]}
{"bus":"0","type":"tx","ts":39634,"frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[48,0,0,170,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":39641,"frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[33,3,0,0,96,252,129,176]}]}
{"bus":"0","type":"tx","ts":39667,"frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[3,34,2,189,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":39672,"frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[16,13,98,2,189,166,1,24]}]}
{"bus":"0","type":"tx","ts":39675,"frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[48,0,0,170,170,170,170,170]}]}
{"bus":"0","type":"rx","ts":39682,"frame":[{"id":402522235,"dlc":8,"rtr":false,"extd":true,"data":[33,3,0,0,96,252,129,176]}]}
{"bus":"0","type":"tx","ts":39740,"frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[3,34,2,189,170,170,170,170]}]}

</details>

pyrousg, 2024-04-04:
> {"bus":"0","type":"tx","frame":[{"id":402391163,"dlc":8,"rtr":false,"extd":true,"data":[48,0,0,170,170,170,170,170]}]}

I understand, but if you could share json, it should be great, i am not able to program it. thank you

MrSaiclops, 2024-04-05:
> I understand, but if you could share json, it should be great, i am not able to program it. thank you

Alright, here it is:
[flows.json](https://github.com/meatpiHQ/wican-fw/files/14885210/flows.json)

pyrousg, 2024-04-05:
but it seems alive 
[image]

MrSaiclops, 2024-04-05:
Hmmmm. In all of the MQTT nodes, sure you've changed the middle section of the topic: `wican/xxxxxxxxxxxx/can/` to match your model, not mine.
My config.yml has this to create the sensors:
```yaml
mqtt:
  sensor:
    - name: "Amb Temp"
      state_topic: "CAR1/Interior_Temp"
      unit_of_measurement: "F"
      value_template: "{{ value_json.int_temp | round(1) }}"
    - name: "Plug Voltage"
      state_topic: "CAR1/Plug-Voltage"
      unit_of_measurement: "V"
      value_template: "{{ value_json.plug_volt }}"
    - name: "Battery SoC"
      state_topic: "CAR1/SoC"
      unit_of_measurement: "%"
      value_template: "{{ value_json.soc }}"
    - name: "Plug Inserted"
      state_topic: "CAR1/Plug-Inserted"
      value_template: "{{ value_json.plug }}"

Please keep in mind my configuration is far from perfect and I didn't intend to share it. But if it's able to work for you, I'm glad.

pyrousg, 2024-04-05:
Thank you so much. still not working, i am not sure why... How you see i am connected, so i was change my id wican, i was change to my adress of mqtt. I was looking over [MQTT Explorer] and there was tx and rx mesages... so HA is sending requests and wican reply, but, why i not see in HA i dont understand. I was add sensor according your config. I saw them in entities, but data are not recorded there. I am close, some minor mistake somewhere separated me from the data from egolf :-) Thank you again, i really apriciate it.

pyrousg, 2024-04-06:
> Hmmmm. In all of the MQTT nodes, sure you've changed the middle section of the topic: `wican/xxxxxxxxxxxx/can/` to match your model, not mine. My config.yml has this to create the sensors:
>
> ```yaml [...]

Hey man, it is working... it just need to be car engine on... So it is bad, that it is not possible to monitor all time, but even this functionality will be nice... I will now add some more measurments... i could share, if will work... Thank you very much for your help... Tomas

pyrousg, 2024-04-06:
Do you know, please, if it have effect of selection of protocol? which one i should use? slcan? or another one?

pyrousg, 2024-04-06:
It is nice, that during charging is refresh of soc every 2 minutes...

pyrousg, 2024-04-14:
Hello, i would like to read energy of battery pack, i was read out these data:
{"bus":"0","type":"rx","ts":36841,"frame":[{"id":1914,"dlc":8,"rtr":false,"extd":false,"data":[33,1,178,170,170,170,170,170]}]}
{"bus":"0","type":"tx","ts":16790,"frame":[{"id":1808,"dlc":8,"rtr":false,"extd":false,"data":[3,34,21,18,170,170,170,170]}]}

how to construct reading of rx? And calculation from data?

var obj = Object(); 
for (let i = 0; i < msg.payload.frame.length; i++) {
    if (msg.payload.frame[i].id === **1914** && msg.payload.frame[i].data[1] === **33** && msg.payload.frame[i].data[2] === 1) {
        obj.energy = msg.payload.frame[0].data[4];
        msg.payload = obj;
        msg.topic = "CAR1/energy";
        return msg;
    }
}

Thank you Tomas

MrSaiclops, 2024-04-28:
@pyrousg, Can you please start your own issue to request help? This one is now cluttered with numerous messages that mostly pertain to **your** vehicle which is **not** a 2019. My original question has still not been answered by @meatpiHQ and I suspect that it's because the conversation is difficult to track with all of your messages.

pyrousg, 2024-04-29:
> @pyrousg, Can you please start your own issue to request help? This one is now cluttered with numerous messages that mostly pertain to **your** vehicle which is **not** a 2019. My original question [...]

I am appologize, should i reduce my post? to clarify your thread? Tomas

meatpiHQ, 2024-04-30:
@pyrousg please open a new "GitHub issue". It's easier to keep track of things this way.

[image]

meatpiHQ, 2024-04-30:
@MrSaiclops I went through your issue again, and im still unable to find right bytes for the odometer. 

Usually odometer PID is 166. 

[image]

DerHerrW, 2024-05-21:
@MrSaiclops: Regarding ODOmeter - Could you try the request / response from the e-up? the e-Golf could be similar; at least raw SOC is identical.

The request 32,2,189 with id 2021 triggers a multipart response (here 2) from the BCU. After receiving the first part it is necessary to request the remaining messages with ID 2021, data ([48,0,0,170,170,170,170,170]). The combined payload of first and second part of the ODO-response is  [98, 2, 189, xx, bb, cc, dd, xx, xx, xx, xx, xx, xx]. kilometer ist bb*2^16+cc*256+dd

If you understand german, you can find my tool unter  https://forum.openwb.de/viewtopic.php?p=91886#p91886

I would very much appreciate your feedback.

MrSaiclops, 2024-05-23:
@DerHerrW Thank you for reaching out! I'm a bit confused about how to execute your suggestions. 
1. Send ``{ "bus": "0", "type": "tx", "frame": [{ "id": 2021, "dlc": 8, "rtr": false, "extd": false, "data": [32, 2, 189, 170, 170, 170, 170, 170] }] };``
2. Save the response which should look something like: ``[98,2,189, xx, bb,dd, xx, xx]``
3. Send ``{ "bus": "0", "type": "tx", "frame": [{ "id": 2021, "dlc": 8, "rtr": false, "extd": false, "data": [48, 0, 0, 0, 170, 170, 170, 170] }] };``
4. Save the second response

This is where I'm lost, how do I combine the second response with the first? 

Is there any chance you could look at my original post to see how the information is formatted in NodeRed and give me instructions in a similar manner?

DerHerrW, 2024-05-23:
I do not use node red but i can show you how I coded (quit&dirty) an OBD2 communication in python. A great help was this wiki page: https://en.m.wikipedia.org/wiki/ISO_15765-2. Here is the code I use in a callback function that gets called if the wican receives an answer to a obd2 request. The code is under the apache license 2.0:

Test if message is multipart (4 higher bytes of data byte 0): 1 - part one of a multipart message 2 - follow part of a multipart message; 0 - single part message.
if 1: the number of bytes to receive is in the lower nibble of byte 0 and byte 1. The payload is the last 6 bytes. if we are interested in the following part we send the continue message (the 100 ist a requested pause of 100ms between each of the parts, does not matter as there will be only one)
if 2: a follow up message is received (tpType==2). The first data byte is the index of the message (message 3 may be received before 2 etc). I simply ignore this. The remaining 7 data bytes are simply attached at the end of the payload from the first message.
`
    tpType =  data[0] // 16        # Oberen Nibble von Byte 0 extrahieren
    cb_rx.messageComplete = False
    if tpType == 1:
        # Erster Teil einer mehrteiligen Botschaft
        # Anzahl zu empfangenen Bytes: 12 Bit aus unterem Nibble Byte 0 und ganzes Byte 1
        cb_rx.bytesToReceive = (data[0] & 15)*256 + data[1]
        cb_rx.payload = [id]
        cb_rx.payload.extend(data[2:8])
        cb_rx.bytesReceived = 6
        logging.debug(f'Ersten Teil einer mehrteiligen Botschaft empfangen: {cb_rx.payload}')
        # zu Sende-ID des Senders die passende Empfänger-ID suchen:
        if id == configuration.ODO_RESP_ID:
            flowCtrl = messageFromData(configuration.ODO_REQ_ID, [48,0,100,170,170,170,170,170])
        elif id == configuration.SOC_RESP_ID:
            flowCtrl = messageFromData(configuration.SOC_REQ_ID, [48,0,100,170,170,170,170,170])
        # Fordere alle weitere Botschaften an mit 100ms Pause zwischen den Frames
        logging.debug(f'Aufforderung für Folgeteile absetzen: {flowCtrl}')
        client.publish(wican_topicTx, flowCtrl)
    elif tpType == 2:
        # Botschaft ist ein Folgeteil einer mehrteiligen Botschaft. Der Index sollte im unteren Nibble von Byte 0 stehen
        # hier wird einfach gehofft, daß die Botschaften in der richtigen Reihenfolge ankommen. Sie werden stumpf angehängt.
        cb_rx.payload.extend(data[1:8])     # Je Nachfolger sollten 7 Nutzbytes kommen
        cb_rx.bytesReceived += 7
        if cb_rx.bytesReceived >= cb_rx.bytesToReceive:
            cb_rx.messageComplete = True
        logging.debug(f'Mehrteilige Botschaft komplett: {cb_rx.payload}')
    elif tpType == 0:
        # Einteilige Botschaft
        cb_rx.payload = [id]
        cb_rx.payload.extend(data[1:8])
        cb_rx.bytesReceived = 7
        cb_rx.messageComplete = True
        logging.debug(f'Einteilige Botschaft: {cb_rx.payload}')
    else:
        logging.warning('Botschaft mit unbekanntem CAN-TP-Botschaftstyp oder FlowControl empfangen.')`

DerHerrW, 2024-06-19:
> Im also seeking for odometer reading.
> Got these messages.
>
> *{
>   "bus": "0",
>   "type": "tx",
>   "ts": 22534,
>   "frame": [
>     {
>       "id": 2015,
>       "dlc": 8,
>       "rtr": false,
>       "extd": false,
>       "data": [
>         2,
>         1,
>         166,
>         170,
>         170,
>         170,
>         170,
>         170
>       ]
>     }
>   ]
> }
> **{
>   "bus": "0",
>   "type": "rx",
>   "ts": 22544,
>   "frame": [
>     {
>       "id": 2024,
>       "dlc": 8,
>       "rtr": false,
>       "extd": false,
>       "data": [
>         6,
>         65,
>         166,
>         0,
>         11,
>         79,
>         36,
>         0
>       ]
>     }
>   ]
> }
> *
> How should i parse message to get readable message?
> Car scanner shows 74115.6km

(11*65536+79*256+36*1)/10=74 115.6

JabeBRD, 2024-06-19:
What im doing wrong, cant get RX message as return for odometer, battery soc is returning as expected??
Im sending two messages, with 5s interval 
`{"bus":"0","type":"tx","frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,91,170,170,170,170,170]}]}`
`{"bus":"0","type":"tx","frame":[{"id":2015,"dlc":8,"rtr":false,"extd":false,"data":[2,1,166,170,170,170,170,170]}]}`
[image]

DerHerrW, 2024-06-19:
Start bit is 32, length is 24 Bits. I wonder why SoC works with start bit 0 and length 1.

JabeBRD, 2024-06-20:
Thank you, its working now.
But should it be readable when charging? Now its readable only when ignition is turned on (from "key").

DerHerrW, 2024-06-20:
Same with my eUp. Some time after charge start the DC/DC Converter is shut down and my wican hibernates. Maybe the ECU is also shut down.

JabeBRD, 2024-06-20:
Its actualy 2020 Passat GTE, but very close to golf in parts.
I have also e-Up, but my car scanner didnt find odometer reading, or it was not listed by default for that car, and that 166 request didnt work, do you have working request address?

DerHerrW, 2024-06-20:
A couple of posts before: https://github.com/meatpiHQ/wican-fw/issues/108#issuecomment-2122237514

JabeBRD, 2024-06-23:
I can confirm that Passat GTE uses same messages as e-golf for odometer readings, same RX and TX messages as in first post.
Odometer reading just like in e-Up like DerHerrW said, and this can be read when car is charging, no need for ignition on from key.
First send
```
{
  "bus": "0",
  "type": "tx",
  "frame": [
    {
      "id": 402391163,
      "dlc": 8,
      "rtr": false,
      "extd": true,
      "data": [ 3, 34, 2, 189, 170, 170, 170, 170 ]
    }
  ]
}
```
receive message with PID 189
```
{
  "bus": "0",
  "type": "rx",
  "ts": 25020,
  "frame": [
    {
      "id": 402522235,
      "dlc": 8,
      "rtr": false,
      "extd": true,
      "data": [ 16, 13, 98, 2, 189, 171, 1, 34 ]
    }
  ]
}
```
after that send second message
```
{
  "bus": "0",
  "type": "tx",
  "frame": [
    {
      "id": 402391163,
      "dlc": 8,
      "rtr": false,
      "extd": true,
      "data": [ 48, 0, 0, 170, 170, 170, 170, 170 ]
    }
  ]
}
```
then receive final message with PID 33
```
{
  "bus": "0",
  "type": "rx",
  "ts": 25023,
  "frame": [
    {
      "id": 402522235,
      "dlc": 8,
      "rtr": false,
      "extd": true,
      "data": [ 33, 148, 30, 224, 97, 175, 76, 192 ]
    }
  ]
}
```
Settings for WiCan
[image]
Im using Home Assistant, so i just set up helper to combine odo1 + odo2 and get complete odometer reading.
Hope this helps someone. That odometer with CAN id 2024 and PID 166 works only when ignition on from key, just ignore it, it is not needed.

typxxi, 2024-08-07:
Has anyone the adresses needed to read the e-GOLF SOC ?
And may be current range ?
Odometer is not that helpfull right now, but current range would be great too.

I hope someone had achieved that before but I had no success so far in my search and tried to use the e-Up codes that should be similiar but the car is not replying at all. I only get the online state in HA, once the car comes into Wifi range.

And then it does not react on the codes send cause on mqtt explorer I do not see an answer. 
Here is the call send 
{ "bus": "0", "type": "tx", "frame": [{ "id": 2015, "dlc": 8, "rtr": false, "extd": false, "data": [3, 34, 2, 140, 170, 170, 170, 170] }] };

BUT, a long BUT
in case I am in the car, ignition is on and I do connect with CarScanner to the car the SOC will arrive once i chose "all sensors" in the ui and filtered that to "state" where I then get 3 sensors shown, first and last are SOC sensors and within a few seconds the % figure will appear.

In the same moment the mqtt Explorer will show the SOC updated and so on. 
It will stop working the moment I am going back out of this filtered sensor overview with just 3 sensors. 
If I leave that communication via mqtt will stop even though I am still connected on the phone with the car.

My battery entry on the Meatpi also looks a bit different with with these
`2029 BMS_SOC 4 0 32 8 V/2.55 1000` = e-Up compared to Passat GTE below seen above
while these here from the Passat are
'2029 Battery_SOC 91 2 24 8 (B3-50)/1.9 5000'

Any ideas where to look - or did I got a faulty meatpic Adapter from Mouser USA shipped to germany ?
I do not think so even though the behaviour is a bit weird but at least I learned to reproduce it and test it over and over again when the transmission reaches mqtt explorer and when not.

Thanks a lot

typxxi, 2024-08-07:
> Start bit is 32, length is 24 Bits. I wonder why SoC works with start bit 0 and length 1.

You seem to have a lot clue about the e-GOLF and I am struggling hard to get the e-GOLF replying to my call for SOC as seen in the Passat GTE, which I have tried out after my previous e-Up settings had some flaws or the whole did not work properly. 

The Passat GTE worked - somehow even though the reported figures were 2% higher than the right ones from carscanner and it suffered from the same flaw the e-Up settings had caused I guess.

Here Passat GTE
[image]

I had taken the e-Up settings before for the e-GOLF where the results were accurate those I got from carscanner Pro, so a bit better .
[image]

BUT and this is a big but: 

mqtt shows / gets the SOC reply only in 1 scenario: 
#1 ignition has to be turned on
#2 then car scanner pro has to start
#3 and connect to the car
#4 and finally I have to open the sensor page and use the filter to show all sensors with "state"

And then, when I see the 3 sensors that match criteria and those sensors start to show figures, only then mqtt receives the messages of BMS SOC 52,12334% or so. 

If I press back on the app and leave that sensor menu the mqtt explorer shows no more  updates. 
If I turn off ignition I get the same result. 

What is going wrong ?
Here is the automation 

```
action:
  - repeat:
      sequence:
        - service: mqtt.publish
          data:
            qos: "0"
            retain: true
            payload: >-
              { "bus": "0", "type": "tx", "frame": [{ "id": 2015, "dlc": 8,
              "rtr": false, "extd": false, "data": [3, 34, 2, 140, 170, 170,
              170, 170] }] };
            topic: wican/1234567890/can/tx
        - service: mqtt.publish
```
It is working but only while the car scanner app is running the sensor page and ignition is turned.

I hope you know the ids needed to use when ignition is turned of. I had hoped GTE settings might work, but they did not.

DerHerrW, 2024-08-07:
You transmit with a Broadcast id 2015. Maybe try id = 2021?

JabeBRD, 2024-08-07:
I currently have same request for e-Up and Passat GTE
`{"bus":"0","type":"tx","frame":[{"id":2021,"dlc":8,"rtr":false,"extd":false,"data":[3,34,2,140,170,170,170,170]}]}`
And yes PID 91 gives different value than PID 140.
You need to use control units that are "online" while charging, you can scan what units are online with Car Scanner while charging.

grobius, 2024-08-26:
Hi, I've been following this interesting thread, but can't sort out one thing. As I understand the OBD port goes via the CAN GW which I think is "sleeping" when the car is parked.

Have you been able to "wake" it and get SoC and ODO while parked (and no ignition on)?

AS a tip there is another job beeing done by Thomas Karlsen to reverse engineer the comfort CAN messages, which is needed to e.g. turn on the HVAC (like in teh Carnet App).

https://github.com/thomasakarlsen/e-golf-comfort-can

JabeBRD, 2024-08-26:
With EV and hybrid cars, some control units wake when vehicle is in charging, this is when you are able to read these control units.

molelightn, 2025-04-20:
> I can confirm that Passat GTE uses same messages as e-golf for odometer readings, same RX and TX messages as in first post. Odometer reading just like in e-Up like DerHerrW said, and this can be [...]

Hi there,

i really strugle to take over your settings for odo1 and odo2. The new firmware has a much different setup for manual pids and i am a bit lost with the settings found in your screenshot. Do you have your odo readings already available as per new firmware UI 4.x? 
I do get readings but they make no sense.
E-Golf 2019

ty
br

JabeBRD, 2025-04-20:
Please post your readings and odometer reading.
I dont use these settings in wican anymore, i have everything setup in Home Assistant, Wican sends raw values only.
Odometer and some others values use these "long" messages, messages that start with 33 or 34 (this is messages second or third reply), if you read more than one values that use long messages from same control unit, it is possible that these messages get mixed, as only way to identify these messages is from timestamp.
I dont know if wican can identify messages from timestamp, but HA can do it. Wican provides timestamp for every message.

molelightn, 2025-04-21:
Hello JabeBRD, 

first of all thank you for your help, below my settings, i guess this is completely wrong:

[image]

[image]

this is what i receive:

[image]

Kaffeetasse09, 2025-06-04:
Hi,
my eGolf is from 2020.
I tried the following setup.
WiCAN setup for SoC (implemented in WiCAN-SW)
[image]

WiCAN setup for Range (Custom PID)
[image]

At MQTT-Broker I can find the following:
every 0.5 seconds:
`{"bus":"0","type":"rx","ts":57510,"frame":[{"id":401604624,"dlc":8,"rtr":false,"extd":true,"data":[32,16,0,0,0,0,0,0]}]}`

every 5 seconds:
`{"bus":"0","type":"rx","ts":57967,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[4,98,2,140,116,170,170,170]}]}`
`{"bus":"0","type":"rx","ts":57316,"frame":[{"id":2029,"dlc":8,"rtr":false,"extd":false,"data":[3,127,22,17,170,170,170,170]}]}`

First code is representing the SoC (45,7%), second one seems to be the answer for my custom PID...
So everthing is working fine...only me request is wrong.

How do I need to setup my custom PID's to get "Range" and "Odometer"?

JabeBRD, 2025-06-04:
Which control unit are you querying??
Those two messages come from engine 01 control unit, try querying control unit dash 17 or battery 8C.
That second reply is basicly saying error and there is no `PID 166` in message, if that is right PID then that should be found usually as fourth number in reply, see your SoC reply as example.

https://www.goingelectric.de/wiki/VW-e-up-OBD2-SG8C

molelightn, 2025-06-05:
This is how Range is working on my Wican, couldn't follow up Odometer yet.

INIT:      ATSP6;ATSH710;ATCP18; ATCRA77A;ATFCSH710;

[image]

Kaffeetasse09, 2025-06-05:
> This is how Range is working on my Wican with E-Golf, couldn't follow up Odometer yet.
> 
> INIT: ATSP6;ATSH710;ATCP18; ATCRA77A;ATFCSH710;
> 
> [image]

Many thanks for your reply.
Range is now working perfectly. Already added to EVCC config.

@JabeBRD Regading your question about the control unit...I have no clue :-)
Is this defined by the PID...e.g. 222AB62 in case for the range?

I wonder why I can not see the transmitted messages in the MQTT Explorer...I marked the tx-messages in the settings of the WiCan dongle...

I have seen your screenshot for odo1 and odo2 of your config:

[image]

I guess this was taken with an older firmware, right?

JabeBRD, 2025-06-05:
You cant see TX messages with your setup, because you are not sending any, wican uses its own automation system, so there is no need for those messages.
TX messages come from outside wican, example from Home Assistant. 
But if you want to send them, you can and wican will act with them.
And that is what you see my screenshot was made for, you can use both ways. Infact, i send tx messages and my Home Assistant reads RX messages and makes those into sensors, there is no settings made in wican except connection to my MQTT server.
You can use it atleast three ways. 1 wican full automated, 2 you send TX wican returns with filtered reply/value, 3 you send TX wican returns unfiltered RX and you make it to sensor
I found out that last one was best for me, atleast at that time, i can filter messages so that it wont mix with other messages, as that can happen when dealing with "long" messages, ones that has 2 or 3 parts, these extension look all the same, only timestamp can identify them as right messages.

ccremer224, 2025-06-06:
The new e-golf.json :

      "pid": "2202BD",
      "parameters": {
        "ODOMETER": "[B5:B7]"

gives completely wrong odometer values for my e-Golf (2020):
The value doesn't change when the vehicle was moved. So it's the wrong address for Odometer.

JabeBRD, 2025-06-06:
In that PID `22 02 BD` calculation should be `B1*65536 + B2*256 + B3`

Spexx-, 2025-06-09:
For Odometer I will get correct values with following calculation: (B6<<16)|(B7<<8)|B9

molelightn, 2025-06-09:
@Spexx- do you request with custom PID? if yes, can you please share a Screenshot?

Thank you

molelightn, 2025-06-10:
works perfect - thank you very much

Kaffeetasse09, 2025-06-11:
> works perfect - thank you very much

Sorry for late response...thanks to all of you explaning again and again... :-)
May I ask for some screenshots or maybe the downloadad config (JSON-File...only the vehicle specific areas)?
I was not able to get it running...even with your detailed information.
Topic was not listed in MQTT broker...

Kaffeetasse09, 2025-06-17:
It's me again....
I have issues to get all 3 "request" running. After my last comment everything was working. After reconnecting the car 2 days later, only the topic of the request which was handled with a custom PID was sent out. 2 topics handeld via vehicle specific PID (in case for the Golf Soc and Odometer) were not send out.
I did some tests with defining only 2 requests and so on....it was always the same... a combination of vehicle specific PID (provided by firmware) and custom PID end up with sending only the topic which is linked to the custom PID.
I'm using firmware 4.12 on my MeatPi. 

Currently I'm using only SoC and Odometer (provided by firmware). Yesterday and today everthing is working.

To all the eGolf users outside...do you have a similar behaviour? In case not, which firmware are you using?

Regards, Timo

ccremer224, 2025-06-19:
> vehicle_profiles_eGolf_v3.json

Doesn`t work for my e-Golf 2020 👎 
Loading that vehicle profile No values at all appear any more in HomeAssistant.

Spexx-, 2025-06-19:
@ccremer224 
Are you using the same name for the PIDs as before? During my tests I found out, that the HomeAssistant integration does not automatically update the sensors, if new sensors will be added(?). The best way is probably to remove the WICAN device in HA and add it once again.

Why did you ignore the PID-Init? Maybe you can compare my screenshots with your configuration, I don't use the "Specific Initialisation", only "PID-Init". Try it in the first step only with two PIDs (from different ECU-IDs), for example SOC and Range.

[image]

[image]

[image]

ccremer224, 2025-06-20:
When I use vehicle_profiles_eGolf_v3.json  in the   _Vehicle Profiles:_  from the WiCAN Setup Page (Automation) I only get the e-Golf as vehicle which is fine, but I get no response in Home Assistant anymore from the WiCAN.
If I use the Custom PID as shown below, I get response in Home Assistant as shown below.

[image]

[image]

Kaffeetasse09, 2025-06-20:
@Spexx- 
I have tested your file now. I have added the destination for all 6 values to "MQTT_Topic" and added my specific topic names...
All 6 requests are working currently. Topics are readable within MQTT-Explorer.
Thanks for your support.

Let's wait after next charging session if this is stable...

molelightn, 2025-06-21:
For me receiving of data with HA Plugin wasn't working very stable. Since i switched over to send it via MQTT it works like a charm.

JabeBRD, 2025-12-23:
You can get plug status, plugged in/out, locked/unlocked with this message
```
{"bus":"0","type":"tx","frame":[{"id":1893,"dlc":8,"rtr":false,"extd":false,"data":[3,34,29,218,170,170,170,170]}]}
```
Lock status is in data bit 5 and plug status in data bit 6.

Spexx-, 2025-12-24:
@Webwanze 
Charging_mode should represent following states:

0 = “Error, check wallbox & vehicle!” / "Fehler, Wallbox & Fahrzeug prüfen!"
1 = “Idling (no vehicle)” / "Im Leerlauf (kein Fahrzeug)"
2 = “Vehicle is charging” / "Fahrzeug wird geladen"
3 = “Waiting for vehicle” / "Auf Fahrzeug warten"
4 = “Charging process completed” / "Ladevorgang beendet"
5 = “Error, check wallbox & vehicle!” / "Fehler, Wallbox & Fahrzeug prüfen!"
else “Unknown status” / "Unbekannter Status"

tronikos, 2026-10-26:
FYI, https://github.com/meatpiHQ/wican-fw/pull/907 has all working parameters for a VW e-Golf 2019.
https://github.com/meatpiHQ/wican-fw/pull/981 has only the ones worth polling.