meatpiHQ/wican-fw #111: Disable AP

jay-oswald, 2024-03-26:
I would like to have the option to disable AP on the WiCan

The way my other smart devices work is I can disable AP, but if the device can't connect to my home WiFi AP is re enabled temporarily to enable it to he reconfigured.

There is also a limit with Bluetooth where it can only be on when it's in AP only mode, I'm also hoping that it could work with station plus Bluetooth. As I would like to be able to use home wifi at home, then connect via Bluetooth to use car scanner app when away from home, which isn't possible at the moment.

njh, 2024-08-13:
Yes please!

I would rather not go round advertising that I have a WiCAN device in my car.

meatpiHQ, 2024-08-14:
Ok looks like this is the most popular feature request, I'll add this in the next update. But the problem is that when you're on the road away from your wifi it will still be broadcasting. I did solve this by adding a button on WiCAN Pro but unfortunately WiCAN doesn't have a button.

jay-oswald, 2024-08-17:
@meatpiHQ could this logic work?

Mode 1: AP only, no station (default)

Mode 2:

Leave station mode on constantly.

When device is first plugged in enable AP for 5 mins. If nothing connects in those 5 mins it just turns off.

That way if someone needs to update details, they can unplug it, then plug it back in to enable the AP and connect to it. Could also then add an option on web UI to reboot without AP to disable again after your done.

Would let the AP be off 99% of the time, and easy enough to enable it by unplugging and replugging when needed

njh, 2024-08-17:
@jay-oswald I like that idea. Is it standard that OBD-II ports are always powered, even if ignition is off?

jay-oswald, 2024-08-17:
@njh yes there are constant power pins on obd2 port. Which im pretty sure wican already uses

stav242, 2024-10-12:
yes please. just want the AP to stop broadcasing, i only use station mode. i dont need the AP - its a exposed vulnerability being there if not required.

meatpiHQ, 2025-12-15:
@tweak19 already implemented.

tweak19, 2026-01-02:
+1 works also for me. 
Only poor bug in 4.12 is the vehicle import. 
THX!