meatpiHQ/wican-fw #113: Support for multiple wifi stations

djungelola, 2024-03-31:
Really think support for multiple wifi stations are needed, for example home, work and mobile hotspot to be able to get correct information at all possible locations.

meatpiHQ, 2024-03-31:
@djungelola this was hard to do in the old IDF, wifi kept breaking. I Haven't tried on the new one yet if it's any better. I'll try again soon after the auto PID request.

tqma1, 2024-07-04:
Any news regarding this topic?
Same situation here: home and mobile hotspot would be needed

asherry99, 2025-06-15:
I would love to see this supported someday as well! +1

pulpier, 2025-08-31:
Any news on this? Will this work with thd old WiCan or only the new Pro version?

meatpiHQ, 2026-01-28:
It's only supported on the PRO version

dral3x, 2026-01-28:
Got it, thank you guys!