meatpiHQ/wican-fw #116: How to get SoC for Opel Mokka-e in HA

jove01, 2024-04-05:
Hello
I have a Opel Mokka-e. With the PSA-addon in HA and EVCC all works fine. But now, it seems, that Opel cut all external way to get info from the car. So I am looking for an alternativ and found meatpi and the project here.
After step by step an the helpfully site with the example, I get my first MQTT-Message in the explorer. But it is not the SoC, it is the payload of the automation.
In CarScanner see the SoC (and other) but I did not know how to tranform the information from CarScanner to this PID.
[image]
Can someone help?
Thanks

chris58008, 2024-04-06:
Did you look at this guide? https://github.com/meatpiHQ/wican-fw/wiki/EV-Battery-SoC-in-Home-Assistant-%E2%80%90-Example

jove01, 2024-04-06:
Yes, that is my guideline. But the result in the mqtt-explorer ist not the Soc, I see

{"bus":"0","type":"rx","ts":21782,"frame":[{"id":1995,"dlc":8,"rtr":false,"extd":false,"data":[5,98,3,8,2,69,204,204]}]}

 

Tomorrow I can show it as a copy from the explorer.

Thanks

jove01, 2024-04-07:
Today I get no connection to the ECU (yesterday no problem)
I found this [https://github.com/meatpiHQ/wican-fw/issues/17#issuecomment-1456925171](url) with the settings for CarScanner. It did not help.
Tomorrow I will look deeper in this thread, or do you have a tip.

meatpiHQ, 2024-04-07:
@jove01 please post a screenshot of the device configuration page.

jove01, 2024-04-07:
> @jove01 please post a screenshot of the device configuration page.

I will do it in the evening

jove01, 2024-04-07:
First the settings in WiCan

[image]
[image]

[image]
[image]
[image]

Than HA Automation
`service: mqtt.publish
data:
  qos: 0
  retain: false
  topic: wican/5432049xxxxx/can/tx
  payload: >-
    {"bus":"0","type":"tx","ts":21767,"frame":[{"id":1931,"dlc":8,"rtr":false,"extd":false,"data":[3,34,3,8,170,170,170,170]}]}
`
The Result
`{"bus":"0","type":"tx","ts":21767,"frame":[{"id":1931,"dlc":8,"rtr":false,"extd":false,"data":[3,34,3,8,170,170,170,170]}]}`

Car Scanner
[image]

I reinstalled Car Scanner, but **no connection to ECU**!

Than I tried it with iCar Pro and SkyScanner, no Problem

Maybe the ECU-Problem ist the basic-problem. And I did not know, what ist happend to yesterday, I had the connection and I see  the SoC!

jove01, 2024-04-07:
So tried the HA AddOn PSA Car controller again. And after many tries the add On works. So I will  stop the WiCan implementation. Maybe later I try it again, I think it is a good project.
Godd luck and many thnanks.

closed 2024-04-07 by jove01