meatpiHQ/wican-fw #117: Hardware Revision Detection

mickeyl, 2024-04-10:
Although we have our own custom hardware up and running now, we have about 30 WiCAN that we still use for testing. I have recently got my first WiCAN USB variant.

I wonder whether you implemented any revision detection hardware? From earlier projects where I was involved, I remember that people used resistors / custom voltages they could read from the software to decide which hardware revision they're on.

Is anything along these lines implemented in your design? Or is it merely encoded in NVS once you initially preprogram the device?

meatpiHQ, 2024-04-15:
@mickeyl there's no reversion detection each has there own binary. I investigated the reversion detection recently for a new product, but I couldn't really find a nice way to do it. Having a voltage divider is a good way to do this but the ADC on the esp32 is pretty crappy.

https://github.com/meatpiHQ/wican-fw/releases/tag/v2.98

mickeyl, 2024-04-15:
Alright, thanks!

closed 2024-04-15 by mickeyl